Portisch/RF-Bridge-EFM8BB1 #4: Original Firmware

dellk, 2018-02-22:
Hi,
Can you please include the original firmware (as .hex) for the Sonoff EFM8BB1. I'd like to flash back. Since the new firmware only one remote is recognized by the RF Bridge.

dellk, 2018-02-23:
Flashed OpenMQTTGateway instead of Espurna to test and at least one of the old remotes is being recognized again. 

Original EFM8BB1 firmware would be good nonetheless.

AlbertWeterings, 2018-02-25:
Thank you for the great work on this project. Sorry I agree with @dellk and also want to be able to get back to the original firmware. At this point your firmware is not stable enough for use for sure I'm willing to test new versions later on.

Portisch, 2018-02-26:
https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/8b3597517c67628677eae568aef3a520dcb0ae9f

closed 2018-02-26 by Portisch

dellk, 2018-02-26:
Thank you. Will test and update.

dellk, 2018-02-26:
Unfortunately it doesn't work. Flashing goes fine, but nothing gets received or sent.