Portisch/RF-Bridge-EFM8BB1 #6: Getting x: 0x43 in return

mcoenen1994, 2018-02-28:
Hi,

I'm trying to program the EFM but getting x: 0x43 in my terminal for a really long time.
Am I doing something wrong?

Thanks in advance,

Maikel

ErikAndren, 2018-02-28:
Is this using the instructions from https://github.com/conorpp/efm8-arduino-programmer ?

Portisch, 2018-02-28:
0x43 means wrong CRC on the received data. (Between host and Arduino)

ErikAndren, 2018-03-01:
Are you connecting the ground pin?
Are the cables long?

mcoenen1994, 2018-03-01:
I've connected 3.3, GND, C2D and C2CLK.
cables are around 20cm, I will try with shorter ones when I'm home.

mcoenen1994, 2018-03-05:
@ErikAndren Tried with shorter cables but no luck...
I'm using a clone arduino mega 2560, could that be a problem?

ErikAndren, 2018-03-05:
I'm using a clone nano myself with sucess but who knows?
To be noted is that I'm using a 5V to 3.3 V level shifter. 
I do not know if this really is necessary.

Are you powering the board via its usb-interface?

mcoenen1994, 2018-03-05:
I tried with powering via usb and via external supply, no difference...
Also for the bridge I tried the usb and 3.3 supply

ErikAndren, 2018-03-05:
Running out of ideas here. 
Try different pins on the arduino, try different arduino, try different host.

mcoenen1994, 2018-03-05:
I'm getting an other error with different pins:

Connected
x: 0x81
PI initiated
127.0.0.1 - - [05/Mar/2018 16:36:25] "POST / HTTP/1.1" 500 -
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1997, in __call__
    return self.wsgi_app(environ, start_response)
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1985, in wsgi_app
    response = self.handle_exception(e)
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1540, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/Library/Python/2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/Users/mcoenen/Downloads/efm8-arduino-programmer-master/prog_server.py", line 117, in hello
    programmers[port].prog(firmware)
  File "/Users/mcoenen/Downloads/efm8-arduino-programmer-master/prog_server.py", line 42, in prog
    assert(0x84 == struct.unpack('B', self.ser.read(1))[0])
error: unpack requires a string argument of length 1

AlbertWeterings, 2018-03-05:
What kind of Arduino do you use in between your PC and EMF8 @mcoenen1994 ? I also struggled for some time but finally got it to work. With a Leonardo I used this code and pins are 3 for C2Ck 4 for CD.

[C2_programmer.zip](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/1781781/C2_programmer.zip)

mcoenen1994, 2018-03-05:
@AlbertWeterings I'm using a Mega 2560 clone. It doesn't have a port-D if I'm right.

AlbertWeterings, 2018-03-05:
@mcoenen1994 I expect the clone has the same chip, maybe parts are different routed to the outside of the board. I would try to write a small sketch (blinking LED's) to check if you are on the right ports. Otherwise if you have a other Arduino around use that one.

Portisch, 2018-03-05:
Do you use Windows? I have a small (just for testing) self written Windows program to send hex files over the COM port to the Arduino with the efm8-arduino-programmer firmware.

ErikAndren, 2018-03-06:
https://www.silabs.com/community/mcu/8-bit/knowledge-base.entry.html/2015/06/08/how_to_use_efm8_star-8aHu

Might be useful, especially the part about 1kOhm pull up resistor on C2CK

AlbertWeterings, 2018-03-06:
@mcoenen1994 I found this [project ](https://github.com/lhartmann) by Lucas Hartmann eventualy it did not compile but after making [these ](https://github.com/lhartmann/c2_prog_wifi/pull/3/commits/bf877c82740ba6efbf1ba941b44467914c916ac0) changes to httpServer.h  and c2_prog_wifi.ino it compiled without a problem in the Arduino IDE. 

Than still it did not work but with the additional pull up resistor mentioned by @ErikAndren it started to work.  

A few thing you need to keep in mind pin mapping in c2.h are pointing to GPIO pins (don't make my mistake and take D0 and D2 on a LoLin board). Microsoft Edge and Internet explorer don't work, firefox works. Programming takes minutes sometimes it look if it got stuck but keep waiting it will continue.

If you write @Portisch firmware the bridge will beep when programming is ready.

mcoenen1994, 2018-03-06:
With an other arduino it succeeded! I now used an original arduino uno. I think it went wrong with the pin-mapping of the arduino mega clone but not sure.

Thanks for the input!

ErikAndren, 2018-03-06:
Did you use a level-shifter or did you flash directly from the arduino (i. e. did you use 5V i/o voltage)?

mcoenen1994, 2018-03-06:
I used the Arduino Uno with only the usb connection to my macbook. The efm was connected via the C2 pins and the ground.

It beeped after programming but not yet able to learn my Sonoff motion detector, but probably that has nothing to do with the efm:)

AlbertWeterings, 2018-03-06:
@mcoenen1994  Portisch firmware is not yet able to handle RF codes from all hardware see [#5](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/5) I hope @Portisch finds time soon to make updates that add more hardware and solves some issues.

closed 2018-03-13 by Portisch