Portisch/RF-Bridge-EFM8BB1 #7: Linker errors

evugar, 2018-03-11:
Tried to import project to Simplicity Studio and build it, got a bunch of L127 and L128 linker errors. How to fix this?

=========================================

16:35:00 **** Incremental Build of configuration Keil 8051 v9.53 - Release for project RF-Bridge-EFM8BB1-master ****
make all 
PATH=/c/Prog/SiliconLabs/SimplicityStudio/v4/developer/toolchains/keil_8051/9.53/BIN:/c/Prog/SiliconLabs/SimplicityStudio/v4/support/8051:/bin:/c/Prog/SiliconLabs/SimplicityStudio/v4/features/com.silabs.external.java.windows.x86_64.feature_1.8.0.92/jre/bin/server:/c/Prog/SiliconLabs/SimplicityStudio/v4/features/com.silabs.external.java.windows.x86_64.feature_1.8.0.92/jre/bin:/c/Prog/SiliconLabs/SimplicityStudio/v4/features/com.silabs.external.java.windows.x86_64.feature_1.8.0.92/jre/lib/amd64:/c/Windows/system32:/c/Windows:/c/Windows/System32/Wbem:/c/Windows/System32/WindowsPowerShell/v1.0/:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/c/devel/avr8-gnu/bin:/c/devel/Python27/:/c/devel/Python27/Scripts:/c/Program Files (x86)/ZyXEL PLA Series Utility/ZyXEL PLA Series Utility/cml:/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/c/Prog/git/bin:/c/Prog/git/usr/bin:/c/Prog/arduino-1.8.2/hardware/tools/avr/bin:/c/Program Files/Intel/WiFi/bin/:/c/Program Files/Common Files/Intel/WirelessCommon/:/c/WINDOWS/system32:/c/WINDOWS:/c/WINDOWS/System32/Wbem:/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/c/devel/Doxygen/bin/Release64:/c/Prog/protoc/bin:/c/Program Files (x86)/ATMEL/FLIP 2.4.6/bin:%USERPROFILE%/AppData/Local/Microsoft/WindowsApps:/c/Program Files (x86)/ZyXEL PLA Series Utility/ZyXEL PLA Series Utility/cml:.:/c/prog/SiliconLabs/SimplicityStudio/v4
 
Building file: ../src/Globals.c
Invoking: Keil 8051 Compiler
C51 "@src/Globals.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.

C51 COMPILATION COMPLETE.  0 WARNING(S),  0 ERROR(S)
Finished building: ../src/Globals.c
 
Building file: ../src/InitDevice.c
Invoking: Keil 8051 Compiler
C51 "@src/InitDevice.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.

C51 COMPILATION COMPLETE.  0 WARNING(S),  0 ERROR(S)
Finished building: ../src/InitDevice.c
 
Building file: ../src/RF_Bridge_main.c
Invoking: Keil 8051 Compiler
C51 "@src/RF_Bridge_main.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.
*** WARNING C294 IN LINE 70 OF C:\Users\Vugar\SimplicityStudio\v4_workspace\RF-Bridge-EFM8BB1-master\src\RF_Bridge_main.c: unreachable code

C51 COMPILATION COMPLETE.  1 WARNING(S),  0 ERROR(S)
Finished building: ../src/RF_Bridge_main.c
 
Building file: ../src/RF_Handling.c
Invoking: Keil 8051 Compiler
C51 "@src/RF_Handling.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.

C51 COMPILATION COMPLETE.  0 WARNING(S),  0 ERROR(S)
Finished building: ../src/RF_Handling.c
 
Building file: ../src/SILABS_STARTUP.A51
Invoking: Keil 8051 Assembler
AX51 "@src/SILABS_STARTUP.__ia" || test $? -lt 2

MACRO ASSEMBLER AX51 V3.12.0.0
COPYRIGHT KEIL ELEKTRONIK GmbH 2000 - 2013
"c:\Prog\SiliconLabs\SimplicityStudio\v4\developer\toolchains\keil_8051\9.53\BIN\AX51.exe" "C:\Users\Vugar\SimplicityStudio\v4_workspace\RF-Bridge-EFM8BB1-master\src\SILABS_STARTUP.A51" ERRORPRINT SET (SMALL,SILABS_STARTUP=1) DEBUG MACRO NOMOD51 INCDIR(C:\Users\Vugar\SimplicityStudio\v4_workspace\RF-Bridge-EFM8BB1-master\inc;C:/Prog/SiliconLabs/SimplicityStudio/v4/developer/sdks/8051/v4.1.1//Device/shared/si8051base;C:/Prog/SiliconLabs/SimplicityStudio/v4/developer/sdks/8051/v4.1.1//Device/EFM8BB1/inc;C:/Prog/SiliconLabs/SimplicityStudio/v4/developer/sdks/8051/v4.1.1//Device/EFM8BB1/peripheral_driver/inc) PRINT(.\src\SILABS_STARTUP.lst) COND SYMBOLS PAGEWIDTH (120) PAGELENGTH (65) OBJECT(.\src\SILABS_STARTUP.OBJ)

ASSEMBLY COMPLETE.  0 WARNING(S), 0 ERROR(S).

Finished building: ../src/SILABS_STARTUP.A51
 
Building file: ../src/uart.c
Invoking: Keil 8051 Compiler
C51 "@src/uart.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.
*** WARNING C185 IN LINE 27 OF C:\Users\Vugar\SimplicityStudio\v4_workspace\RF-Bridge-EFM8BB1-master\src\uart.c: 'TX_Finished': different memory space

C51 COMPILATION COMPLETE.  1 WARNING(S),  0 ERROR(S)
Finished building: ../src/uart.c
 
Building file: ../lib/efm8bb1/peripheralDrivers/src/uart_0.c
Invoking: Keil 8051 Compiler
C51 "@lib/efm8bb1/peripheralDrivers/src/uart_0.__i" || test $? -lt 2

C51 COMPILER V9.53.0.0 - SN: K1RLC-00939Z
COPYRIGHT Copyright (C) 2012 - 2014 ARM Ltd and ARM Germany GmbH. All rights reserved.

C51 COMPILATION COMPLETE.  0 WARNING(S),  0 ERROR(S)
Finished building: ../lib/efm8bb1/peripheralDrivers/src/uart_0.c
 
Building target: RF-Bridge-EFM8BB1-master.omf
Invoking: Keil 8051 Linker
LX51 "@RF-Bridge-EFM8BB1-master.lnp" || test $? -lt 2

LX51 LINKER/LOCATER V4.66.30.0 - SN: K1RLC-00939Z
COPYRIGHT ARM Germany GmbH 1995 - 2014
@RF-Bridge-EFM8BB1-master.lnp "./lib/efm8bb1/peripheralDrivers/src/uart_0.OBJ",
"./src/Globals.OBJ",
"./src/InitDevice.OBJ",
"./src/RF_Bridge_main.OBJ",
"./src/RF_Handling.OBJ",
"./src/SILABS_STARTUP.OBJ",
"./src/uart.OBJ"
TO "RF-BRIDGE-EFM8BB1-MASTER.OMF.CRBUILD" REMOVEUNUSED PRINT(.\RF-Bridge-EFM8BB1-master.m51) PAGEWIDTH (120) PAGELENGTH (65) CLASSES( ECODE(C:0x0 - C:0x1ffe), HCONST(C:0x0 - C:0x1ffe), XDATA(X:0x0 - X:0xff), HDATA(X:0x0 - X:0xff))

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  PCA0_halt
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  _PCA0_writeChannel
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  _PCA0_writeChannel
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1000F47H

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 100001FH

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1000E6FH

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_halt
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1000FADH

Program Size: data=79.2 xdata=199 const=195 code=4162
LX51 RUN COMPLETE.  0 WARNING(S),  7 ERROR(S)
make: *** [RF-Bridge-EFM8BB1-master.omf] Error 1

16:35:02 Build Finished (took 2s.563ms)

ErikAndren, 2018-03-11:
If I recall correctly the problem is that the pca_0.c file is excluded from building for some strange reason. If you dig around a bit in the project / build settings you should be able to find the exclusion pattern and remove it.

Portisch, 2018-03-11:
I do have same problem when opening the project first time. Just go to the project properties and search for the project modules tab. Thenclick apply. Then it is compiling without any problems.

closed 2018-03-11 by Portisch