Portisch/RF-Bridge-EFM8BB1 #9: Can't write efm8

ciotlosm, 2018-03-14:
Hey, sorry to pollute the isues here, but you guys seem to be really knowledgable about writing firmware on this chip. It seems I messed up trying to write some data on the chip and now I can't upload anymore the custom firmware to test myself the behaviour. I've opened this issue:
https://github.com/conorpp/efm8-arduino-programmer/issues/5 on the repo for the uploader, however I am running out of ideas on how to fix it. 

I really hope you guys can give me some direction as I'd love to test as well changes to your firmware.

Portisch, 2018-03-14:
The 0x43 from the https://github.com/conorpp/efm8-arduino-programmer doesn't mean the EFM8 chip is bricked. Only that the data between host and https://github.com/conorpp/efm8-arduino-programmer have wrong crc. So the https://github.com/conorpp/efm8-arduino-programmer is ignoring the data and don't flash it to the EFM8 chip.

So you will need to fix the host application to do the command 0x03 right.

ciotlosm, 2018-03-14:
Seems it was because of the serial speed set in both the Arduino sketch and the server uploader. The Arduino Uno I was now using was not handling the initial speed. I switched to 57600 in both the sketch and server uploader and it uploaded.

closed 2018-03-14 by ciotlosm