Portisch/RF-Bridge-EFM8BB1 #10: Adding new protocols

Sk4zz, 2018-03-15:
Firstly, thank you for your nice work!
I have some RF power outlets at home which don't work together (yet) with my Sonoff RF bridge running ESPurna. It seems the outlets use the Intertechno v3 protocol and I can sniff the signals with an arduino. I have had a look into the RF_Protocols.h but I have no Idea how to get those timings used to specify a new protocol. Those are not the bucket timings, are they? Here is one example of a sniffed code using the compressed.ino example from the pimatic project.

`b: 260 2532 1312 10752 0 0 0 0 
t:010002000200020200000202000200020000020002000202000200020000020002000200020002000200020002000200020002000200020200000200020002020003
`
Thank you very much!

Portisch, 2018-03-15:
Looks like bucket. No host software is right now supporting this. Bucket isn't defined in the protocols. Just enable manually bucket sniffing by the command 0xB1 and try again.

Sk4zz, 2018-03-16:
Yes, the output I posted are bucket timings. My question was how to get the timings that are required in the RF_Protocols.h to specify a new protocol.
I did some more research and it turns out that my outlets use the KaKu (Klik-aan-klik-uit) remote protocol. I also came across the NewRemoteSwitch library by Randy Simons [https://github.com/hjgode/homewatch/tree/master/arduino/libraries/NewRemoteSwitch](https://github.com/hjgode/homewatch/tree/master/arduino/libraries/NewRemoteSwitch) There is also a description of the protocol in that library:
```
= Protocol of klikaanklikuit A-series devices =

Copied from Wieltje, http://www.circuitsonline.net/forum/view/message/1181410#1181410,
but with slightly different timings, as measured on my device.
        _   _
'0':   | |_| |_____ (T,T,T,5T)
        _       _
'1':   | |_____| |_ (T,5T,T,T)
        _   _
'dim': | |_| |_     (T,T,T,T)

T = short period of ~260µs. Use the ShowReceivedCodeNewRemote example to find the
actual period length for your devices.

A full frame looks like this:

- start pulse: 1T high, 10.44T low
- 26 bit:  Address
- 1  bit:  group bit
- 1  bit:  on/off/[dim]
- 4  bit:  unit
- [4 bit:  dim level. Only present of [dim] is chosen]
- stop pulse: 1T high, 40T low
```
So my question remains the same. Is there a way to determine the following values from this description of the protocol. And if not, what would be necessary to do so?
```
	// normal high signal time on sync pulse
	uint16_t SYNC_HIGH;
	// normal low signal time on sync pulse
	uint16_t SYNC_LOW;
	// bit count of SYNC bits
	uint8_t SYNC_BIT_COUNT;
	// high time of a logic bit 1
	uint16_t BIT_HIGH_TIME;
	// high time of a logic bit 0
	uint16_t BIT_LOW_TIME;
	// duty cycle for logic bit 1
	uint8_t BIT_HIGH_DUTY;
	// duty cycle for logic bit 0
	uint8_t BIT_LOW_DUTY;
	// bit count for this protocol
	uint8_t BIT_COUNT;
```
Right now I'm not trying to learn my remote to the RF bridge. I'm merely trying to understand this whole RF business and I'm hoping for the integration of your modded firmware into ESPurna (maybe even into Tasmota?). Until then I would like to help to extend the number of supported protocols.

While I'm at it: I enabled the RAW support in ESPurna but I don't know, what I should paste into the webinterface to send commands to my outlets. Can you give me a hint on that?

Portisch, 2018-03-16:
Will check it next week. The bucket sniffing 0xB1 is also not supported by espurna yet. So please use the RS232 connection to the EFM8 direct with baud 19200 8N1 if the firmware can sniff the buckets.

If it can be sniffed we can think further to implement a protocol.

Sk4zz, 2018-03-16:
I must confess that I don't know how to establish a serial connection to the EFM8. Here's what I tried: I used a FTDI USB TTL FT232RL board that I had lying around to connect to  the  upper GND, TX, RX,
 3v3 headers in the middle of the following image (which is from the tasmota wiki https://github.com/arendst/Sonoff-Tasmota/wiki/How-to-Flash-the-RF-Bridge) 
[image]
and then tried to send the 0xB1 to the EFM8 using the serial console of atom by typing "0xB1" or "B1" but there was no response.
I'm certainly doing something wrong here and I'm sorry if this is a silly question but I couldn't find any resources on how to connect EFM8. I also searched for ways to establish a serial connection to the EFM8 via C2 but I could't come up with a solution.
Thanks for your help!

Portisch, 2018-03-16:
Take a look near C26. There is a second RX & TX. This is the RS233 header for the EFM8. Also put the switch in off position to cancel the RS232 connection to the ESP8266.

From top to bottom: GND, TX, RX, 3V3.

Portisch, 2018-03-21:
@Sk4zz Please post a sniffed data by command 0xB1. With the data it is maybe possible to see if it is a "normal" protocol like here: https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/12#issuecomment-374895344

Sk4zz, 2018-03-21:
I just had a quick look at the other thread. My outlets seem fairly similar to the Aldi ones. I'll try to sniff the protocol later today.

Sk4zz, 2018-03-21:
Ok, I used @AlbertWeterings instructions from the other thread and was able to sniff the protocol. It seems to be different from the Aldi protocol though.
`
[795587] [RFBRIDGE] Received 0xAA  -> UART sync
[795592] [RFBRIDGE] Received 0xB1  -> bucket sniffing
[795592] [RFBRIDGE] Received 0x05  -> bucket count = 5

[795593] [RFBRIDGE] Received 0x01  
[795593] [RFBRIDGE] Received 0x12  -> 0x0112 = 274ms

[795593] [RFBRIDGE] Received 0x09  
[795593] [RFBRIDGE] Received 0xE2   -> 0x09E2 = 2530ms 

[795593] [RFBRIDGE] Received 0x00
[795594] [RFBRIDGE] Received 0xE6  -> 0x00E6 = 230ms

[795594] [RFBRIDGE] Received 0x05
[795594] [RFBRIDGE] Received 0x03  -> 0x0503 = 1283ms

[795594] [RFBRIDGE] Received 0x2A
[795594] [RFBRIDGE] Received 0x08  -> 0x2A08 = 10760ms

[795595] [RFBRIDGE] Received 0x01
[795595] [RFBRIDGE] Received 0x00
[795595] [RFBRIDGE] Received 0x03
[795595] [RFBRIDGE] Received 0x00
[795595] [RFBRIDGE] Received 0x03
[795595] [RFBRIDGE] Received 0x00
[795596] [RFBRIDGE] Received 0x03
[795596] [RFBRIDGE] Received 0x03
[795596] [RFBRIDGE] Received 0x00
[795596] [RFBRIDGE] Received 0x00
[795596] [RFBRIDGE] Received 0x03
[795597] [RFBRIDGE] Received 0x03
[795597] [RFBRIDGE] Received 0x00
[795597] [RFBRIDGE] Received 0x03
[795597] [RFBRIDGE] Received 0x00
[795597] [RFBRIDGE] Received 0x03
[795597] [RFBRIDGE] Received 0x00
[795598] [RFBRIDGE] Received 0x00
[795598] [RFBRIDGE] Received 0x03
[795598] [RFBRIDGE] Received 0x00
[795598] [RFBRIDGE] Received 0x03
[795598] [RFBRIDGE] Received 0x00
[795598] [RFBRIDGE] Received 0x03
[795599] [RFBRIDGE] Received 0x03
[795599] [RFBRIDGE] Received 0x00
[795599] [RFBRIDGE] Received 0x03
[795599] [RFBRIDGE] Received 0x00
[795600] [RFBRIDGE] Received 0x03
[795601] [RFBRIDGE] Received 0x00
[795602] [RFBRIDGE] Received 0x00
[795602] [RFBRIDGE] Received 0x03
[795602] [RFBRIDGE] Received 0x00
[795603] [RFBRIDGE] Received 0x03
[795603] [RFBRIDGE] Received 0x00
[795603] [RFBRIDGE] Received 0x03
[795603] [RFBRIDGE] Received 0x00
[795604] [RFBRIDGE] Received 0x03
[795605] [RFBRIDGE] Received 0x00
[795606] [RFBRIDGE] Received 0x03
[795608] [RFBRIDGE] Received 0x00
[795608] [RFBRIDGE] Received 0x03
[795609] [RFBRIDGE] Received 0x00
[795610] [RFBRIDGE] Received 0x03
[795610] [RFBRIDGE] Received 0x00
[795611] [RFBRIDGE] Received 0x03
[795611] [RFBRIDGE] Received 0x00
[795612] [RFBRIDGE] Received 0x03
[795612] [RFBRIDGE] Received 0x00
[795613] [RFBRIDGE] Received 0x03
[795613] [RFBRIDGE] Received 0x00
[795613] [RFBRIDGE] Received 0x03
[795613] [RFBRIDGE] Received 0x00
[795613] [RFBRIDGE] Received 0x03
[795614] [RFBRIDGE] Received 0x00
[795614] [RFBRIDGE] Received 0x03
[795614] [RFBRIDGE] Received 0x03
[795614] [RFBRIDGE] Received 0x00
[795614] [RFBRIDGE] Received 0x00
[795614] [RFBRIDGE] Received 0x03
[795615] [RFBRIDGE] Received 0x00
[795615] [RFBRIDGE] Received 0x03
[795615] [RFBRIDGE] Received 0x00
[795615] [RFBRIDGE] Received 0x03
[795615] [RFBRIDGE] Received 0x03
[795616] [RFBRIDGE] Received 0x00
[795616] [RFBRIDGE] Received 0x04

[795616] [RFBRIDGE] Received 0x55  -> UART sync end

Thus the sniffed code is something like
AA B1 05 0112 09E2 00E6 0503 2A08 010003000300030300000303000300030000030003000303000300030000030003000300030003000300030003000300030003000300030300000300030003030004 55
I'm also attaching an image of my remote, the brand seems to be Flamingo.
[image]

Thanks again for your effort and your nice work!

Portisch, 2018-03-21:
The code sniffed by 0xB1 looks ok. But this is different timing as the protocols in the RF_Protocols.h.

https://forum.pilight.org/showthread.php?tid=1110&pid=20914#pid20914

I think you have to use 0xB1 for sniffing and 0xB0 for transmit. Take a look to the readme.

closed 2018-09-17 by Portisch