Portisch/RF-Bridge-EFM8BB1 #13: Bridge R2 V1.0

AlbertWeterings, 2018-03-20:
[image]

Picture of new release RF-Bridge

Portisch, 2018-03-20:
Something different for the firmware or just hardware improvement? The Step-Up to 12V for the SYN115 should bring more distance for transmit, isn't it?

Jason2866, 2018-03-20:
@AlbertWeterings Could you compare the firmware from V1 and V2 of the EFM chip?
If different could you upload the V2 version? I want to flash @Portisch firmware. It gives me a
good feeling if there a problems (with my remotes) that i could go back...
THX in advance

AlbertWeterings, 2018-03-20:
@Jason2866 @Portisch Firmware of EFM is the same in R1 and R2 Portisch firmware is working in both. Flashing ESPURNA is going wrong and becomes instable in R2 just do a Factory reset in the admin tab of ESPURNA and it will work.

Hardware dif.. Is a step-up to 12 volt for SYN115 and there are two IO lines connected to USB D+ and D- via a few diodes. I expect SonOff has created a kind of software USB port on the two IO lines of the ESP

Portisch, 2018-03-21:
@ALL
Yes, the firmware is the same! My flash backup (what wasn't working) is 100% the same. I just had additional 16 bytes in the front. That was the reason why it wasn't working. Removing these bytes shows that the firmware matches to 100%. I add the picture to the doc. THX

closed 2018-03-21 by Portisch

danielo515, 2021-11-03:
So the portish firware works on the R2 version of the sonoff bridge then? Does it have any limitation?