Portisch/RF-Bridge-EFM8BB1 #14: Problem with new Bridge

franki29, 2018-03-22:
Hi all, after i discovered that the code works well with my older bridge, i am wondering why it does not work with the new PCB  layout.  There is a new chip on board and after doing some messuring i noticed, that the SYN115 was replaced with a F115H and the new chip is a 12V "changer" becasue the F115H is working with 12V instead of 3.3 V. The question is now, is it possible to port the code to the new transmitter chip?

Thanks for helping

franki29, 2018-03-22:
Information about the new Chip 
http://www.gaily-gw.com/downloadRepository/9ae79667-c159-40a5-a97e-1efc22fa5682.pdf

Portisch, 2018-03-22:
The SYNF115H should be 1:1 compatible to the SYN115. The only thing you can do:
Take the "old" bridge and start 0xB1 bucket sniffing. Than do a transfer on the "new" bridge like AA A8 04 03 AA AA AA 55. Than we can see on the sniffed data from the "old" bridge the pulse timing of the "new" bridge. As compare you can do it the other way too. Put the "new" bridge into sniffing and make the transfer on the "old" bridge. Make each test 5 times.

franki29, 2018-03-22:
Hi, i already tested receiving the code from the old one to the new one and it was 1:1 OK.
But sending same code from the new one to the old one nothing is received, so i guess that the code is not send out. Maybe  i am searching on the wrong side and the new bridge is just broken.
I will try to mesure and what is quite strange that the PAOUT has 12V. I have no osszilloscope to mesure, so i have to look how i can get access to one.
Never mind, thanks for your help

closed 2018-03-22 by franki29