Portisch/RF-Bridge-EFM8BB1 #15: Rohrmotor24

mstuehn, 2018-04-28:
Hi,
unfortunately I'm not able to pair the sonoff-rf-bridge running espurna1.12.6a (RAW_RF enabled) with the remote of rohrmotor24. I've tried several things:

1. Start learning over web-interface -> timeout
2. tried a different remote (remote for sockets ) -> works
3. use new learning command A9 over mqtt -> beep sound indicated successful learning, telnet showed:
[597079] [MQTT] Received sonoff-bridge/rfraw/set => AAA955
[597086] [RFBRIDGE] Sending RAW MESSAGE 'AAA955'
[598942] [RFBRIDGE] Received 0xAA
[598945] [RFBRIDGE] Received 0xAB
[598947] [RFBRIDGE] Received 0x06
[598948] [RFBRIDGE] Received 0x02
[598949] [RFBRIDGE] Received 0x6C
[598950] [RFBRIDGE] Received 0x0A
[598952] [RFBRIDGE] Received 0x75
[598953] [RFBRIDGE] Received 0x31
[598954] [RFBRIDGE] Received 0x11
[598955] [RFBRIDGE] Received 0x55

So it seems to know this remote, but only if i use A9. What is the correct approach to pair the rohrmotor24-remote and the sonoff-rf-bridge?
Thanks for help.

mstuehn, 2018-05-06:
There seems to be no integration into the espurna framework, I followed these steps and it worked:
1. Activate new learning with command AAA955 via mqtt on address $nodename/rfraw/set
2. Note the answer, in my case AA AB 06 02 6C 0A 75 31 11 55
3. Exchange the ACK-Byte (AB) with the SEND-RAW command (A8) -> AA A8 06 02 6C 0A 75 31 11 55
4. Put this string (without spaces) into the sonoff web-frontend (RF-section) and click "save"

Afterwards the espurna firmware can be used to control the gpios as usual.

closed 2018-05-06 by mstuehn

Portisch, 2018-05-06:
Ist the motor moving? Ich never tried it...

mstuehn, 2018-05-06:
Yes, the shutter opens and closes according to the rf-code of the remote.