Portisch/RF-Bridge-EFM8BB1 #16: Would this firmware work with this device?

henfri, 2018-05-13:
Hello,

thanks for looking into the Firmware. I cannot learn [this ](http://www.seav.it/sites/default/files/prodotti/schede_tecniche/Txx%204%20GB.pdf)device (see: [here](https://github.com/xoseperez/espurna/issues/861))
Would the new firmware help me?

Greetings,
Hendrik

AlbertWeterings, 2018-05-15:
It looks like you have a PT2262 based remote,  it should have work with the original firmware as well as with this. Are you able to open the remote and confirm the type of chip inside?

Jason2866, 2018-06-11:
@henfri  Try it with the portisch firmware. I couldnt learn "2262" Code with orig. firmware.
Now with portisch firmware it works perfect. Take a look at TASMOTA firmware. In newest release the flasher software is included. Flashing EFM Fimware is so easy now.

closed 2018-06-19 by Portisch