Portisch/RF-Bridge-EFM8BB1 #18: Intertechno Code not recognized by 1 channel code learning relay

Jason2866, 2018-06-11:
Hello,
first great software, now my Intertechno Power Outlets work great with the RF-Bridge!
Now learning this codes works (correct). Everything works as it should :-)
I have bought this learning code relay:
https://www.aliexpress.com/item/Smart-Home-433MHz-220V-1-Channel-Wireless-Remote-Control-Switch-DIY-Receiver-Module-Support-433MHz-RF/32826285319.html?spm=a2g0s.9042311.0.0.27424c4d6WXBKO
If i programm it with the Transmitter from the Poweroutlets. It switches correct.
It doesnt react if i send the codes via RF Bridge. 
I have a learning remote too. I can programm this remote with the RF-Bridge with the IT-Codes from the relay. Now when i press the corresponding buttons on it, the relay switches as it should.
So the codes are correct at all. 
I tried this with 2262 Code (Intertechno) and 1527 Code. I have a hardcoded transmitter for both.
On both transmitters the relay is programmable and reacts as wanted.
The same picture here, if i learn this code to the bridge i cant switch the relay via the bridge. 
I cant learn any code sended from the RF-Bridge to the relay. 
Could it be that the RF-Bridge does not send as often the codes as other devices?

Portisch, 2018-06-19:
If anyobody do have problems of decoding a RF signal please use the 0xB1 sniffing command.
This will show the "waveform" of the RF signal. This data can be used to decode/implement a unknown protocol.

Jason2866, 2018-06-19:
Hello @Portisch 
Code with rc remote switching the relay
17:45:35 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:45:35 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:45:43 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:45:44 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:45:52 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:45:52 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:45:53 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:45:53 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:46:23 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010055"}}
17:46:23 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010055"}}
17:46:24 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000055"}}
17:46:24 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009600000055"}}
17:46:24 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000055"}}
17:46:24 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009600000055"}}
17:46:24 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000055"}}
17:46:26 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:46:26 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:46:34 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:46:35 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:46:38 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:46:38 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}

Code sended from Bridge not switching the relay
17:48:55 MQT: stat/sonoff-82589D/RESULT = {"RfSync":11320}
17:48:55 MQT: stat/sonoff-82589D/RESULT = {"RfLow":360}
17:48:55 MQT: stat/sonoff-82589D/RESULT = {"RfHigh":1090}
17:48:55 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:48:56 MQT: stat/sonoff-82589D/RESULT = {"RfCode":"#151514"}
17:48:56 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:48:58 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:48:58 MQT: stat/sonoff-82589D/RESULT = {"Delay":2}
17:48:58 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009600000155"}}
17:48:59 MQT: stat/sonoff-82589D/RESULT = {"RfSync":11320}
17:48:59 MQT: stat/sonoff-82589D/RESULT = {"RfLow":360}
17:48:59 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601010155"}}
17:48:59 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA42C380118009601000155"}}
17:48:59 MQT: stat/sonoff-82589D/RESULT = {"RfHigh":1090}
17:48:59 MQT: stat/sonoff-82589D/RESULT = {"RfCode":"#151511"}
17:49:00 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA055"}}

Jason2866, 2018-06-19:
The issue is not that the protocol is unknow. The original Rf remote (2262 code) works.
If i send this correct code via rf bridge it doesnt switch.
I have a learning remote too. If i learn this remote the codes sended from the rf bridge i can switch
the device! So the codes are in general correct. 
Here the codes sended and workimg from the learning remote
18:05:37 CMD: rfraw 177
18:05:37 MQT: stat/sonoff-82589D/RESULT = {"RfRaw":"ON"}
18:05:37 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AAA055"}}
18:05:41 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 03 0160 03F7 2440 01010110011001100101011001100110010101100101011002 55"}}
18:05:46 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 03 0170 03E7 242C 01010110011001100101011001100110010101100110010102 55"}}
18:05:50 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 03F8 2422 01010110011001100101011001100110010101100101011002 55"}}
18:06:02 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 05 021C 0776 014A 00BE 0F96 0101010104 55"}}
18:06:02 MQT: tele/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 03 021C 0780 0F8C 010102 55"}}

closed 2018-09-17 by Portisch