Portisch/RF-Bridge-EFM8BB1 #19: Compiling with simplicity studio 4 and keil not working

balou1974, 2018-06-24:
Hello!

I have a problem compiling efm8bb1-firmware using studio 4 and keil. It seems the linker is missing parts:

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  PCA0_halt
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L127: UNRESOLVED EXTERNAL SYMBOL
    SYMBOL:  _PCA0_writeChannel
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  _PCA0_writeChannel
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1001137H

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 100001CH

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_run
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1000F7DH

*** ERROR L128: REFERENCE MADE TO UNRESOLVED EXTERNAL
    SYMBOL:  PCA0_halt
    MODULE:  ./src/RF_Handling.OBJ (RF_HANDLING)
    ADDRESS: 1001176H

Program Size: data=80.2 xdata=205 const=252 code=4649
LX51 RUN COMPLETE.  0 WARNING(S),  7 ERROR(S)
make: *** [RF-Bridge-EFM8BB1.omf] Error 1

ErikAndren, 2018-06-24:
See

https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/7

Portisch, 2018-06-24:
Thx @ErikAndren

https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/7#issuecomment-372132829

closed 2018-06-24 by Portisch