Portisch/RF-Bridge-EFM8BB1 #20: upgrade firmware

tartesso26, 2018-06-27:
i´m triying to upgrade your firmaware after upgrade tastoma, in tastoma doc says you can easily upgrade with web gui, uploading file, i go to upgrade section in tastoma device page, and i have the upload bottom there, but ... which file do i have to upload??

tartesso26, 2018-06-27:
i tried desired RF_Bridge.hex but get error, magic word is wrong

ErikAndren, 2018-06-27:
Since web interface upgrading is done in the tasmota firmware I think you are better off asking in that forum; https://github.com/arendst/Sonoff-Tasmota/

Portisch, 2018-06-27:
Magic word means the sign of the file format. The HEX file included in this git is the original HEX file produced by silabs application.

newingit18, 2018-07-01:
For uploading the RF_bridge.hex-file you have to have the Tasmota-Version 6.0.xxx first at the Sonoff-bridge. When this is properly running(check access to the RF-Bridge) you must prepare the hardware for upgrade (see https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433 - the last topic) of the EFM8BB1. Ther are two options - one with cutting two strip lines on the circuit board and one doing it in another way. I used the first method. This worked for me with a success message at the webinterface of the RF-Bridge.

mr-sneezy, 2018-07-02:
@tartesso26 
I think I know what your issue with the 'magic word hex error' is as I just had it and had to debug the problem myself. It was the method I was using to save the file from GitHub... 

One way fails, another way works.
 If I use 'save as' right clicking the RF_bridge.hex, the saved file fails the upgrade process in the Tasmota FW (6.0.0.b) giving me a 'magic word error'. The clue was the file size, big, at 148KB.
Instead, if I open up the RF_bridge.hex file on Github (left click) and view the hex bytes, then click 'raw' on the menu to get a raw view and save THAT hex text list to a txt file, it does work like expected. That resultant file is only 19KB in size.
Somebody who knows Git better can explain why I hope.

mr-sneezy, 2018-07-02:
Oh this is priceless fun :)
Updating the EFM8 firmware fixed the issues I had learning the code from an new PIR I bought last week, so great joy there, but has now broken the learned code of my front door bell so it's not detected anymore at all.  It won't relearn successfully either, so one step forward, and one step back...

Marty56, 2018-07-17:
@mr-sneezy 
The question asked - I think - was not about the specific hardware modification which are simple to understand but more to where in the Tasmota WebUi one can find support of flashing the EFM8 firmware.
I only see support for flashing the Tasmota firmware but no support for flashing the EFM8.

mr-sneezy, 2018-07-17:
LOL, I think you made the exact same assumption (mistake) I did. It's the option below the OTA Url flash option.
It's the 'Upgrade by file upload' option. 
I assumed for a few hours that they were two ways to update just the bridge FW, not the EFM8 chip.
Try it and confirm.

Marty56, 2018-07-17:
@mr-sneezy .Thanks for your tip. I will try.

closed 2018-08-14 by Portisch