Portisch/RF-Bridge-EFM8BB1 #23: Understanding/mimicing Rfraw 166 & 177 codes (Dooya DC2700 remotes)

autofrank, 2018-08-06:
Rf Bridge running 6.1.1 with Portisch firmware upgrade. All upgrades successful.

I have a DC2700 Dooya Remote (https://www.aliexpress.com/item/Dooya-Remote-Curtain-Accessories-Remote-Control-Transmitter-DC1600-For-DT52E/32613885486.html ) for a projector screen. It is a simple three button remote UP/STOP/DOWN

I can sniff the button presses using rfraw 166
21:53:57 CMD: rfraw 166
21:53:57 SRC: WebConsole from 192.168.1.137
21:53:57 RSL: Group 0, Index 1, Command RFRAW, Data 166
21:53:57 RSL: RESULT = {"RfRaw":"ON"}
21:53:57 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
21:54:01 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC13C55"}}
21:54:14 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
21:54:19 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC11E55"}}

This represents three button presses

- DOWN
- STOP
- UP

I can also sniff the code using rfraw 177

21:49:54 CMD: rfraw 177
21:49:54 SRC: WebConsole from 192.168.1.137
21:49:54 RSL: Group 0, Index 1, Command RFRAW, Data 177
21:49:54 RSL: RESULT = {"RfRaw":"ON"}
21:49:54 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
21:49:57 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55"}}
21:50:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 06 12D4 05D2 02A4 01D5 0102 226A 0123422323234242424223424242424223234242232342234223234242424242234223422342234220 55"}}
21:50:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 06 12DE 05C8 028C 0196 00FA 2242 0123322323233232323223323232323223233232232332233223233232323232233223322332233220 55"}}
21:51:15 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05C8 02D5 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232322323232330 55"}}

This represents the same three button presses

I can't find the Dooya protocol and having difficulty in sending the code from the bridge and control the screen

I'm also not sure if it is easier to decode/understand the rfraw166 or rfraw 177 result

Any help would be appreciated 
Thanks

autofrank, 2018-08-06:
I've been trying to decipher this using the example in the HT6P20B post

This is what I have

Code sniffed 
21:49:57 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55"}}

.. and the data part
AA B1 05 12DE 05D2 02CC 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232232323233230 55

separating out the data...
AA B1 05 12DE 05D2 02CC 0172 23A0 0 12 33 22 32 32 33 23 23 23 22 33 23 23 23 23 22 32 33 23 22 32 33 22 33 22 32 33 23 23 23 23 22 33 23 22 32 32 32 33 23 0 55

I need to add the len and the repeat time
this has 51 part = len (dec 51 = 33 hex)
copy repeat of 20 (20 dec = 14 in hex)
replacing the B1 with a B0 to 'send/transmit'

and this gives me 
AA B0 33 05 14 12DE 05D2 02CC 0172 23A0 0 12 33 22 32 32 33 23 23 23 22 33 23 23 23 23 22 32 33 23 22 32 33 22 33 22 32 33 23 23 23 23 22 33 23 22 32 32 32 33 23 0 55

.. but when I enter this with an rfraw command in the console
I get the following but no screen movement

12:25:20 RSL: RESULT = {"RfRaw":"ON"}
12:25:29 CMD: rfraw AAB033051412DE05C802CF017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323055
12:25:29 SRC: WebConsole from 192.168.1.137
12:25:29 RSL: Group 0, Index 1, Command RFRAW, Data AAB033051412DE05C802CF017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323055
12:25:29 RSL: RESULT = {"RfRaw":"ON"}
12:25:31 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}

I tried adjusting the timing aspect to the most common from 10 sample commands but nothing
I tried the same with the other 2 commands and nothing either 

Any help appreciated 

PS _ I have read about adding new protocols but not sure if this is the next route

Portisch, 2018-08-06:
1. AA A6 06 02 B8419AC13C 55
This means the signal got decoded as protocol type 0x02 (ROHRMOTOR24).
I don't know if this can be true or not. The data is 40 bit for this protocol so you can try to send it like this: AA A8 06 02 B8419AC13C 55
or AAA80602B8419AC11E55 for the other command. The AAA60602B8419AC155 looks incorrect.

2. AA B1 05 12DE 05C8 02D5 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232322323232330 55

This should be/is also 40 bit.
But there is something wrong in the sync bit "attachment" on the log. It should look like this:
AA B1 05 12DE 05C8 02D5 0172 23A0 012332232323323232322332323232322323323223233223322323323232323223323232232323233**4** 55

So try this:
AA B0 35 05 14 12DE 05C8 02D5 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232322323232334 55

0xAA: sync start
0xB0: command
0x35: len command
0x05: bucket count
0x14: repeats
buckets 0-4
data
0x55: sync end

autofrank, 2018-08-06:
Thanks @Portisch 

The following two worked perfectly
AAA80602B8419AC11E55 (up)
AAA80602B8419AC13C55 (down)

I re'sniffed' the third command (stop) and it is definitely reading different but consistent at 
16:47:59 RSL: Group 0, Index 1, Command RFRAW, Data 166
16:47:59 RSL: RESULT = {"RfRaw":"ON"}
16:47:59 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
16:48:09 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:09 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:16 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:16 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:17 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:19 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}
16:48:19 RSL: RESULT = {"RfRaw":{"Data":"AAA60602B8419AC155"}}

the longer versions also work for the up and down
AAB035051412DE05C802D5017223A0012332232323323232322332323232322323323223233223322323323232323223323232232323233455
AAB035051412DE05C802D5017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323455

the third command (stop) is also still an issue using this format...
AA B1 06 12D4 05D2 02A4 01D5 0102 226A 0123422323234242424223424242424223234242232342234223234242424242234223422342234220 55
it appears to have an extra timing bucket (6 instead of 5) so I tried increasing the len by 1 (to 35) and the buckets to 6

.. giving me 
AA B0 36 06 14 12D4 05D2 02A4 01D5 0102 226A 0123422323234242424223424242424223234242232342234223234242424242234223422342234224 55

but that didn't seem to work either ..
any thoughts on that one ?

autofrank, 2018-08-06:
sorted it... I completed another sniff of the 'stop' and I got a 5 bucket code and following the same process i got a working code

so for future reference .. the remote is the DC2700
[image]

and the codes that work for me are as follows

screen up (1st button)
AAB035051412DE05C802D5017223A0012332232323323232322332323232322323323223233223322323323232323223323232232323233455

screen stop (2nd button)
AAB035051412DE05D202CF01732242012332232323323232322332323232322323323223233223322323323232323223322332233223322455

screen down (3rd button)
AAB035051412DE05C802D5017223A0012332232323323232322332323232322323323223233223322323323232323223323223232323323455

there is a fourth button position on the bezel but no physical button behind it

Thanks again to @Portisch for creating the fw and for helping me solve this one...

BTW - I use fibaro HC2 as my home automation and will publish a virtual device sometime in the future for these actions

closed 2018-08-06 by autofrank

tadder, 2019-03-27:
Can anyone assist please? I am trying to understand and implement decode of my “Home Easy” switches.

I am using Tasmota 6.5.0 on a Sonoff RF bridge with Portisch RF-Bridge-EFM8BB1-20181127.hex.

‘Sniffing’ produces the output listed below when applied to BitBucketConverter181002.

My problem is that despite reading the suggested explanation I don’t understand how to answer the SYNC_HIGH, SYNC_LOW, and BIT_HIGH_TIME requests (02 is pure guess work).

I should add that I am new to Python and am running under Win10.
Thanks in advance for any pointers.

Brian

`======== RESTART: C:\Users\XXXX\Desktop\BitBucketConverter181002.py ========
Enter B1 line: 16:02:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 0A82 04F6 28D2 010002000200020002000202000002000200020002000202000002020002000002020002000002000200020002000200020200000200020200020000020200000203 55"}}

Creating B0...

Creating A8...
Enter SYNC_HIGH bucket number: 02
Enter SYNC_LOW bucket number: 02
Enter BIT_HIGH_TIME bucket number: 02

A8 command: 'AA A8 10 7F 04F6 04F6 04F6 32 04F6 32 20 00000000 55

B0 command: 'AAB04C041401360A8204F628D201000200020002000200020200000200020002000200020200000202000200000202000200000200020002000200020002020000020002020002000002020000020355'

>>> `

Portisch, 2019-03-27:
Upgrade firmware first!

tadder, 2019-03-27:
Thank you. I only flashed this about a week ago is there newer firmware available? Should I update both Tasmota and Portisch or just Portisch?

tadder, 2019-03-27:
I have found 'RF-Bridge-EFM8BB1.hex' dated 26th March on your github, so will install this instead of the copy that came with Tasmota. However 6.5.0 does appear to be the latest version of Tasmota so I assume I am OK to stick with that?

tadder, 2019-03-29:
I have updated Portich to ‘Keil 8051 v9.53’ and saved the output to ‘HexDump.txt’. However, I wonder if I am saving this output in the correct format as the program aborts after parsing the first line. I have tried formatting with no word-wrap just in case this was placing a hidden character at the line end to no avail.
Also, if I run with no reference to the HexDump file the program tries to download file ‘oscilloscope.png’ and then aborts. How is this part of the code supposed to work?
I think I may be very close to finally getting this working, but help with this matter would be very much appreciated.

brian@debian:~/Programs$ python BitBucketConverter.py -f HexDump.txt -e 192.168.1.xx
###### Processing line 18:16:19 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0500 04D8 280818 55"}}
 ######
Filtered 0xB1 data: AAB1030122050004D828081855
http://192.168.1.76/ax?c2=191&c1=RfRaw%20AA%20B0%200B%2003%2008%200122%200500%2004D8%20280818%2055
Sending command to bridge
Resulting 0xB0 data: AA B0 0B 03 08 0122 0500 04D8 280818 55
###### Command was sent. Enter name of command if your device reacted as expected or just [Enter] if not ######
###### Please no special caracters or spaces

Traceback (most recent call last):
  File "BitBucketConverter.py", line 343, in <module>
    parse_file(options.file)
  File "BitBucketConverter.py", line 294, in parse_file
    strInput = input().replace(" ","")
  File "<string>", line 0

    ^
SyntaxError: unexpected EOF while parsing 

Thank you

tadder, 2019-03-31:
I have noticed today that using the interactive tool without specifying the IP of the RF Bridge i.e.-
BitBucketConverter.py -f HexDump.txt
Does in fact produce an output to the screen.

However, attempting to send the output to the RF Bridge gives the same result i.e.-
BitBucketConverter.py -f HexDumpON.txt -e 192.168.1.76
###### Skipping line '17:25:40 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
' ######
###### Processing line 17:25:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0500 0500 281808 55"}}
 ######
Filtered 0xB1 data: AAB10301220500050028180855
http://192.168.1.76/ax?c2=191&c1=RfRaw%20AA%20B0%200B%2003%2008%200122%200500%200500%20281808%2055
Sending command to bridge
Resulting 0xB0 data: AA B0 0B 03 08 0122 0500 0500 281808 55
###### Command was sent. Enter name of command if your device reacted as expected or just [Enter] if not ######
###### Please no special caracters or spaces

Traceback (most recent call last):
  File "BitBucketConverter.py", line 343, in <module>
    parse_file(options.file)
  File "BitBucketConverter.py", line 294, in parse_file
    strInput = input().replace(" ","")
  File "<string>", line 0

    ^
SyntaxError: unexpected EOF while parsing

Is there a pointer in these results that gives a clue to my issue?
As before, I would be most grateful if someone has anything that might help please.

Portisch, 2019-04-01:
The option to parse a file in the script is not coded by me. I don't know if this is working or not. Start the script on a terminal window and enter a single B1 line from your tasmota console.

tadder, 2019-04-01:
thank you for your reply, I didn't realise that you had not written the parsing part of the script. I will try what you suggest.

marcelhen, 2020-11-05:
Hello,

I have a Sonoff Bridge 433 MHz (Tasmota / Portisch) running via MQTT on the HA. I want to use it to switch my shutters. If I set option RfRaw 166 I get the following code:

tele / tasmota_C3B1F5 / RESULT = {“Time”: “2020-11-02T09: 15: 30”, “RfRaw”: {“Data”: “AAA606012CFF86713355”}}

How can I send that now? If I copy it into the command line, nothing happens.

If I set the RfRaw 177 option, I unfortunately don’t receive anything …

I have already received several other codes with the RfRaw 177 option and have also converted them, so they can be switched with Home Assistant without any problems.

I currently have the following command:

**cover:

- platform: "mqtt"

name: "Rolladen HWR"

command_topic: "cmnd/tasmota_C3B1F5/Backlog"

payload_open: "RfRaw AAA420F80110033001232355; RfRaw 0"

payload_close: "RfRaw AAA606012CFF86713355; RfRaw 0"

payload_stop: "RfRaw AAA606012CFF86715555; RfRaw 0"**

But unfortunately it doesn’t work !!!

Best regards 
Marcel

Mindsuk, 2021-04-12:
@Portisch can you please assist me with my smartwares sh5-tsw-A switch

Elbandi, 2021-06-10:
> Your 0xA8 command to stop will be: B8419AC155 (stop)
> **AAA80602B8419AC15555**
> 
> And this is the problem: ...5555
> The ESP32 firmware will not check the len of the command and stops at the first 0x55 (sync end) received. What firmware your are using on the ESP32?
Is this already fixed? my stop command is AAA80601 5645FA8055 55, and there are two 55 at the end. and stop is not working from rfbridge :(