Portisch/RF-Bridge-EFM8BB1 #24: Understanding why one RF device not usable after EFM8BB1 upgrade ?

mr-sneezy, 2018-08-07:
I'm not sure this is an issue. I've looked around here and forums but my trouble seems unique.
It boils down to not being able to receive or mimic the RF codes from my door bell, that I could BEFORE I updated. I had it working well for months into Home Assistant to send me TXT's when I got a delivery.

I don't have a copy of the code that the door bell produced before update, so all I can do now is try rfraw 166 & rfraw 177 commands. 166 does not produce a result (I guess meaning that it's not understood) and 177 gives me a slightly different string each time I push the bell button now.

I'm wondering, is there a timing difference in the way the Sonoff original EFM8BB1 captured data and the Portisch code ?
If so then perhaps the door bell data speed or symbol length is now out of range ?
I have console data, but like I say, it's not consistent now. 
Cheers,
Martin

Portisch, 2018-08-13:
Why 0xA6?

> With the command the standard PT226x sniffing gets disabled and the sniffing of the defined protocols in RF_Protocols.h is starting.

If it was working with the original firmware just use the original commands for sniffing (0xA4), learning (0xA1) and transmit (0xA5).

mr-sneezy, 2018-08-13:
I have now l spent more time deciphering your readme more fully so I think I know you are saying about RfRaw codes (other than 166 and 177). It's not easy reading material mate, I have trouble following it.
 
 Unfortunately it still doesn't work with RfRaw 0xA1, or 0xA4. I get nothing received by the bridge from this device now (but do with others). 
I have bought a second door bell unit like the first for testing, just in case the first was somehow faulty now (even though still activating the actual wireless bell fine). 
No change with the new one, nothing received with that either. So it's still like the original firmware code is not the same as it used to be, I can't learn the device... 

Using the console and old learning and pressing remote a few times.
```
0:09:30 HTP: Console
10:09:46 CMD: rfkey 2 1
10:09:46 SRC: WebConsole from 192.168.0.19
10:09:46 RSL: Group 0, Index 1, Command RFKEY, Data 2 1
10:09:46 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Start learning"}
10:09:46 SER: Received A0 55
10:10:17 SER: Received A2 55
10:10:17 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Learn failed"}
```
With RfRaw 0xA4 I get only this after setting and button press on remote.
```
10:41:21 CMD: RfRaw 0xA4
10:41:21 SRC: WebConsole from 192.168.0.19
10:41:21 RSL: Group 0, Index 1, Command RFRAW, Data 0xA4
10:41:22 MQT: stat/sonoff_bridge/RESULT = {"RfRaw":"OFF"}
10:41:22 SER: Received A0 55
```
Perhaps I'm still not understanding the way I'm supposed to do this stuff ?

Portisch, 2018-08-13:
Please flash the original firmware for the RF Bridge and test and record the data. Maybe on this data we can see why it isn't working.

On the other side the command 0xB1 should sniff any timing/protocol!?

mr-sneezy, 2018-08-15:
This is what I get from rfraw 0xB1 and pressing the doorbell TX once for about 200mS.

`10:08:06 HTP: Console
10:08:32 CMD: rfraw 177
10:08:32 SRC: WebConsole from 192.168.10.91
10:08:32 RSL: Group 0, Index 1, Command RFRAW, Data 177
10:08:33 MQT: stat/sonoff_bridge/RESULT = {"RfRaw":"ON"}
10:08:33 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
10:08:38 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 00B4 01E0 00D9 006E 13F6 01011010101031101010311031313110013101010110011034 55"}}
10:08:38 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 009D 01EA 13CE 01011010101001101010011001010110010101010110011002 55"}}
10:08:38 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A1 01EA 13C4 01011010101001101010011001010110010101010110011002 55"}}
10:08:38 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 01EA 13C4 01011010101001101010011001010110010101010110011002 55"}}`
I will not have time and means to reflash the bridge and do more testing till I catch up on some high priority stuff...

Portisch, 2018-08-16:
As sniffed with command 0xB1 I can see the sync time is about 5000ms. The default minimum sync time in the source was 6000ms. So try the new master: https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/d112fbb6aba3c6e8d9a653a083b79c00780c29fe

mr-sneezy, 2018-08-16:
OK, will do thanks. 
(5000mS = 5sec, typo for 5000uS ??)

mr-sneezy, 2018-08-16:
This result speaks volumes...  

14:31:11 HTP: Console
14:31:14 SER: Received A4 13 B0 00 AA 01 E0 3D D1 05 55
14:31:15 MQT: tele/sonoff_bridge/RESULT = {"RfReceived":{"Sync":5040,"Low":170,"High":480,"Data":"3DD105","RfKey":"None"}}
14:31:54 CMD: rfkey 2 1
14:31:54 SRC: WebConsole from 192.168.0.19
14:31:54 RSL: Group 0, Index 1, Command RFKEY, Data 2 1
14:31:54 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Start learning"}
14:31:54 SER: Received A0 55
14:31:57 SER: Received A3 13 B0 00 A0 01 E0 3D D1 05 55
14:31:57 MQT: stat/sonoff_bridge/RESULT = {"RfKey1":"Learned"}
14:31:57 CFG: Saved to flash at F9, Count 106, Bytes 3584

and then this from Home Assistant....

[image]

Well done.

closed 2018-08-16 by Portisch