Portisch/RF-Bridge-EFM8BB1 #25: Help decoding SZ Funpower RF 433 DC Fan Controller Remote

aenertia, 2018-08-19:
Hi there; so I have a successful flashed chip and am trying to figure out my DC Fan controller. Bucket Sniffing results in the following.

The Remote system must 'pair' with the base unit (I do have a second unit which I haven't attempted to setup yet), the one I am trying to sniff however is mounted high and there is little possibility of getting up there at this point.

Here is the dump's

#on/off/1/2/3/4/5/reverse/1/on/off
23:48:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0380 012A 163A 01010110101010100110100110011010011010101010010110 55"}}
23:48:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0384 012D 164E 01010110101010100110100110011010011010101010010110 55"}}
23:48:32 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0384 012E 164E 01010110101010100110100110011010011010101010010110 55"}}
23:48:38 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0382 0126 164E 01010110101010100110100110011010101010101001100110 55"}}
23:48:39 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0381 012D 163A 01010110101010100110100110011010011010101001100110 55"}}
23:48:40 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 038D 011E 164E 01010110101010100110100110011010101010100110011010 55"}}
23:48:41 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0388 012A 1630 01010110101010100110100110011010011010100110011010 55"}}
23:48:43 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 038F 011E 164E 01010110101010100110100110011010101010101001011010 55"}}
23:48:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0381 0122 164E 01010110101010100110100110011010101010100110100110 55"}}
23:48:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 038B 0126 1644 01010110101010100110100110011010101010101010011010 55"}}
23:48:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0397 0116 164E 01010110101010100110100110011010101010100101101010 55"}}
23:48:54 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 038C 0128 1630 01010110101010100110100110011010011010100101101010 55"}}
23:48:58 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0382 0128 163A 01010110101010100110100110011010011010101001100110 55"}}
23:49:00 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0381 0129 163A 01010110101010100110100110011010011010101010010110 55"}}
23:49:11 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037C 012C 1644 01010110101010100110100110011010101010101010010110 55"}}
23:49:11 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0380 012A 163A 01010110101010100110100110011010011010101010010110 55"}}
##on/off bucket sniff
00:03:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037C 0127 164E 01010110101010100110100110011010101010101010010110 55"}}
00:03:32 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0131 038E 016E 132E 10101001010101011001011001100101100101010101101000 55"}}

aenertia, 2018-08-19:
Replaying any of the data raw doesn't result in the hoped for beep from the base unit.

Portisch, 2018-08-20:
Example:
`23:48:31 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0380 012A 163A 01010110101010100110100110011010011010101010010110 55"}}`

The last bit is sometimes wrong because of a bug in the firmware. It have to be every time the index number of the last bucket (sync bucket). So here are 3 buckets and the data should be:
0101011010101010011010011001101001101010101001011**2** instead of
01010110101010100110100110011010011010101010010110

Then you choose your number of repeats: a number of 20 (0x14) should be enough.

Assemble the 0xB1 data for the 0xB0 comand:
AA B0 xx 03 14 0380 012A 163A 01010110101010100110100110011010011010101010010112 55

Determine the length "xx":
`03 14 0380 012A 163A 01010110101010100110100110011010011010101010010112` == 0x21 bytes

0xB0 complete:
AA B0 21 03 14 0380 012A 163A 01010110101010100110100110011010011010101010010112 55

The other way would be to decode the 24 bit of data for each command like:
010101101010101001101001100110100110101010100101 | 12 (sync)	data
1 1 1 0 0 0 0 0 1 0 0 1 0 1 0 0 1 0 0 0 0 0 1 1					binary
E09483															hex

As the bucket timing is 75 & 25% duty cycle you can use also comand 0xA5 or comand 0xA8. 0xA5 is the simple one:
AA A5 163A 012A 0380 E09483 55

aenertia, 2018-08-20:
Am still struggling ; tried with your strings but no dice.

Here is some more dump of On / Off. Note that there seems to be an answer that comes back from the base unit (the second entry) after I press the on button on the remote (have not pressed it for some time). Is this part of the keying?

08:26:20 CMD: RfRaw 177
08:26:20 RSL: RESULT = {"RfRaw":"ON"}
08:26:20 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
08:26:24 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037F 012F 1644 01010110101010100110100110011010101010101010010110 55"}}
08:26:25 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0385 0129 163A 01010110101010100110100110011010011010101010010110 55"}}
08:26:27 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0382 012E 1644 01010110101010100110100110011010101010101010010110 55"}}
08:26:27 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0389 012B 1630 01010110101010100110100110011010011010101010010110 55"}}
08:26:34 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037F 012F 1644 01010110101010100110100110011010101010101010010110 55"}}
08:26:55 CMD: RfRaw 176
08:26:55 RSL: RESULT = {"RfRaw":"ON"}
08:27:30 CMD: RfRaw  AA B0 21 03 14 0380 012A 163A 01010110101010100110100110011010011010101010010112 55
08:27:30 RSL: RESULT = {"RfRaw":"ON"}

aenertia, 2018-08-20:
and a bunch more repeated on/off

08:38:32 CMD: RfRaw 177
08:38:32 RSL: RESULT = {"RfRaw":"ON"}
08:38:32 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
08:38:34 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037F 012F 1644 01010110101010100110100110011010101010101010010110 55"}}
08:38:40 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 037F 012F 1644 01010110101010100110100110011010101010101010010110 55"}}
08:38:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0384 012E 1644 01010110101010100110100110011010101010101010010110 55"}}
08:38:52 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0387 012B 1644 01010110101010100110100110011010101010101010010110 55"}}
08:38:58 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0385 0129 164E 01010110101010100110100110011010101010101010010110 55"}}
08:39:02 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0387 0129 164E 01010110101010100110100110011010101010101010010110 55"}}
08:39:19 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0380 012A 1644 01010110101010100110100110011010101010101010010110 55"}}
08:39:23 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0385 0129 164E 01010110101010100110100110011010101010101010010110 55"}}
08:39:27 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0385 012C 1644 01010110101010100110100110011010101010101010010110 55"}}
08:39:34 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0385 0129 164E 01010110101010100110100110011010101010101010010110 55"}}
08:39:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0381 0129 1644 01010110101010100110100110011010101010101010010110 55"}}
08:39:57 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0386 0128 1644 01010110101010100110100110011010101010101010010110 55"}}

gerardovf, 2018-09-10:
@aenertia try this:

$ python ./BitBucketConverter.py "AA B1 03 0381 0129 1644 01010110101010100110100110011010101010101010010110 55" 20
'**_RfRaw AAB02103140381012916440101011010101010011010011001101010101010101001011055_**'

[raw-sniffing-procedure](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#raw-sniffing-procedure)

[B1 to B0 tool](https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/31)

aenertia, 2018-09-11:
Thanks will give it a shot when I get back home from my overseas work trip.!

Portisch, 2018-09-11:
As I have written here: https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/25#issuecomment-414214476 it looks like old firmware or a bug. The sync bucket is missing...

> @aenertia try this:
> 
> $ python ./BitBucketConverter.py "AA B1 03 0381 0129 1644 01010110101010100110100110011010101010101010010110 55" 20
> 'RfRaw AAB02103140381012916440101011010101010011010011001101010101010101001011055'

This should not work because bucket 2 is missing!

This maybe work:
AAB02103140381012916440101011010101010011010011001101010101010101001011 _2_ 55

closed 2018-10-24 by Portisch