Portisch/RF-Bridge-EFM8BB1 #26: strange signal

tartesso26, 2018-08-24:
i have a remote control with 3 buttons, i´m triying to use it with rf-brigde, but when i try to read buttons using RfRaw 166 i get this

first button = {"RfRaw":{"Data":"AAA60602390801011155"}}
                   = {"RfRaw":{"Data":"AAA60602390801011E55"}}
second button = {"RfRaw":{"Data":"AAA606023908010155"}}
third button = {"RfRaw":{"Data":"AAA60602390801013C55"}}

so, i think i have to send 
rfraw AAA80602390801011155    for first button
rfraw AAA806023908010155    for second button
rfraw AAA80602390801013C55 for third button

but only first button is working

if try reading with B1 then i got

for first button, i try 4 reads and get this results
 {"RfRaw":{"Data":"AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55"}}
{"RfRaw":{"Data":"AA B1 07 011B 02C7 12FC 01CC 008C 05E6 1FA4 25010110101001011001010101100101010101010101010110010101010101011001010110010101102501011010100101100101010110010101010101010101011001010101010101100101011001010116 55"}}
 {"RfRaw":{"Data":"AA B1 04 05E6 016C 02D1 12CA 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55"}}
 {"RfRaw":{"Data":"AA B1 04 0596 02D3 0165 132E 021211212122121122121212112212121212121212121211221212121212121122112211221122112030 55"}}

for second button try multiples reads, and most don´t work, only once, and get 
{"RfRaw":{"Data":"AA B1 06 0550 02D4 016C 01B4 012C 1210 0212112121221211221212121122121212121212121212112212121212121211221122112211221125 55"}}

for third button, try 4 reads and get
{"RfRaw":{"Data":"AA B1 04 05F0 016B 02CF 12CA 0121221212112122112121212211212121212121212121221121212121212122112122121121221213 55"}}
{"RfRaw":{"Data":"AA B1 06 1289 062C 0161 02CA 01B6 1EA0 01232332323223233223232323322323232323232323232332232323232323233323322323323232322325 55"}}
{"RfRaw":{"Data":"AA B1 04 05F0 016D 02D0 12CA 0121221212112122112121212211212121212121212121221121212121212122112122121121221213 55"}}
 {"RfRaw":{"Data":"AA B1 06 1284 062C 0164 02C9 01B8 1EAA 01232332323223233223232323322323232323232323232332232323232323233323322323323232322325 55"}}

how can i do to get rf-brigde works?? any idea

thanks, your work is fantastic

Portisch, 2018-08-27:
It detected protocol 0x02. I don't know if this is correct. The second command is getting sniffed right but the ESP firmware do have a bug: https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23#issuecomment-411288705

Your data will be:
first button: 3908010111
second button: 3908010133
third button: 3908010155

**You can use command 0xA8 also as universal transmit like this:**
AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55
AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010133 55
AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010155 55

**Or as protocol 0x02: (timing is a little bit different)**
AA A8 06 02 3908010111 55 (this button was already working)
AA A8 06 02 3908010133 55 (in your sniffing it was 0x3C not 0x33)
AA A8 06 02 3908010155 55 (was sniffed correct but defekt by the ESP firmware)

tartesso26, 2018-09-16:
thanks a lot, i´ll try using all your comments, your work is great

tartesso26, 2018-09-16:
do exits an update for firmware fixing these problems ???