Portisch/RF-Bridge-EFM8BB1 #27: Sniffing Heidemann HX Gong protokoll

phototobi, 2018-08-29:
Hi,
i thought it could be possible to sniff door bell info from Heidemann Hx tx using sonoff rfbridge with espurna. I see a lot of traffic on rfin but not when i trigger the door bell. Now i like to understand if it could be a solution to hack the hardware and use your great firmware. I only like to see if someone rings the door bell. Do you think it is possible that way?

Thanks in advance
Tobias

Jason2866, 2018-09-03:
Hi @phototobi,
yes that could work. You can easily flash this firmware with Tasmota.
Tasmota is a custom firmware like Espurna. Tasmota has the great benefit that the flasher
for the EFM chip is included. No extern device needed! Simply two wires to connect.
See https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433

Portisch, 2018-09-17:
For sniffing unknown protocols use bucket sniffing 0xB1

closed 2018-09-17 by Portisch