Portisch/RF-Bridge-EFM8BB1 #29: Sniffing Dooya DC90 remote control

jk2020, 2018-09-03:
Looks like Sniffing by command 0xA6 does not work for DC90 remote control

So used Bucket sniffing using command 0xB1 and calculated the command by means of the example of Bucket Transmitting using command 0xB0

Actually got now following commands for UP/DOWN which work instead of my DC90 remote control. My successful commands are:
DOWN
`AAB035051A128E061802D1014E1E50012332232323322332323232322332322332232323232323233223322332323223323223232323323455`
UP
`AAB0370612128E062202AC016001221F9A012332232323322332323232322332322332232323232323233223322332323223323232232323233555`

However I face problems finding the command for STOP.

Analyzed dozens of commands and put together new command string - see below - but no success yet:

```
AAB0330419058C02CB016E1324012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330414059602CA016E131A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330414059602CD016B131A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412059602C701721374012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412059602CC016C131A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058202CA016D1324012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02C90170131A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412059602CA01721310012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412059602CD016C136A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0350519059602CC017101B8131A012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512059602CE016D01CC1522012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512057802CA017001B8132E012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202CD016B01B8132E012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02CB016D01AE1392012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202CA016F01B81392012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02C9017301AE1338012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202CA016901AE1338012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202CA016F01B8132E012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02D0016601B81338012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202C9016F01B814B4012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02CF016801B813BA012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058202C9016F01B8132E012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512057802CF016801B81342012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02C9016D01AE1324012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512059602C9016E01B81338012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02C9016E01AE1310012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0350512058C02CC016D01B8131A012211212122112212121212112212112211212121212121221122112212121122112211221122112455
AAB0370612058202CA016E01B801221374012211212122112212121212112212112211212121212121221122112212121122112211221122112555
AAB0370612058C02CD016F01C2012C1324012211212122112212121212112212112211212121212121221122112212121122112211221122112555
AAB0370612056E02C6017301BD012714F0012411212124112414141414112412112211212121212121241122112212121122112211221122112555
AAB0330412057802C9016E132E012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02C9016E1342012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02C7016D131A012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02CD016B1310012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058202CA016C146E012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412059602C80173137E012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02C701721324012211212122112212121212112212112211212121212121221122112212121122112211221122112355
AAB0330412058C02C90171132401221121212211221212121211221211221121212121212122112211221212112211221
```

Any hints or ideas what I need to do to get the command for my stop button?

Portisch, 2018-09-04:
Please try the new DC90 branch. I implemented a new DC90 protocol.
Maybe you can now sniff the commands by command 0xA6.
If this is working you can transmit the sniffed data with command 0xA8, protocol 0x07.

It should result in data like this (decoded from you 0xB0 sniffed data):
Up: BA097F5111
Down: BA097F513C
Stop: ??

jk2020, 2018-09-04:
I can confirm that the UP/DOWN data you had decoded from 0xB0 sniffed data works with
```
RfRaw AAA80602BA097F513C55
RfRaw AAA80602BA097F511155
```
Have dowloaded the zip from DC90 branch and have flashed with hex file RF-Bridge-EFM8BB1.hex you have created.

Sonoff RF Bridge is up and running
`10:02:35 CMD: status 0
10:02:35 SRC: WebConsole from 192.168.2.3
10:02:35 RSL: Group 0, Index 1, Command STATUS, Data 0
10:02:35 MQT: stat/sonoff-bridge/STATUS = {"Status":{"Module":25,"FriendlyName":["sonoff-bridge"],"Topic":"sonoff-bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:02:35 MQT: stat/sonoff-bridge/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:43","StartupUTC":"2018-09-04T09:01:52","Sleep":0,"BootCount":169,"SaveCount":225,"SaveAddress":"F9000"}}
10:02:35 MQT: stat/sonoff-bridge/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-08-31T15:15:32","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:02:35 MQT: stat/sonoff-bridge/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Free VirusFI",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
10:02:35 MQT: stat/sonoff-bridge/STATUS4 = {"StatusMEM":{"ProgramSize":472,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000000","23B617CE","00000000"]}}
10:02:35 MQT: stat/sonoff-bridge/STATUS5 = {"StatusNET":{"Hostname":"sonoff-bridge-3976","IPAddress":"192.168.2.62","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:A3:6F:88","Webserver":2,"WifiConfig":3}}
10:02:35 MQT: stat/sonoff-bridge/STATUS6 = {"StatusMQT":{"MqttHost":"nodered.jfr.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A36F88","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:02:35 MQT: stat/sonoff-bridge/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 04 09:02:35 2018","Local":"Tue Sep 04 10:02:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:11","Sunset":"19:27"}}
10:02:35 MQT: stat/sonoff-bridge/STATUS10 = {"StatusSNS":{"Time":"2018-09-04T10:02:35"}}
10:02:35 MQT: stat/sonoff-bridge/STATUS11 = {"StatusSTS":{"Time":"2018-09-04T10:02:35","Uptime":"0T00:00:43","Vcc":3.170,"Wifi":{"AP":1,"SSId":"Free VirusFI","RSSI":64,"APMac":"80:2A:A8:CA:7E:9B"}}}`

executing sniff via 0xA6
`10:05:31 CMD: rfraw aaa655
10:05:31 SRC: WebConsole from 192.168.2.3
10:05:31 RSL: Group 0, Index 1, Command RFRAW, Data aaa655
10:05:31 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
10:05:31 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}`

Red LED is blinking when pressing DC90 remote but however now commands displayed

Portisch, 2018-09-04:
Ok, you used protocol 0x02 (ROHRMOTOR24). And yes, it looks like the same. The sync of the two remotes is the same. The bit timing is a little bit different. So I tried to increase the tolerance. Please try again with the DC90 branch if now with 0xA6 the data can be sniffed.

jk2020, 2018-09-04:
Yes, used 0x02 because it did not work with 0x07

Downloaded updated version and flashed to sonoff rf bridge
```

10:56:26 HTP: Console
10:56:30 MQT: tele/sonoff-bridge/STATE = {"Time":"2018-09-04T10:56:30","Uptime":"0T00:00:13","Vcc":3.170,"Wifi":{"AP":1,"SSId":"Free VirusFI","RSSI":62,"APMac":"80:2A:A8:CA:7E:9B"}}
10:56:32 CMD: rfraw aaff55
10:56:32 SRC: WebConsole from 192.168.2.3
10:56:32 RSL: Group 0, Index 1, Command RFRAW, Data aaff55
10:56:32 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
10:56:32 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA0055"}}
10:56:41 CMD: rfraw aaa655
10:56:41 SRC: WebConsole from 192.168.2.3
10:56:41 RSL: Group 0, Index 1, Command RFRAW, Data aaa655
10:56:41 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
10:56:41 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
```
no change, e.g. sniffing of DC90 not possible with 0xA6

jk2020, 2018-09-04:
PS: after reloading flash and reboot of bridge I get following, if i press UP/DOWN button:
```
11:06:09 HTP: Console
11:06:13 CMD: rfraw aaff55
11:06:13 SRC: WebConsole from 192.168.2.3
11:06:13 RSL: Group 0, Index 1, Command RFRAW, Data aaff55
11:06:13 MQT: stat/sonoff-bridge/RESULT = {"RfRaw":"ON"}
11:06:13 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AA0055"}}
11:06:14 MQT: tele/sonoff-bridge/STATE = {"Time":"2018-09-04T11:06:13","Uptime":"0T00:00:13","Vcc":3.170,"Wifi":{"AP":1,"SSId":"Free VirusFI","RSSI":66,"APMac":"80:2A:A8:CA:7E:9B"}}
11:06:17 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F7C014002A8DD04BF55"}}
11:06:24 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F7C014002B2DD04BF55"}}
11:07:00 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F86014002A8DD04BF55"}}
11:07:03 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F86014A02A8DD04BF55"}}
11:07:18 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F86014002A8DD04BF55"}}
11:07:21 MQT: tele/sonoff-bridge/RESULT = {"RfRaw":{"Data":"AAA41F7C014002B2DD04BF55"}}
```
however nothing for STOP button

jk2020, 2018-09-04:
after another reboot followin appears on consolue when pressing UP/DOWN buttongs
```
11:10:15 HTP: Main Menu
11:10:17 HTP: Console
11:10:20 MQT: tele/sonoff-bridge/STATE = {"Time":"2018-09-04T11:10:20","Uptime":"0T00:00:13","Vcc":3.170,"Wifi":{"AP":1,"SSId":"Free VirusFI","RSSI":60,"APMac":"80:2A:A8:CA:7E:9B"}}
11:10:25 SER: Received A4 1F 90 01 36 02 9E DD 04 BF 55
11:10:25 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":8080,"Low":310,"High":670,"Data":"DD04BF","RfKey":"None"}}
11:10:35 SER: Received A4 1F 86 01 40 02 B2 DD 04 BF 55
11:10:35 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":8070,"Low":320,"High":690,"Data":"DD04BF","RfKey":"None"}}
11:10:46 SER: Received A4 1F 86 01 40 02 9E DD 04 BF 55
11:10:46 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":8070,"Low":320,"High":670,"Data":"DD04BF","RfKey":"None"}}
11:11:02 SER: Received A4 1F 86 01 36 02 A8 DD 04 BF 55
11:11:02 MQT: tele/sonoff-bridge/RESULT = {"RfReceived":{"Sync":8070,"Low":310,"High":680,"Data":"DD04BF","RfKey":"None"}}
```

have not entered any RfRaw command

Portisch, 2018-09-04:
The original sniffing will not work... I have done some research about the DC90 remote. It is from the same company like the Rohrmotor24. So the protocol is the same - just the timing is looking different.
Please do mor 0xB1 bucket sniffing. Like a minimum of 10 times for each button. But press the button on the remote short only - not permanent. Or do also a sniffing with the button pressed permanent.

jk2020, 2018-09-04:
Finally that one worked for STOP button

```
AA B1 04 05A0 02C6 016D 1310 
0 12 21 1212-12 21 12 21-21212121- 12 2121 12- 21 121212-12121212- 21 12 21 12- 212121 12- 21 12 21 12- 21 12 21 12 3 55
1011 1010 0000 1001 0111 1111 0101 0001 0101 0101
B    A    0    9    7    F     5   1    5    5
RfRaw AAA80602BA097F515555
```
Many thanks for your help.
Also your tutorials are great work!

closed 2018-09-04 by jk2020