Portisch/RF-Bridge-EFM8BB1 #30: EV1527 Remote Issue

peteakalad, 2018-09-05:
I have a remote which contains a EV1527 chip controlling a blind controller (blind genie).

I am using a sonoff bridge with 6.2 tasmota, and this would learn the codes of the remote and correctly activate the blind controller with the stock RF firmware, but unfortunately it is not reliable as the signal sometimes needs to be sent multiple times.  I can't just always send the code multiple times as a 2nd code will stop the blind opening if the 1st code was actually received ok.

If I hold the remote button down for a second or two it always works.  

I've applied this RF-Bridge-EFM8BB1 firmware to the RF chip in the hope that I could make the RF bridge repeatly send the same data in a single tranmission like holding the remote button down.

So far this hasn't worked - once this RF firmware is applied (tried the hex here and in the tasmota repository) the RF bridge can learn the code, and does so with the same parameters as the stock firmware but when transmitting it no longer works.  I suspect the timings are just not quite right for my receiver of the EV1527 signal.

I could try and get a rtl sdr capture of the EV1527 remote signal if it helps.

Is there a command to cause the data to be repeated more often than it is by default?  And is it the same as the stock or different for the 0xA5 transmit command?

Thanks!

Portisch, 2018-09-06:
I don't know if the repeat is the same as in the original firmware. And it is hardcoded in the firmware.

RF-Bridge-EFM8BB1/inc/RF_Protocols.h

`#define RF_TRANSMIT_REPEATS		8`

Feel free to increase the repeating and report!

peteakalad, 2018-09-06:
Portisch thanks for the pointer - will see if I can get the IDE setup and increase that repeat number significantly see if it makes a difference.

Failing that will try and do some digging and compare the signals transmitted stock vs custom FW via rtlsdr captures.

Will report back, cheers!

peteakalad, 2018-09-06:
No luck with a longer repeat.

Also compared the signals between stock and custom fw via rtlsdr captures, and high and low lengths seemed to be 10% ish too great with custom.  Did try reducing the lengths appropriately with no luck.  Guess it must be some other difference in the signal causing the receiver not to work.

Then tried extracting with 0xB0 and transmitting with converted rfraw command from the python script on the other issue, worked great!

So {"RfKey1":{"Sync":14060,"Low":480,"High":1380,"Data":"AC6108"}} with custom fw didn't work with my device, where it was working on stock.  Def transmitting LED lights and received by my rtlsdr.

Using bucket sniff of the remote;
22:04:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 054A 01F3 3714 01100110010110101001011010101001101010100110101012 55"}}
22:04:57 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 055D 01D5 0988 36D8 01100110010110101001011010101001101010100110101013 55"}}
22:04:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0543 01F4 36E2 01100110010110101001011010101001101010100110101012 55"}}
22:05:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0543 01F4 36E2 01100110010110101001011010101001101010100110101012 55"}}

Then this transmitted worked well...
RfRaw AAB0210314054301F436E20110011001011010100101101010100110101010011010101255

closed 2018-09-10 by Portisch