Portisch/RF-Bridge-EFM8BB1 #31: B1 to B0 help tool

gerardovf, 2018-09-05:
After learning how bit bucket works from here https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/23 I decided to prepare a python script to help calculate the right 'B0' message to send using 'RfRaw' command in Tasmota from the received 'B1' sniffing message (rename file from  'BitBucketConverter.txt' to 'BitBucketConverter.py'.

[BitBucketConverter.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2353034/BitBucketConverter.txt)

In the command line give the 'B1' message string and the retries value (in decimal):
i.e. **_BitBucketConverter.py "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20_**
>>> 
Command Line : "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20
'RfRaw AAB01C041407EB015700FD3EBC01010101010110100101010110101010010355'
>>>

peteakalad, 2018-09-06:
Great work!

I was stumped trying to get the B1 / B0 working till I tried this - got my device to work at last with this firmware!

Many thanks for this - could be something to add to the wiki.

gerardovf, 2018-09-07:
Thanks @peteakalad 

I've included it here: [Sonoff RF Bridge 433](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433)

gerardovf, 2018-09-09:
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#raw-sniffing-procedure

Portisch, 2018-09-10:
May you can include a "nibble" to readable hex number convert!?

Example by this data:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/26#issue-353911232

AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55

bucket 0: 05E6
bucket 1: 016C
bucket 2: 02D1
bucket 3: 12D4

A sync is normally first high than low.
In the sniffed data the last nibble should be always the snyc start (bucket 3):
012122121211212211212121221121212121212121212122112121212121212211212122112121221 _3_
After a high snyc pulse a low time is following. This nibble is the first in the data (bucket 0):
_0_ 121221212112122112121212211212121212121212121221121212121212122112121221121212213

So the real data is:
sync: 30, data: 12122121211212211212121221121212121212121212122112121212121212211212122112121221

each 12 or 21 is one bit. The string length of 12122121211212211212121221121212121212121212122112121212121212211212122112121221 is 80 chars. This 80 chars divided by 2 is the amount of bits: 40

Now the convert to hex numbers:
bucket 2 is the long pulse, bucket 1 the short.
The data is normally starting with a high pulse.

bucket series: 12
Means short high, long low: so I guest a logical 0

bucket series: 21
Means long high, short low: so I guest a logical 1

Converted data:

```
12122121211212211212121221121212121212121212122112121212121212211212122112121221
0 0 1 1 1 0 0 1 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0 1 0 0 0 0 0 0 0 1 0 0 0 1 0 0 0 1
```
Convert this binary data to hex:
0x3908010111

For sure this is only working for bit orientated protocols.
Like with your data it isn't working:
AA B1 04 07F0 0128 00F2 3822 010101010101101001010101101010100103 55

Sync: 30, data: 1010101010110100101010110101010010
bit 0: 10
bit 1: 01
```
1010101010110100101010110101010010
0 0 0 0 0 ? 1 ? 0 0 0 ? 1 1 1 ? 0
```
11 or 00 isn't defined...

gerardovf, 2018-09-10:
All right!
Let's start with your proposal:

Command Line : "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20

Bucket 0: 07EB (2027)

Bucket 1: 0157 (343)

Bucket 2: 00FD (253)

Bucket 3: 3EBC (16060)

**0 01 01 01 01 01 01 10 10 01 01 01 01 10 10 10 10 01  3**

'RfRaw AAB00B041407EB015700FD3EBC0155'
 
Command Line : "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55" 20

Bucket 0: 05E6 (1510)

Bucket 1: 016C (364)

Bucket 2: 02D1 (721)

Bucket 3: 12D4 (4820)

**0 01 21 22 12 12 11 21 22 11 21 21 21 22 11 21 21 21 21 21 21 21 21 21 22 11 21 21 21 21 21 21 22 11 21 21 22 11 21 21 22  3**

'RfRaw AAB00B041405E6016C02D112D42255'

Portisch, 2018-09-10:
Not correct, the first nibble is the second part of the sync - not data.
The sniffing is searching for the longest bucket. After the longest bucket ocurre again the data recording is starting.

So not:
**0 01 01 01 01 01 01 10 10 01 01 01 01 10 10 10 10 01 3**

It is:
**30 10 10 10 10 10 11 01 00 10 10 10 11 01 01 01 00 10**

That are 4 "bits": 01, 10, 11 and 00
May here is the answer: [A bit of theory](http://tinkerman.cat/decoding-433mhz-rf-data-from-wireless-switches-the-data/)

The reason why I am asking is because if you have the data in hex you can use the command [0xA8.](https://github.com/Portisch/RF-Bridge-EFM8BB1#transmiting-by-command-0xa8)
Also this data can be used to define new protocols in the [RF_Protocols.h](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/inc/RF_Protocols.h)

AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55

```
30   12 12 21 21 21 12 12 21 12 12 12 12 21 12 12 12 12 12 12 12 12 12 12 21 12 12 12 12 12 12 12 21 12 12 12 21 12 12 12 21
sync 0  0  1  1  1  0  0  1  0  0  0  0  1  0  0  0  0  0  0  0  0  0  0  1  0  0  0  0  0  0  0  1  0  0  0  1  0  0  0  1
hex 3908010111
```

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 12D4
SYNC_LOW: bucket 0: 05E6
BIT_HIGH_TIME: bucket 2: 02D1
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x42 (66%)
BIT_LOW_TIME: bucket 1: 016C
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x21 (33%)
BIT_COUNT + SYNC_BIT_COUNT: 0x28 (40, SYNC_BIT_COUNT = 0)
Data: 3908010111

`AA A8 xx 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55`
Protocol is 0x7F (unknown), Len xx is counting bytes: 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 == 0x11

`AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55`

Maybe a printout of the statistic of the sniffed data would also be usefull to check if this protocol is already defined. Like in this example it is protocol 0x02 (ROHRMOTOR24).

gerardovf, 2018-09-10:
Quick and dirty. I miss how to get the 'SYNC_BIT_COUNT' value (I used '0')

```
Command Line : "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55" 20

'RfRaw AAB033041405E6016C02D112D4012122121211212211212121221121212121212121212122112121212121212211212122112121221355'

Sync: 30    Data: 12 12 21 21 21 12 12 21 12 12 12 12 21 12 12 12 12 12 12 12 12 12 12 21 12 12 12 12 12 12 12 21 12 12 12 21 12 12 12 21 
Sync               0  0  1  1  1  0  0  1  0  0  0  0  1  0  0  0  0  0  0  0  0  0  0  1  0  0  0  0  0  0  0  1  0  0  0  1  0  0  0  1  
Hex: 3908010111

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 12D4
SYNC_LOW: bucket 0: 05E6
BIT_HIGH_TIME: bucket 2: 02D1
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x42 (66%)
BIT_LOW_TIME: bucket 1: 016C
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x21 (33%)
BIT_COUNT + SYNC_BIT_COUNT: 0x28 (40, SYNC_BIT_COUNT = 0)
Data: 3908010111
7F12D405E602D142016C21283908010111
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 12D4 05E6 02D1 42 016C 21 28 3908010111' == 0x11

'AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55'
```

Portisch, 2018-09-10:
The sync bit count is protocol defined. Default 0!? Maybe it can be defined if the bit  count MOD 8 != 0.

Like the WS_1200 protocol. 7 sync bits and 64 data bits. The sum is 71 bits: 
71 mod 8 = 7.

gerardovf, 2018-09-11:
```
This program needs at least one parameter (i.e. "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55")
Second parameter can be the number of repetitions (i.e. "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55" 20)
Default repetitions value is 20 (hex 14)
```

Next versions should be more 'user friendly'. Maybe a '-v' (verbose) option in the command line...

[BitBucketConverter.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2369962/BitBucketConverter.txt)

gerardovf, 2018-09-11:
[BitBucketConverter.py](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2371308/BitBucketConverter.txt)

```
Usage: BitBucketConverter.py [options]

Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  -i INPUT, --input=INPUT
                        input 'B1' string
  -r REPEAT, --repeat=REPEAT
                        number of times to repeat
  -d, --debug           show debug info
  -v, --verbose         show more detailed info
None
-------------------------------------------------
Command Line : -i "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55"

'RfRaw AAB033041405E6016C02D112D4012122121211212211212121221121212121212121212122112121212121212211212122112121221355'

'AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55'
-------------------------------------------------
Command Line : -v -i "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55"
Repeat: 20

'RfRaw AAB033041405E6016C02D112D4012122121211212211212121221121212121212121212122112121212121212211212122112121221355'

'AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55'
-------------------------------------------------
Command Line : -d -v -i "AA B1 04 05E6 016C 02D1 12D4 0121221212112122112121212211212121212121212121221121212121212122112121221121212213 55"
Repeat: 20

'RfRaw AAB033041405E6016C02D112D4012122121211212211212121221121212121212121212122112121212121212211212122112121221355'

Sync: 30    Data: 12 12 21 21 21 12 12 21 12 12 12 12 21 12 12 12 12 12 12 12 12 12 12 21 12 12 12 12 12 12 12 21 12 12 12 21 12 12 12 21 
Sync               0  0  1  1  1  0  0  1  0  0  0  0  1  0  0  0  0  0  0  0  0  0  0  1  0  0  0  0  0  0  0  1  0  0  0  1  0  0  0  1  
Hex: 3908010111

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 12D4
SYNC_LOW: bucket 0: 05E6
BIT_HIGH_TIME: bucket 2: 02D1
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x42 (33%)
BIT_LOW_TIME: bucket 1: 016C
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x21 (33%)
BIT_COUNT + SYNC_BIT_COUNT: 0x28 (40, SYNC_BIT_COUNT = 0)
Data: 3908010111
7F12D405E602D142016C21283908010111
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 12D4 05E6 02D1 42 016C 21 28 3908010111' == 0x11

'AA A8 11 7F 12D4 05E6 02D1 42 016C 21 28 3908010111 55'

```

ErikAndren, 2018-09-11:
I'd suggest that you fork the portisch RF-bridge git repository and add the python script there in order to get it properly revisioned?

gabrielklein, 2018-09-11:
Thank you - nice tool! :) some suggestion:

* Remove the '...' around the RfRaw command 
* Allow directly copying things like: 18:30:23 MQT: /sonoff/bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 02FD 0E48 14CF 05E6 503C 000000030102010103010103010101010101010104 55"}}
(It would speed up copy and past from the console)
* Send directly the command to the web interface, similar to curl 'http://10.0.0.201/ax?c2=191&c1=RfRaw%20AA%20A8%200B%207F%200102%2002C6%2017E8%2042%200C12%2021%200%20%2055'  
* Have a "direct input interface", it means something where I can just "copy commands" and get the result (no need to "start the command").

Greate job :)

gerardovf, 2018-09-12:
@gabrielklein Can you try if this is working and if you miss something else?

[BitBucketConverter.py](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2374903/BitBucketConverter.txt)

The script will keep asking for a string until you give an empty input.
I love your approach for a tool that makes all the work. I just copied the idea from "decode-status.py" here [tools](https://github.com/arendst/Sonoff-Tasmota/tree/development/tools) (@arendst, @ascillato2)

@Portisch would you mind putting this tool in your repository?
I don't feel like creating a fork for such a simple script as @ErikAndren suggested!

gabrielklein, 2018-09-12:
It's great :+1: 

I've just added a try - catch
```
        try:
           strInput = getInputStr()
           if (len(strInput) > 0):
               main(strInput, options.repeat)
           else:
               break
        except:
          print "Unexpected error:", sys.exc_info()[0]

```
I receive an error due to json decoding (as it's not json when it's coming back).

It's makes things really easy! You record events and just copy the messages in your tool to test if you can reproduce something.

What would be even more amazing is to respect the timing between messages - but you cannot do that with your script as you only have time in seconds :)

Thank you as it really simplify our life trying to replay what we receive!!

-------

Concerning the point of ErikAndren, you fork the project, modify your script, and then do a "pull request" - but you will spend lot of energy if you are not used to GIT - but it's great to learn if you want to participate a bit more in opensource projects!

mcspr, 2018-09-12:
@gerardovf You can create a gist instead, see https://gist.github.com
A link to it can be added to the README here.

Portisch, 2018-09-12:
> @gerardovf You can create a gist instead, see https://gist.github.com
> A link to it can be added to the README here.

This looks like a good way

gerardovf, 2018-09-12:
I'm trying to use python3

https://gist.github.com/gerardovf/15109405e8a53dd075bbe650d70fafc6

Portisch, 2018-09-17:
319914226746a5736e83049ef66b59b22b32d334

Jason2866, 2018-09-20:
@gerardovf 
THX Tool workes great!!

gerardovf, 2018-09-20:
@Jason2866 Thanks! It's just something I thought I needed for myself. If you find some improvements could be made just ask.
Anyway, the whole merit must be given to @Portisch, his code is working flawlessly.

gerardovf, 2018-10-03:
@haegar33 I can understand your confusion. This is a 'work in progress' tool :-)
Please try to start the python3 version with '-d -v' options in the command line (use verbose and debug).
I'm trying to make the tool more user friendly but I need feedback.
You can also try this version 
[BitBucketConverter181002.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2441611/BitBucketConverter181002.txt)

ciB89, 2018-10-03:
That version does not work for me. I btw also only get an A8 code. Trying to get my Flamingo Sockets with its remote to work (FA500R). I get a bunch of different B1 codes and I think without your tool I won't be able to figure it out.

gerardovf, 2018-10-03:
@ciB89 Please try to start the python3 version with '-d -v' options in the command line (use verbose and debug).
The new version (BitBucketConverter181002.txt) should also work with python2 (it's just that you won't be able to send the http command -using the '-e' option-)

ciB89, 2018-10-03:
The new version(181002) has a syntax error in line 258 (missing the brackets for print()) - and it returns the error 'Unexpected error: <class 'NameError'>'

haegar33, 2018-10-03:
I guess you have to make a decision whether to go with Python 2 or 3. For example "raw_input" does not exist in V3. Also some loops in your code will crash as V3 is more sensitive to type conversion. Example:

`for i in range(0, iLength/2))`
needs:
`for i in range(0, int(iLength/2))`

to be valid Python3 code. Actually I would prefer a much simpler V2 version without the sendCommand stuff (needing some modules I don't have..)

Jason2866, 2018-10-04:
@gerardovf 
I can confirm that BitBucketConverter181002.txt works with Python 2.7
started with options -d -v 
```

c:\Users\Hans\.platformio\python27\Scripts>BitBucketConverter.py -d -v
Enter B1 line: le/sonoff-82589D/RESULT = {"RfRaw":{"Data":"AA B1 04 0359 016A 00E6 24C2 01200120202020202020102001200120202020010101010123 55"}}
AA B1 04 0359 016A 00E6 24C2 01200120202020202020102001200120202020010101010123 55

Creating B0...
AA B1 04 0359 016A 00E6 24C2 01200120202020202020102001200120202020010101010123 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 4
Bucket 0: 0359 (hex), 857 (dec)
Bucket 1: 016A (hex), 362 (dec)
Bucket 2: 00E6 (hex), 230 (dec)
Bucket 3: 24C2 (hex), 9410 (dec)
szOutAux: 'AAB0xx04140359016A00E624C20120012020202020202010200120012020202001010101012355'
iLength: 23 (hex), 35 (dec)
szOutFinal2Send: 'AAB02304140359016A00E624C20120012020202020202010200120012020202001010101012355'

Creating A8...
szDataStr: '01200120202020202020102001200120202020010101010123'
Sync: 30    Data: 12 00 12 02 02 02 02 02 02 01 02 00 12 00 12 02 02 02 00 10 10 10 10 12
strFinalBits: '              0  0  1  1  1  1  1  1  1  0  0  1  1  1  0  '
0011 (3-3h)
1111 (15-Fh)
1001 (9-9h)
Hex: 3F9

```

b1g1an, 2018-10-05:
Hi, I grabbed some codes from a Cotech remote and tried one with BitBucketConverter181002.txt 

I get the below but don't know what I'm supposed to do with the results, don't really understand what's going on at all, just trying to follow the instructions, any help gratefully received...

```
python ./BitBucketConverter181002.py -d -v
Enter B1 line: {"RfRaw":{"Data":"AA B1 04 0219 03F9 0BCC 1C7A 01101001100110101001100101100110010101010110011023 55"}}
AA B1 04 0219 03F9 0BCC 1C7A 01101001100110101001100101100110010101010110011023 55

Creating B0...
AA B1 04 0219 03F9 0BCC 1C7A 01101001100110101001100101100110010101010110011023 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 4
Bucket 0: 0219 (hex), 537 (dec)
Bucket 1: 03F9 (hex), 1017 (dec)
Bucket 2: 0BCC (hex), 3020 (dec)
Bucket 3: 1C7A (hex), 7290 (dec)
szOutAux: 'AAB0xx0414021903F90BCC1C7A0110100110011010100110010110011001010101011001102355'
iLength: 23 (hex), 35 (dec)
szOutFinal2Send: 'AAB0230414021903F90BCC1C7A0110100110011010100110010110011001010101011001102355'

Creating A8...
szDataStr: '01101001100110101001100101100110010101010110011023'
Sync: 30    Data: 11 01 00 11 00 11 01 01 00 11 00 10 11 00 11 00 10 10 10 10 11 00 11 02 
strFinalBits: '              1  '
Hex: 
Enter SYNC_HIGH bucket number: 
```

gerardovf, 2018-10-05:
@b1g1an There's some work pending to make it user friendly. In order to get to the final data you need to enter any value between 0 and 3 at the "Enter SYNC_HIGH bucket number" like prompts.
Your goal is to get the final 'B0' message. Then just enter in the console:
`RfRaw AAB0230414021903F90BCC1C7A0110100110011010100110010110011001010101011001102355
`

b1g1an, 2018-10-05:
@gerardovf Thanks for the reply, got some codes out of it now but unfortunately the socket doesn't respond, does this look right for sniffing one button held for several seconds...

```
15:44:45 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E6 0428 0BE0 1C66 01101001011010011001010101101010011010011010010123 55"}}
15:44:45 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E5 0424 0BEA 1C5C 01101001010101100110010110101001010101101010010123 55"}}
15:44:46 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01D9 0430 0BE0 1C7A 01101001011001010110101010101010011001101010010123 55"}}
15:44:46 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 01AE 04AD 0138 0A87 1C70 01101001010110010101100110011010100110011010010134 55"}}
15:44:46 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E7 042A 0BE0 1C66 01101001011010011001010101101010011010011010010123 55"}}
15:44:47 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01EA 0422 0BEA 1C7A 01101001010101100110010110101001010101101010010123 55"}}
15:44:47 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01DC 0437 0BCC 1C84 01101001011001010110101010101010011001101010010123 55"}}
15:44:47 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E2 042B 0BE0 1C7A 01101001010110010101100110011010100110011010010123 55"}}
15:44:48 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E1 042F 0BE0 1C70 01101001011010011001010101101010011010011010010123 55"}}
15:44:48 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01EA 0428 0BE0 1C70 01101001010101100110010110101001010101101010010123 55"}}
15:44:48 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E0 0433 0BD6 1C7A 01101001011001010110101010101010011001101010010123 55"}}
15:44:49 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01DD 042E 0BEA 1C7A 01101001010110010101100110011010100110011010010123 55"}}
15:44:49 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E0 0435 0BE0 1C7A 01101001011010011001010101101010011010011010010123 55"}}
15:44:49 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E9 042D 0BE0 1C70 01101001010101100110010110101001010101101010010123 55"}}
15:44:50 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E5 042D 0BE0 1C84 01101001011001010110101010101010011001101010010123 55"}}
15:44:50 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E2 042F 0BE0 1C7A 01101001010110010101100110011010100110011010010123 55"}}
15:44:50 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E4 0430 0BE0 1C7A 01101001011010011001010101101010011010011010010123 55"}}
15:44:51 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E5 042B 0BEA 1C7A 01101001010101100110010110101001010101101010010123 55"}}
15:44:51 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E1 042D 0BE0 1C7A 01101001011001010110101010101010011001101010010123 55"}}
15:44:52 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E2 0431 0BEA 1C7A 01101001010110010101100110011010100110011010010123 55"}}
15:44:52 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01DF 0431 0BEA 1C7A 01101001011010011001010101101010011010011010010123 55"}}
15:44:52 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01EB 0428 0BE0 1C70 01101001010101100110010110101001010101101010010123 55"}}
15:44:53 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01E6 042C 0BEA 1C84 01101001011001010110101010101010011001101010010123 55"}}
15:44:53 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01DF 042D 0BE0 1C7A 01101001010110010101100110011010100110011010010123 55"}}
```

gerardovf, 2018-10-05:
@b1g1an I find strange your captures. You should get the same data all the time.
You are getting  `01101001011010011001010101101010011010011010010123` or `01101001010101100110010110101001010101101010010123` or `01101001011001010110101010101010011001101010010123` or `01101001010110010101100110011010100110011010010123` 
You're also getting 4 bitbuckets (in my tests I get only 3).
You'll have to wait for @Portisch expertise to give us some advise.

gerardovf, 2018-10-05:
@b1g1an Meanwhile try any of these commands in Tamota's console:

`RfRaw AAB023041401E604280BE01C660110100101101001100101010110101001101001101001012355`

`RfRaw AAB023041401E504240BEA1C5C0110100101010110011001011010100101010110101001012355`

`RfRaw AAB023041401D904300BE01C7A0110100101100101011010101010101001100110101001012355`

`RfRaw AAB023041401E2042B0BE01C7A0110100101011001010110011001101010011001101001012355`

b1g1an, 2018-10-06:
@gerardovf Many thanks for your help, unfortunately none of those commands did anything however, I did find some info regarding your query about the commands being different.

In this article, which appears to be the same sockets albeit I have the UK version...

https://www.instructables.com/id/Control-CoTech-Remote-Switch-With-Arduino-433Mhz/

...it says...

```
This means that the Arduino is reading the reciever. The mots interesting thing is the binary code. This remote sends a 24 bit code. This may differ from different units.

1010 0111110111100000 0111
^ ^ ^ ^ ------------------------ ^ ^ ^ ^
The first 4 digit is the "name" of the remote. The last 4 is the the name of the button that is pushed. The 16 nr in between is a random code.

If you press the same button again it will show the same four nr (1010) in the start and the same four nr (0111) in the end. The 16 nr i between will probably change. Each button on the remote can send 4 different 24bit binary codes.
```

...which seems to explain it. Would also suggest these are Flamingo sockets rebranded for Clas Ohlson.

Does any of that help?

gerardovf, 2018-10-09:
@Portisch regarding your question in Discord about sending RfRaw commands using http... that's what the BitBucketConverter.py script does (using pycurl) if a 'device' IP is set in the command line (e.g. '-e 192.168.0.123'). Currently is commented out if not running Python3.

Portisch, 2018-10-09:
Sending commands over http is already working. But how about receiving/sniffing?
My problem: I use a Phoenix Contact ILC155ETH PLC. This PLC do not have any feature like MQTT.
So I need a (easy) way to exchange data between the PLC and the RF bridge.

gerardovf, 2018-10-09:
@Portisch a quick search in Google... It seems some other people have same problem: [mqtt on PLC](https://plcnext-community.net/index.php?option=com_easydiscuss&view=post&id=593&Itemid=224&lang=en)

gerardovf, 2018-10-09:
@Portisch It would be easy to install a MQTT Mosquitto broker in a Raspberry Pi and then use the PLC to communicate with the Pi using a serial port. I can send you python examples on how to get the MQTT messages and how to use the Pi serial port.

ciB89, 2018-10-22:
I'm at it again - I need to manually fill in SYNC_HIGH, SYNC_LOW and BIT_HIGH_TIME - how do I get these numbers?

These are some of my results:

```
13:55:54 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 01F8 03F3 00FB 0064 00AA 134C 2121102121102121101021212110211010102121211021102110102125 55"}}
13:55:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 011E 0A64 00DB 04FA 2A08 010203020300030203000300030003000303000003030000030003030003000003000303000003000303000300000300030003000300030300000300030003030004 55"}}
13:55:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0130 04F4 00DE 2396 01010202020101010102020201010101020002000101010001000101000100010000010101020000010100010000010001010100000100010103 55"}}
13:55:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0174 03D0 0122 27BA 01010101011001100101010101010110010101100110011003 55"}}
13:55:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0176 03CC 012C 27C4 01010101011001100101010101010110010101100110011003 55"}}
13:55:56 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0178 03CF 0136 27BA 01010101011001100101010101010110010101100110011003 55"}}

13:56:45 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 019A 03E9 00F0 1342 2121102121102110102121102121102121212121211010102110102123 55"}}
13:56:45 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 011A 0A64 00DC 04FB 2A08 010203000300030003000300030003000303000003030000030003030003000003000303000003000303000300000300030003000300030300000300030003030004 55"}}
13:56:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 012F 04F5 00DE 23A0 01010202000101010102020001010101020000000101010001000101000100010000010101000000010100010000010001010100000100010103 55"}}
13:56:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0176 03CD 0127 27BA 01010101011001100101010101010110010101100110011003 55"}}
13:56:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0175 03CC 27C4 01010101011001100101010101010110010101100110011002 55"}}
13:56:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0170 03CB 012C 27CE 01010101011001100101010101010110010101100110011003 55"}}
```

Portisch, 2018-10-22:
> 13:56:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0175 03CC 27C4 01010101011001100101010101010110010101100110011002 55"}}

The longest bucket is every time the last nibble at the end of the data:
0101010101100110010101010101011001010110011001100**2**

Data left:
> 0101010101100110010101010101011001010110011001100

These are 49 nibbles. The number of nibbles can't be odd.
One nibble is related to the sync:
010101010110011001010101010101100101011001100110**0**

```
SYNC_HIGH: bucket 0: 0x0175µs
SYNC_LOW:  bucket 2: 0x27C4µs
```

Data left:
> 010101010110011001010101010101100101011001100110

These are 48 nibbles what are 24 bits.
bucket 1 is longer than bucket 0.
```
BIT_HIGH_TIME: bucket 1: 0x03CCµs
BIT_LOW_TIME:  bucket 0: 0x0175µs
```
So bit 1 is "10", a bit 0 is "01".:
```
010101010110011001010101010101100101011001100110
0 0 0 0 0 1 0 1 0 0 0 0 0 0 0 1 0 0 0 1 0 1 0 1
0x050115
```

ciB89, 2018-10-25:
I still don't get how to calculate/choose the SYNC_HIGH, SYNC_LOW and BIT_HIGH_TIME. The helper tool is asking me for these values - or do I use a wrong version?

gerardovf, 2018-10-25:
@ciB89 Unfortunately those values seem to be hard to guess in an automatic way. We still rely on @Portisch expertise...

Portisch, 2018-11-06:
@gerardovf You need to update your source to the actual master:  
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Decode-0xB1-sniffed-data
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h

ciB89, 2018-11-10:
One more question: Once I have got be B0 code - how can I test if it acutally works? Switching to send mode "RFRAW 176" and then "RFRAW AAB0230414017903D0012C27C4 0301010101011001100101010101010110010101100110011055"? - if this then does not turn my socket on I try a different b1 code? Or do I need to define a new protocol first (or what do I need this for?)

Also, my RF Bridge used to make a "beep" sound whenever I switched to b1 sniffing, but it does not make that sound anymore... I had to reflash the device because I changed my wifi setup, and I'm afraid that this screwed something up (since the beeping is missing). Also before I could hold the buttons on my remote to receive many many codes, but now it seems that it only receives one code and then I have to press the button again. I know this is off-topic, but I need to know if I have to fix my bridge first before I can actually get this to work ;)

gerardovf, 2018-11-14:
> 
> 
> @gerardovf You need to update your source to the actual master:
> https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Decode-0xB1-sniffed-data
> https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h
> 
> Update:
> I have done a fork of your script:
> https://gist.github.com/Portisch/6dd5d6d20465521d76538a2907ba5e55
> 
> This will do also search the sync pattern in the buckets.

@Portisch would you mind taking ownership of the python script? You are the one that gave me the idea for the tool and should receive all the credits for your wonderful work.

joecool99, 2019-02-27:
@Portisch, @gerardovf:
Thank you very much for RF-Bridge-EFM8BB1firmware and the BitBucketConverter.py script :-)

I've created a Node RED node out of it: 
[BitBucketConverterNode.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2911035/BitBucketConverterNode.txt)
using the Node RED python 3 node:
https://flows.nodered.org/node/node-red-contrib-python3-function

It can be used for example with MQTT to convert the B1 codes from sonoff bridge to the B0 codes.
[example flow.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2911106/example.flow.txt)

JoHaNMcR, 2019-05-05:
hello can anybody help me here, im getting this rfraw data but doesnt make sense with the patters you guys get

`10:00:54 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0262 00AA 1A2C B09091A1A1A091A091A1A091A1A0909181A1A18091909090 55"}}
10:00:54 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02A8 0294 281818 55"}}
10:00:55 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 0244 00B4 1A36 B09091A1A1A091A091A1A091A1A09091A1A1A1A091909090 55"}}
10:00:55 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 02BC 029E 281818 55"}}
10:01:00 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00B4 02F8 0064 02DA 381A18 55"}}
10:01:03 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 02BC 02EE 281818 55"}}
10:01:03 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0258 00B4 1A18 B09091A1A1A091A091A1A091A1A09091A1A1A1A091909090 55"}}
10:01:03 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0294 02A8 281818 55"}}
10:01:05 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 024E 00B4 1A36 B09091A1A1A091A091A1A091A1A09091A1A181A091909090 55"}}
10:01:05 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 028A 02B2 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0262 0096 1A2C B09091A1A1A0918091A1A091A1809091A1A181A091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02BC 02A8 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0258 00AA 1A4A B09091A181A091A09181A09181A0909181A181A091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 029E 0294 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0258 00A0 1A54 B09091A1A1A091A091A1A091A1809091A181A18091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 02A8 0294 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 0262 00A0 1A4A B09091A18180918091818091818090918181A1809180909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00F0 026C 1A54 A0909181818091809181809181809091818181809180909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0258 00B4 1A54 B09091A1A18091A091A1809181A090918181818091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 029E 029E 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0276 00B4 1A54 B0909181A18091809181809181A090918181818091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 029E 02A8 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 026C 00BE 1A54 B09091A181A091A091A1A091A1809091A1A181A091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02A8 029E 281818 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 0276 0096 1A5E B09091A1A1A0918091818091818090918181818091909090 55"}}
10:01:06 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 0294 0294 281818 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0262 00AA 1A4A B09091A1A1A091A091A1A09181A0909181A1818091909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00B4 029E 029E 281818 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0276 1A4A A0909181818091809181809181809091818181809180909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0258 00AA 1A4A B09091A1A1A091A091A18091818090918181818091909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 029E 02A8 281818 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 026C 00AA 1A54 B09091A1A180918091818091818090918181818091909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02A8 02B2 281818 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 026C 00B4 1A54 B0909181818091A091818091818090918181818091909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0294 02A8 281818 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AAAAB1040104026200B41A54B09091A1A180918091A1A09181A0909181818180918090909055"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 026C 1A54 A0909181818091809181809181809091818181809180909090 55"}}
10:01:07 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0262 00B4 1A54 B09091A1A1A0918091A18091818090918181818091909090 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0294 029E 281818 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 0262 00B4 1A4A B09091A18180918091A18091A1809091A1A1A18091909090 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00BE 029E 02A8 281818 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0262 00AA 1A5E B09091A1A1A091A091A18091A18090918181818091909090 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 02A8 029E 281818 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 0258 00B4 1A68 B09091A1818091809181A0918180909181A1818091909090 55"}}
10:01:08 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02A8 0294 281818 55"}}
[... cut ...]
10:01:27 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 026C 00AA 1A54 B09091A1A1A0918091818091818090918181818090918090 55"}}
10:01:27 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 02A8 029E 281818 55"}}
10:01:27 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 026C 00AA 1A54 B09091818180918091818091A18090918181818090918090 55"}}
10:01:27 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 02A8 02A8 281818 55"}}
10:01:27 MQT: tele/rf_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00F0 0276 00AA 1A4A B0909181818091809181809181A09091818181809090918090 55"}}`

yortiz1979, 2019-08-29:
Hi i wonder if any one could help me understand a little bit more this I need some help on this remote for my owning.

the remote has three four bottom

this is the control remote im using
REMOTE like in this link: https://turkish.alibaba.com/product-detail/6-0m-3-0m-large-size-retractable-full-cassette-awning-with-led-60646476303.html

OPEN
2.CLOSE
STOP
4 . LIGHT ON/OFF
I have the rf bridge v2 succesfully installed latest tamota and upgraded the latest firmware ( RF-Bridge-EFM8BB1.hex)

while using the RfRaw AAB155 and pressing the buttom few time 3-5 secs i get the folowing codes:

17:13:51 CMD: RfRaw AAB155 17:13:51 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"} 17:13:51 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}} 17:13:55 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 02B2 0384 180808 55"}} 17:14:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01D6 012C 0ECE 02F8 5E42 4818181818181818181818180818181818080808081818281808083B3B080808083B3B380B083B3B08083B0808083B083B0808083B083B3B083808080B080808381B08083B3B08 55"}} 17:14:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5CE4 481818181818081818080808080808080808080808080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}} 17:14:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0136 0ECE 02F8 5CC6 481818181818181818181808080818180818181818080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}} 17:14:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02EE 5CC6 481818181818081808080808080808080808080808080828080808080808083B080808080808083B08080808083B3B3B38080B3B3B38080B3B0808380B380B08083B080808080808080838 55"}} 17:14:06 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 02F8 5E74 4818181818181818180808080808080808080808080808280808080808380B08080808083808080B080808380B083B3B080808083B3B080808083B0808080838080B18083B3B08080808380B 55"}} 17:14:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5E4C 4818181818181808080808080808080808080808081818280808080808380B08080808083808080B080808380B083B3B080808083B3B080808083B0808080838080B08083B3B08080808380B 55"}} 17:14:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 029E 0122 0398 28090809 55"}} 17:14:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02F8 5CEE 481818181818180808080808080808080808080808180828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55"}} 17:14:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 02EE 5CC6 481818181818181818080808080808180808080808080828080808083808080B08080838080B3B08083808080B3B380B083B3B380B083B3B083B0808380B0808380B0808080838080808 55"}} 17:14:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 02F8 5E4C 481818181818181808180808180808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55"}} 17:14:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0EC4 02EE 5E56 481818181818180808080808080808080808080808080828080808083B0808080808083B08380B08083B0808083B3808080B083B3808080B083B0838080808080B080838080B0808083B0808 55"}} 17:14:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 0334 5CEE 481818181818181818080808080808081808080808080828081818180B3808181808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}} 17:14:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 0122 0ECE 032A 5CC6 481818181818080808080808080808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}} 17:14:17 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 0122 0ECE 0334 5CD0 481818181818181818181808180808080808080808080828080808080B3808080808080B3B083808080B38080B3B380B380B3B380B380B3808080B08080838080B0808380808080B3808 55"}} 17:14:19 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 028A 03B6 180808080808 55"}} 17:14:24 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0316 0190 0334 280918 55"}} 17:14:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01CC 012C 0ECE 0334 5E4C 4818181818181818180818080808080808180808080818280808080B083B3808080B08380B38080B083B3B3B083B083B3B083B083B380B3B3B0838080B38080808080B083B 55"}} 17:14:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CEE 48181818181808080808080808080808080808080808082838180B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55"}} 17:14:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CD0 48181818181808180808080808080808080808080808082838080B0838080808080B08380B08380B083808080B380B0808380B380B0808380B3B0808080808083808080B0838080B08380808 55"}} 17:14:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C2 012C 0ECE 02D0 5CC6 48181818181818080808080808080808080808080808082838AAB10501CC01220ECE033E5E5648181818181818181818080808080808080808080808082808080B0808080838080B0808083B380B080808083B380B380B3B380B380B3B3B0838080B083808080B3808080B08080808 55"}} 17:14:36 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 01CC 0136 00DC 0EC4 0334 5E4C 58182818181818181818080808181808080808080808083808080C0808080848080C0808084C480C080808084C480C480C4C480C480C4C4C08480 [... cut ...]
rfraw AA B0 53 05 08 01C2 012C 0EC4 02EE 5CDA 4818181818080808080808080808080808080808080808280808380B08380B08380B083B0808083B08380B380B083B3B380B083B3B38080B3B08083B38080B080808380B083808 55;rfraw 0
rfraw AA B0 56 05 08 01C2 012C 0EC4 02E4 5E56 48181818181818080808080808080808080808080808082808083B08380B08083B0838080B08080808380B0838080B0808080838080B08080808380B0808083B3B380B380B083B08380B 55;rfraw 0
rfraw AA B0 56 05 08 01C2 0136 0EC4 02EE 5CD0 48181818181818181808080808080808080808080808082808083B0808083B083B08080808080808080808083B38080B38080B38080B38080B380B08380B083B380B3B08083B08080838 55;rfraw 0`

thank you very much in advance for any guidance on this issue!

jotadominguez, 2020-02-27:
@yortiz1979 This same this is happening to me, the first try work right away, and then i kept trying the other B0 and none of them worked. 
And the one that worked, didnt work any more! :(

Any tips?
I have a 3 button remote SCT-R1, brand TUBE

yortiz1979, 2020-02-27:
I just dropped the idea since i did not received and help or advice in the forum and i never found any solution! If you find something please share! And good luck!!

wernerhp, 2021-03-13:
Any advice for https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/31#issuecomment-489406119 ?  I have similar problem.

jhonniedj, 2022-02-02:
For anyone who want to parse B0 from B1 data, please find my node-red flow here,
which I reverse-engineered from the BitBucketConverter.py.
Since the pyhton3 functions nodes made my node-red crash.

[B0 builder node-red.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/7987727/B0.builder.node-red.txt)
I wanted to parse this raw data to detect protocols (like pilight or rtl_433, but no luck yet)

cook1emr, 2022-06-23:
> For anyone who want to parse B0 from B1 data, please find my node-red flow here, which I reverse-engineered from the BitBucketConverter.py. Since the pyhton3 functions nodes made my node-red crash. [...]

did you get this working in the end?

jhonniedj, 2022-06-23:
> > For anyone who want to parse B0 from B1 data, please find my node-red flow here, which I reverse-engineered from the BitBucketConverter.py. Since the pyhton3 functions nodes made my node-red crash.
> > [B0 builder node-red.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/7987727/B0.builder.node-red.txt) I wanted to parse this raw data to detect protocols (like pilight or rtl_433, but no luck yet)
> 
> @jhonniedj did you get this working in the end?

I was able to reproduce the command, but not to decode it from raw commands.
I found out that I had 2 action (the store) remotes from Flamingo (KAKU sort of) which did respond differently, 1 was decodable by portisch, but the other was not, while it was able to being decoded by pilight (with usb-arduino-nano-433 decoder)

Unfortunately I was unable to connect raw output to a nice decoding-script in node-red  (like Pilight or RTL_433) in order to decode all my store-bought 433 remotes

cook1emr, 2022-06-23:
@jhonniedj I got the same error from the test Inject node

jhonniedj, 2022-06-23:
> @jhonniedj I got the same error from the test Inject node

what version of nodejs and node-red are you running?, it could be that replaceAll is not implemented yet,
maybe .replace(/ /g, "") will work in stead of .replaceAll(" ","")?

peterchs, 2022-06-24:
The online tool is working again now
https://bbconv.hrbl.pl/
May need to strip out the time, so the entry is like:
`{"RfRaw":{"Data":"AA B1 05 02B2 058C 0064 01CC 0FB4 481A38 55"}}`

jhonniedj, 2022-06-27:
> Hi looks like 14.8 node.js, looks like i need node.js 15, however i cant work out how to update the docker container for that
> 
> [image]
> 
> changing to replace does work though

Hereby the file that should resolve the "TypeError: szInpStr.replaceAll is not a function" error:
[B0.builder.node-red-without-replaceall.json.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/8991979/B0.builder.node-red-without-replaceall.json.txt)
(I replaced the other 2 functions also with .replace, apparently .replaceAll was used on 3 places)

chrismox, 2023-04-19:
> https://bbconv.hrbl.pl/ is down again it seems, can it be hosted elsewhere? Github for instance?

As you say, still down. I'm a bit lost without it, I guess I'll have to try to work out how to get the Python script running...

alexmercerscott, 2023-04-21:
Still down, bumping so it goes back on someones radar.

jhonniedj, 2023-04-21:
> Still down, bumping so it goes back on someones radar.

[convertB1.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/11295845/convertB1.txt)
Would this help? i made a nodejs file to convert.
If you rename convertB1.txt to convertB1.js, and install nodeJS, you can run it from command line like this:

node convertB1.js

Then the program needs a B1 string, that will be converted to a B0 string.
Does it help?

[image]

cook1emr, 2023-04-21:
> > Still down, bumping so it goes back on someones radar.
> 
> [convertB1.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/11295845/convertB1.txt)
> I made a nodejs file to convert.
> If you rename convertB1.txt to convertB1.js, and install nodeJS, you can run it from command line like this:
> 
> node convertB1.js
> 
> Then the program needs a B1 string, that will be converted to a B0 string.
> Does that help you out?
> 
> [image]
> 

I second this, I still use the flow in nodered. Which closes my dependence on the site

[24 more replies]