Portisch/RF-Bridge-EFM8BB1 #32: How can you reset EFM8BB1?

gerardovf, 2018-09-08:
@Portisch After using several 'B0' commands (I'm using Tasmota) the EFM8BB1 stops sending 'ACK' (MQT: gvf/cega/bridge1/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}) and I just get "MQT: gvf/cega/bridge1/stat/RESULT = {"RfRaw":"ON"}"
Restarting RF-Bridge-EFM8BB1 has no effect. Only power cycling the device.

I use "gvf/cega/bridge1/cmnd/rfraw aac002c055" to confirm that the EFM8BB1 doesn't work (it stops sending a beep)
I have to check if the problem happens when I send several 'B0' commands in a short period of time.

Portisch, 2018-09-10:
I enabled the watchdog timer: https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/eb4ac0df2915e25dcd06a150289b849cfaa17690

Untested!!

gerardovf, 2018-09-20:
@Portisch Just to keep you updated on this issue. It's very rare but it's still happening. It can take days before the EFM8BB1 stops sending 'ACK'.
I installed your watchdog timer version [(eb4ac0d)](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/eb4ac0df2915e25dcd06a150289b849cfaa17690).
The only difference I can think of is that I'm sending long 'B0' payloads through MQTT.
Today after not receiving ACK, I first did a soft reboot on the bridge with no success and then I powered down the bridge (it's connected to a Sonoff S20 with Tasmota that allows me to take the power out from the bridge remotely)

Portisch, 2018-09-21:
The limit of the command data len is:
`#define RF_DATA_BUFFERSIZE		112`

If you send more data bytes than RF_DATA_BUFFERSIZE the data will truncated. The uart ist than just waiting for the sync end (0x55) and start transmit of the truncated data. But this will be done with the "real" len what mabye higher than RF_DATA_BUFFERSIZE. So the code tries to access data from RF_DATA[>RF_DATA_BUFFERSIZE] what can lead to an access violation.

I tried to fix it here: b24bed890ef4243954fbb5468a050ba705e37080
You should get a AKN but the RF transmit would not work because the data is truncated if more than 112 bytes are getting received.

gerardovf, 2018-09-21:
@Portisch I'll check it inmediatly!
I see '**RF_DATA_BUFFERSIZE**' is used in '**RF_DATA[RF_DATA_BUFFERSIZE]**'
In '**RF_Handling.c**' there's an index **actual_byte** used in several assigments (**RF_DATA[actual_byte] =** )
Everytime the index is incremented (**actual_byte++;**) it would be safe to check it's value (**if (actual_byte > sizeof(RF_DATA))**). It is already done in **void Bucket_Received(uint16_t duration)**
I don't see it done in **void PCA0_channel0EventCb()**

I'm receiving spurious data (maybe from a neighbour's remote?) ¿does this data make sense?
```
18-09-20 10:16:06 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423F001180370DEF1AE55"}}'
18-09-20 10:16:07 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423DC00D20352DEF0D755"}}'
18-09-20 10:16:07 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423E600F0033EDEF1AE55"}}'
18-09-20 10:16:08 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423E6010E0366DEF1AE55"}}'
18-09-20 10:16:09 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423E601180352DEF1AE55"}}'
18-09-20 10:16:10 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA42404035C032ADEF1AA55"}}'

18-09-20 22:47:01 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA4242C00FA033EDEF1AE55"}}'
18-09-20 22:47:01 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423E6012C017CDEF1AF55"}}'
18-09-20 22:47:01 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423F000F00352DEF1AE55"}}'

18-09-20 23:12:58 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA422A601040352DEF1AE55"}}'
18-09-20 23:13:00 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA423E600F00348DEF1AE55"}}'
18-09-20 23:13:01 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA4240401360384DEF1AE55"}}'
18-09-20 23:13:01 Received topic 'gvf/cega/bridge1/tele/RESULT' With message: '{"RfRaw":{"Data":"AAA4221000FA037AEF78D755"}}'

```

simo878, 2018-10-07:
I am having the same problem as described (also running tasmota 6.2.1) 
The signals im sending are:
AAB017041401E60101021221F21011001010101010110100101355

gerardovf, 2018-10-07:
@simo878 Try using last patch from @Portisch. Since I installed it I had no more problems.
[b24bed8](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/b24bed890ef4243954fbb5468a050ba705e37080)

simo878, 2018-10-07:
@gerardovf  awesome, just flashed it, will see how it goes!

ErikAndren, 2018-10-09:
I'm trying the b24bed8 build and I have still the same problem with non-responding EFM-chip.
I am sending bucket commands.

gerardovf, 2018-10-09:
@ErikAndren I can confirm that this [last version](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/b24bed890ef4243954fbb5468a050ba705e37080/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex) is working for me since more than two weeks (I'm using Tasmota 6.2.0).

Portisch, 2018-10-09:
@ErikAndren What data, how often and so on do you send?

ErikAndren, 2018-10-10:
Via home assistant (HA):

Every sunset HA sends:
 payload_on: "AAB04E051401260AA000E70522298601000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003030000030003000303000455"

and 5 seconds later HA sends:
payload_on: "AAB04E0514012F0AA000E90522297C01000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003030000030003000300030455"

Every sunrise HA sends:
    payload_off: 
"AAB04E051401260AA000E70525298601000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003000300030003000303000455" 

adn 5 seconds later HA sends:
    payload_off: "AAB04E0514012E0AA000E90522297C01020303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003000300030003000300030455"

gerardovf, 2018-10-10:
@ErikAndren You're missing the 'command' part in your info. I understand you are using topic "cmnd/mydevice/rfraw"

Portisch, 2018-10-10:
Over the webinterface console: no problems

> 00:01:51 CMD: RfRaw AAB04E051401260AA000E70522298601000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003030000030003000303000455
> 00:01:51 RSL: RESULT = {"RfRaw":"ON"}
> 00:01:53 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:02:03 CMD: RfRaw AAB04E0514012F0AA000E90522297C01000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003030000030003000300030455
> 00:02:03 RSL: RESULT = {"RfRaw":"ON"}
> 00:02:05 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:02:14 CMD: RfRaw AAB04E051401260AA000E70525298601000303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003000300030003000303000455
> 00:02:14 RSL: RESULT = {"RfRaw":"ON"}
> 00:02:16 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:02:23 CMD: RfRaw AAB04E0514012E0AA000E90522297C01020303000300000303000003000300030003000303000003030000030003000300030300000300030300030000030300030000030003000300030003000300030455
> 00:02:23 RSL: RESULT = {"RfRaw":"ON"}
> 00:02:25 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}

ErikAndren, 2018-10-10:
I'm using the "cmnd/sonoff/RfRaw" MQTT command
The problem is intermittent e. g. it works for a couple of days then it just stops responding until I power cycle the RF bridge.

I will try to update the Tasmota.

Portisch, 2018-10-10:
Do not update Tasmota. Just wait until this happen again. Then try if the RF Bridge (Tasmota) ist running or freezed. If running the use the console with this command:
> RfRaw 177

The bridge should beep

ErikAndren, 2018-10-10:
Good point. I will wait for the next time and report back the result.

simo878, 2018-10-10:
Reporting back, it's been 3 days and not a single error since flashing that build (previously would error out after a few hours every time)

Portisch, 2018-10-10:
The bridge can't receive UART data if something is ongoing. For this are the chip resources too limited. Maybe Tasmota can implement a queue for transfer data to the RF chip!?

ErikAndren, 2018-10-10:
Sure, all commands received while processing a command could be dropped but it stops responding to any command even though it is done with the current command.

Tasmota implementing a queue would be handy but the EFM chip should still be able to cope with this situation

gerardovf, 2018-10-11:
@Portisch In "uart.c", ¿could you try moving the acknowledge of the interrupt **AFTER** the actual read/write is performed?
`	//Buffer and clear flags immediately so we don't miss an interrupt while processing
	uint8_t flags = SCON0 & (UART0_RX_IF | UART0_TX_IF);
	SCON0 &= ~flags;`

gerardovf, 2018-10-15:
@Portisch My approach is not to blame RF-Bridge firmware. When I get no confirmation "**_{"RfRaw":{"Data":"AAA055"}}_**" I first send a restart command to Tasmota from a script in the Raspberry Pi (using 'curl http://192.168.0.123/cm?cmnd=Restart%2099'). If that doesn't help then I power down the Sonoff bridge using a Sonoff S20.

Portisch, 2018-10-15:
I know, but it isn't a fault by the EFM8BB1 firmware. I am sure if you sniff the UART data on the pins at J3 you will not see any data between Tasmota and the RF chip anymore when it stopps working. So the problem is the UART master -> Tasmota

Jason2866, 2018-10-15:
@Portisch @gerardovf 
Yes, bug in Tasmota. Fix will be online soon.
If you cant wait change in latest dev version... in file support.ino
Find void SerialSendRaw(char *codes, int size) and add after
uint8_t code;
add line
size = strlen(codes);
Bug found and fixed by Andrethomas

mioke77, 2019-04-27:
Same issue for me, but with the latest version
Sonoff RF Bridge 433 - R2 V1.0
  Sonoff-Tasmota v6.5.0
  RF-Bridge-EFM8BB1-20181127.hex

@Jason2866, the UART issue has been resolved in this latest version?

Jason2866, 2019-04-27:
@mioke77 There are no know problems. I use this version a few month with my 2 Rf Bridges.

mioke77, 2019-05-05:
Sonoff RF Bridge 433 - R2 V1.0
Sonoff-Tasmota v6.5.0
RF-Bridge-EFM8BB1-20181127.hex

It happened again :(

I can access it via http, I can send RfRaw 177 but it does not beep

`08:14:13 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
08:14:48 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
08:15:08 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
08:16:49 CMD: rfraw 177
08:16:49 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
08:17:10 CMD: rfraw 177
08:17:10 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
08:20:02 MQT: tele/sonoff/STATE = {"Time":"2019-05-05T08:20:02","Uptime":"0T13:46:20","Vcc":3.174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"Wifi":{"AP":1,"SSId":"NET2.4GHz","BSSId":"F8:94:E6:FA:B4:47","Channel":2,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:03"}}
08:20:55 CMD: rfraw 177
08:20:55 MQT: stat/sonoff/RESULT = `{"RfRaw":"ON"}`

@Portisch could be a hardware issue with my RF bridge?

zox-gh, 2019-11-04:
I can confirm I suffer from the same issue, after several days of working properly. 
Exactly the same as described in previous posts. 
tasmota v6.5.0 + Portisch

After power cycle it works again but only few transmissions. Then it stops completely, does not receive signal even from native Sonoff Rf fob.

Certainly not an issue with rf bridge hardware as I have 2 and tested both. The same behavior.

troelssiggaard, 2020-03-09:
I'm having the exact same issues as described above. I send RF raw codes over MQTT to Sonoff RF Bridge R2 with Tasmota v8.1.0 flashed with Portisch firmware. After a power cycle of the bridge I can send a few codes, then after a few hours the EFM8BB1 stops responding to the RawRF codes  I send - a reboot of the devices has no effect - I need to do a full physical power cycle of the bridge before the RF chip starts responding again.

Did anyone come up with a solution to this? Some kind of combination of Tasmota version + Portisch version? The codes I send are for my Dooya binds:

Ex.:
"RfRaw AA B0 35 05 08 12FC 05DC 02D0 017C 22B0 0123322323233223233223323232233232233232232323232332322323323232233232322332323224 55"

This issues is making my bridge useless for the purpose of automatic blinds, so I would love to know how to fix this!

gerardovf, 2020-03-09:
@troelssiggaard Have you checked if Tasmota doesn't complain about blank spaces?
Have you tried: "RfRaw AAB035050812FC05DC02D0017C22B0012332232323322323322332323223323223323223232323233232232332323223323232233232322455"?
@Jason2866 @Portisch

troelssiggaard, 2020-03-09:
> @troelssiggaard Have you checked if Tasmota doesn't complain about blank spaces?
> Have you tried: "RfRaw AAB035050812FC05DC02D0017C22B0012332232323322323322332323223323223323223232323233232232332323223323232233232322455"?
> @Jason2866 @Portisch
@gerardovf 
I mean, it works for some time, so I don't see why some spaces would make it not work eventually.

gerardovf, 2020-03-09:
> @troelssiggaard Have you checked if Tasmota doesn't complain about blank spaces?
> Have you tried: "RfRaw [...]

You're right! In file 'xdrv_06_snfbridge.ino': SerialSendRaw(**RemoveSpace**(XdrvMailbox.data));

Jason2866, 2020-03-09:
This does mean Tasmota does remove white spaces and the are no problem.
It seems not every Portisch firmware is stable with Rf raw. Try the one (is not the latest)
from Tasmota github https://github.com/arendst/Tasmota/blob/development/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181127.hex

troelssiggaard, 2020-03-09:
@Jason2866  That is the Portisch hex that I've used. But have tried several versions...

Jason2866, 2020-03-09:
Sorry cant help any further. My two RF Bridges do run several month without a problem.
I use rf raw too.

LucReynders, 2020-03-14:
Same here.
Installed tasmota 8.xxx and portisch on my second rfbridge.
As soon as I send a rfraw with data the bridge blocks and needs a power restart. The command does not create the rf putput.
My old rfbridge accepts these command without any problem.

gerardovf, 2020-03-14:
> Same here.
> Installed tasmota 8.xxx and portisch on my second rfbridge.
> As soon as I send a rfraw with data the bridge blocks and needs a power restart. The command does not create the rf putput.
> My old rfbridge accepts these command without any problem.

No problem on my bridge (Tasmota 8.1.0.4). Maybe has to do with the messages you are sending? (length,...)

LucReynders, 2020-03-15:
Solved the problem :

- installed Tasmotizer to flash (struggled some time with installing this on Linux Deb laptop)
- re-flashed RFBridge with Tasmota (as explained in the special flash proc for RFBridge)
- downloaded the tasmota project from github
- firmwate upgrade with Portisch selecting the latest hex file from the download (with correct wiring)

But it seems that I have to use different B0 codes than for my old Bridge ?
Has the sniffed B1 and bitbucketconverter been changed due to updates ?
(I had to re-snif codes and rund bitbuck again for my next rfbridge ,  old B0 codes do not activate my rf receivers)

LucReynders, 2020-05-26:
Not solved.  The bridge still gets blocked after some unpredictable tile (hours or day).  The webui is still working but the sending/receiving gets blocked.  Only a power down/up restarts the bridge.
I am sending rfraw Commands 2 times with a 1 sec delay To make sure that they get received at the receiving end.
Should I use a bigger delay between 2 transmits ?
What is the recommended delay between 2 transmits ?

vdiogo, 2020-06-16:
Suffering from same issue here: have to do a power cycle for the rfraw commands to work again

jnt2007, 2020-09-09:
The same. Is it hardware issue with new Sonoff RF revisions?

Jason2866, 2020-09-09:
Try this version (not the latest) https://github.com/arendst/Tasmota/raw/development/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20190220.hex 
some newer versions do not run stable in our test in raw mode

jnt2007, 2020-09-09:
@Jason2866 are you sure that the version which you recommend is a not the latest?
_RF-Bridge-EFM8BB1-20190220.hex_ it is a latest Portisch fw **0x04**.

Jason2866, 2020-09-09:
The build numbering is not changed always. We deleted the version in Tasmota github which haf problems with rfraw. You can try the others too.

lustyffh, 2021-01-05:
I also face the same problem with Tasmota 9.1. Apparently, the issue should be reopened.

jnt2007, 2021-01-06:
@lustyffh try to flash version which recommends @Jason2866. For me it works like a charm more than 3 months!

I just forgot to wrote a comment that @Jason2866 recommendation helpful.

istvanruman, 2021-01-24:
Hi, Have the same issue. I have to do a power cycle for the rfraw commands to work again. The link of @Jason2866 is not availabe. van you help me with the working flash version?

Jason2866, 2021-01-24:
The firmware is here https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffRfBridge_efm8bb1

istvanruman, 2021-01-24:
Thank you, I will try

lustyffh, 2021-01-24:
> @lustyffh try to flash version which recommends @Jason2866. For me it works like a charm more than 3 months!

Hi, just writing to confirm that this fix works indeed. 2 weeks now without a single halt.

istvanruman, 2021-02-02:
Works fine for me too. Thanks