Portisch/RF-Bridge-EFM8BB1 #33: Add codes for X10 protocol remote control

ubeaut, 2018-09-21:
Hi,
I have managed to get the firmware to turn on and off X10 devices by sending rfraw commands.
Is it possible to have the X10 put into the firmware as a known protocol so it shows up when the remote is used and the results show up in the tasmota console like the other remotes that work with tasmota.
X10 have house codes and device numbers. Below are 2 examples that I have used and they work.
Are they added to the protocols.h?

Here are the codes I managed to get to work:
Code A2 device
ON:
RfRaw AAB02E0501236E116C022A06A69C360122232322222222222322222323232323222222232222222223232322232323232455
OFF:
RfRaw AAB02E050123641180022F069A9C360122232322222222222322222323232323222223232222222223232222232323232455

The code for A9 device
ON:
RfRaw AAB02E05012378116C023006999C360122232322222322222322222323222323222222222222222223232323232323232455
Off:
RfRaw AAB02E0501236E1176021A06B09C360122232322222322222322222323222323222223222222222223232223232323232455

Thanks

ubeaut, 2018-09-22:
Hi,
I have managed to install the Silicon Labs Studio 4 and have imported the firmware code and can compile the hex file.
I do not know what to put into the rf_protocols.h file for the X10 remote.
I don't understand the high, low sync times etc.
If I could get  the code to paste into the rf_protocls.h and any other files I am happy to compile and see what happens.
Any help would be great.
Thanks

Portisch, 2018-09-24:
This protocol can't be defined. The bit timing isn't working "normal". You can only use bucket mode!

closed 2018-09-24 by Portisch

Portisch, 2018-09-24:
Sure, you can modify the source to support your protocol. I won't.
You have to do the protocol disassemble/assemble by your mqtt client/broker.

The "normal" bit timing is a defined by a high/low sequence.
In your sniffed data there are buckets like this: 22222222222
So the timing of each bucket ist the same. What is it? Bit 0 or bit 1...
This can't be decoded in "bit" mode. Only bucket mode.

ubeaut, 2018-09-24:
I would *love* to modify the source to support my protocol if only I knew
what to put into the source. I just want it to show up on the bridge
console so I can use node-red and mqtt to do other things. I don't need it
to turn on and off from the bridge at all.

I am not a programmer or have any electronic experience either.

I am on the right track with what I have done below?

I have tried this:
I did this:python BitBucketConverter.py -d -i "AA B1 05 236E 1176 021A 06B0
9C36 01222323222223222223222223232223232222232222222222232322232323232324
55"

'RfRaw
AAB02E0514236E1176021A06B09C360122232322222322222322222323222323222223222222222223232223232323232455'

Sync: 40    Data: 12 22 32 32 22 22 32 22 22 32 22 22 32 32 22 32 32 22 22
32 22 22 22 22 22 32 32 22 32 32 32 32 32
Sync               0
Hex:

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 06B0
SYNC_LOW: bucket 0: 236E
BIT_HIGH_TIME: bucket 2: 021A
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0xa (89%)
BIT_LOW_TIME: bucket 1: 1176
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x59 (89%)
BIT_COUNT + SYNC_BIT_COUNT: 0x1 (1, SYNC_BIT_COUNT = 0)
Data:
7F06B0236E021Aa1176591
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 06B0 236E 021A a
1176 59 1 ' == 0x0B

'AA A8 0B 7F 06B0 236E 021A a 1176 59 1  55'

Then I converted the HEX values to decimal and then I edited the
RF_Protocols.h and changed the rohrmotor24 in that file and added my number
in there.
I successfully compiled the hex file.
I uploaded it to the bridge and............................it still didn't
work

*What I changed RF_Protocols.h Below:*

/*
 * Rohrmotor24
 *
https://github.com/bjwelker/Raspi-Rollo/tree/master/Arduino/Rollo_Code_Receiver
 */
#define ROHRMOTOR24_IDENTIFIER                  0x02

SI_SEGMENT_VARIABLE(ROHRMOTOR24, static DUTY_CYLCE_PROTOCOL_DATA,
SI_SEG_CODE) =
        {
                1712,           // SYNC_HIGH
                9070,           // SYNC_LOW
                0,                      // SYNC_BIT_COUNT
                538,            // BIT_HIGH_TIME
                4470,           // BIT_LOW_TIME
                10,                     // BIT_HIGH_DUTY
                89,                     // BIT_LOW_DUTY
                1,                      // BIT_COUNT
                8                       // REPEAT_DELAY
        };