Portisch/RF-Bridge-EFM8BB1 #34: Support for Brennenstuhl Devices

sroettgermann, 2018-09-23:
Hi there,

is there currently any support for Brennenstuhl devices (remotes). I tried to learn the codes for the power sockets, but I'm not getting any results.

Best regards
Sebastian

Portisch, 2018-09-23:
No results with bucket sniffing?

sroettgermann, 2018-09-23:
When I'm sending the command I'll receive nothing. 

`01:13:27 CMD: RfRaw AAB155
01:13:27 SRC: WebConsole from 192.168.178.20
01:13:27 RSL: Gruppe 0, Index 1, Befehl RFRAW, Daten AAB155
01:13:27 RSL: RESULT = {"RfRaw":"ON"}`

Portisch, 2018-09-23:
It maybe not 433MHz!? I would need all information you have about the device.

sroettgermann, 2018-09-23:
Here you can find the full information about the device: https://www.amazon.de/Brennenstuhl-Funkschalt-Set-Funksteckdosen-Innenbereich-Kindersicherung/dp/B001AX8QUM/ref=sr_1_1?ie=UTF8&qid=1537713006&sr=8-1&keywords=brennenstuhl%2Brcs%2B1000&th=1

It is a 433,92 MHz device. The Sonoff Bridge recognizes the signal, since the red LED is blinking when I'm pushing a button on the remote. But I don't get any output on the console.

On their website they are mentioning, that the devices are good to use in raspberry pi projects. ^^ https://www.brennenstuhl.com/de-DE/produkte/funksteckdosen/funkschalt-set-rcs-1000n-comfort-1-handsender-2-schaltempfaenger-1000w

Portisch, 2018-09-24:
This plug should be detected in normal sniffing mode!? (0xA4)
https://gist.github.com/0xFelix/2151e4d1c4c4ff75dccaef1dd14e26fb

But 0xB1 (bucket sniffing) should work also. Maybe you have another 433MHz device to test the sniffing?

iamsven, 2018-09-29:
I am trying to use [Brennenstuhl RC CE1 3001](https://www.amazon.de/gp/product/B074DVCXWM/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1 ) devices. Only sniffing with 0xB1 shows results, but I am not able to translate them into proper commands for sending. I used your Python tool for the translation.

Here is an example of pressing the "A" button on the remote several times.

`19:12:11 MQT: /Smarthome/Sonoff-RF-Bridge-1/stat/RESULT = {"RfRaw":"ON"}
19:12:11 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
19:12:23 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0421 025B 019D 0B68 1BE4 01200101200101200101012020202020200120010101202034 55"}}
19:12:24 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041E 0252 01A0 0B72 1BEE 01200101012001010101202020200120012001200101202034 55"}}
19:12:26 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0421 0252 01A0 0B68 1BDA 01200101010101012020010101202020200101200101202034 55"}}
19:12:28 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0422 0237 0197 0B7C 1BE4 01100101101010011010012010100101202020010101101034 55"}}
19:12:28 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 049A 01CB 0129 012A 0A3D 1BD0 01100101100101100101011010101010100110010101101045 55"}}
19:12:28 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041E 024F 01A0 0B7C 1BD0 01200101012001010101202020200120012001200101202034 55"}}
19:12:29 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0428 01D0 019D 0B5E 1BE4 01100101010101011010010101101010100101100101101034 55"}}
19:12:29 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0494 01C7 012D 0122 0A49 1BD0 01100101101010011010011010100101101010010101101045 55"}}
19:12:30 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0428 024F 019C 0B68 1BDA 01200101010101012020010101202020200101200101202034 55"}}
19:12:31 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0494 01C7 012C 0122 0A46 1BDA 01100101101010011010011010100101101010010101101045 55"}}
19:12:31 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041F 0249 01A2 0B72 1BDA 01100101200101200101012010202010200120010101202034 55"}}
19:12:31 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 049A 01E5 0113 0A47 1BD0 01100101011001010101101010100110011001100101101034 55"}}
19:12:32 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041F 01C8 019E 0B72 1BEE 01100101010101011010010101101010100101100101101034 55"}}
19:12:33 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0426 0240 0197 0B72 1BDA 01100101101010011020011010100101101020010101101034 55"}}
19:12:36 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041C 0254 019B 0B7C 1BDA 01200101200101200101012020202020200120010101202034 55"}}
19:12:36 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 049F 01D5 0126 0A39 1BC6 01100101011001010101101010100110011001100101101034 55"}}
19:12:36 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 041D 01CA 019E 0B72 1BD0 01100101010101011010010101101010100101100101101034 55"}}`

What would be the steps to extract the proper code?

gerardovf, 2018-10-02:
@iamsven I'm using B0 mode with 2 devices and it is working flawlessly. I just send the (long) command in the MQTT payload.
The only drawback is that I cannot assign the commands to any of the 16 buttons in the bridge.

In the python script I entered :
19:12:26 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0421 0252 01A0 0B68 1BDA 01200101010101012020010101202020200101200101202034 55"}}
And I got:
>>> 
Command Line : -d -v
AA B1 05 0421 0252 01A0 0B68 1BDA 01200101010101012020010101202020200101200101202034 55

Creating B0...
AA B1 05 0421 0252 01A0 0B68 1BDA 01200101010101012020010101202020200101200101202034 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 5
Bucket 0: 0421 (hex), 1057 (dec)
Bucket 1: 0252 (hex), 594 (dec)
Bucket 2: 01A0 (hex), 416 (dec)
Bucket 3: 0B68 (hex), 2920 (dec)
Bucket 4: 1BDA (hex), 7130 (dec)
szOutAux: 'AAB0xx05140421025201A00B681BDA0120010101010101202001010120202020010120010120203455'
iLength: 25 (hex), 37 (dec)
szOutFinal2Send: 'AAB02505140421025201A00B681BDA0120010101010101202001010120202020010120010120203455'

Creating A8...
szDataStr: '01200101010101012020010101202020200101200101202034'
Sync: 40    Data: 12 00 10 10 10 10 10 12 02 00 10 10 12 02 02 02 00 10 12 00 10 12 02 03 
strFinalBits: '              0  0  1  0  1  1  1  0  0  1  '
0010 (2-2h)
1110 (14-Eh)
Hex: 2E
iLength: 0C (hex), 12 (dec)

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 0B68
SYNC_LOW: bucket 4: 1BDA
BIT_HIGH_TIME: bucket 0: 0421
BIT_HIGH_DUTY: (100% / (bucket 3 + bucket 0)) * bucket 3: 0xc (12%)
BIT_LOW_TIME: bucket 1: 1BDA
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x57 (87%)
BIT_COUNT + SYNC_BIT_COUNT: 0xa (10, SYNC_BIT_COUNT = 0)
Data: 2E
7F0B681BDA0421c1BDA57a2E
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 0B68 1BDA 0421 c 1BDA 57 a 2E' == 0x0C

A8 command: 'AA A8 0C 7F 0B68 1BDA 0421 c 1BDA 57 a 2E 55

B0 command: 'AAB02505140421025201A00B681BDA0120010101010101202001010120202020010120010120203455'

>>>

haegar33, 2018-10-03:
Unfortunately I got the newer version called **RC CE1 4001**. With this set none of the B0 codes are working. When sniffing I get:

14:26:05 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 03A5 03C2 0267 019C 0B7C 1C3E 30020202020230020230300202023030020230303002020245 55"}}

14:26:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 039E 0451 0265 0197 0B86 1C48 30020202020230303030300230020202023030023002020245 55"}}

14:26:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 03A3 0439 0267 0190 0B72 1C3E 30020202023002300202023030300230020230023002020245 55"}}

only pressing the D-ON Button (We have 4 plugs A-D)

iamsven, 2018-10-03:
@gerardovf, unfortunately, the commands you provided (`AAB02505140421025201A00B681BDA0120010101010101202001010120202020010120010120203455` and `AA A8 0C 7F 0B68 1BDA 0421 c 1BDA 57 a 2E 55`) didn't work.

I run the sniffing again, but the output commands didn't work either:
```
Enter B1 line: 13:08:02 MQT: /Smarthome/Sonoff-RF-Bridge-1/tele/RESULT = {"RfRaw":{"Data":"AA B1 04 041C 01C9 0B72 1BD0 01100101101010011010011010100101101010010101101023 55"}}
AA B1 04 041C 01C9 0B72 1BD0 01100101101010011010011010100101101010010101101023 55

Creating B0...
AA B1 04 041C 01C9 0B72 1BD0 01100101101010011010011010100101101010010101101023 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 4
Bucket 0: 041C (hex), 1052 (dec)
Bucket 1: 01C9 (hex), 457 (dec)
Bucket 2: 0B72 (hex), 2930 (dec)
Bucket 3: 1BD0 (hex), 7120 (dec)
szOutAux: 'AAB0xx0414041C01C90B721BD00110010110101001101001101010010110101001010110102355'
iLength: 23 (hex), 35 (dec)
szOutFinal2Send: 'AAB0230414041C01C90B721BD00110010110101001101001101010010110101001010110102355'

Creating A8...
szDataStr: '01100101101010011010011010100101101010010101101023'
Sync: 30    Data: 11 00 10 11 01 01 00 11 01 00 11 01 01 00 10 11 01 01 00 10 10 11 01 02
strFinalBits: '              1  '
Hex:
Enter SYNC_HIGH bucket number: 0
Enter SYNC_LOW bucket number: 1
Enter BIT_HIGH_TIME bucket number: 2
iLength: 0B (hex), 11 (dec)

The data for command 0xA8 will be:
SYNC_HIGH: bucket 0: 041C
SYNC_LOW: bucket 1: 01C9
BIT_HIGH_TIME: bucket 2: 0B72
BIT_HIGH_DUTY: (100% / (bucket 0 + bucket 2)) * bucket 0: 0x56 (86%)
BIT_LOW_TIME: bucket 1: 01C9
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0xd (13%)
BIT_COUNT + SYNC_BIT_COUNT: 0x1 (1, SYNC_BIT_COUNT = 0)
Data:
7F041C01C90B725601C9d1
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 041C 01C9 0B72 56 01C9 d 1 ' == 0x0B

A8 command: 'AA A8 0B 7F 041C 01C9 0B72 56 01C9 d 1  55

B0 command: 'AAB0230414041C01C90B721BD00110010110101001101001101010010110101001010110102355'
```

Is there a special strategy behind wihich of the sniffed codes I have to use?
Eventough I'am pressing always the same button, the data portion of the command changes...

Here are the results from [Sniffing Brennenstuhl Button "A"](https://gist.github.com/iamsven/da2c7398cd359c7d4774f7cd7c815eaa)

gerardovf, 2018-10-03:
@iamsven Maybe you should try first with another remote/device...
I'm not an expert, that's @Portisch task, but for me it's strange to see messages with 5, 6, 7 bit buckets. Those are just 'time' values. I think 3 bit buckets should be enough. In my tests I just get 3, 4 bit buckets (and I discard the 4 bit bucket 'bad' ones)
I did all my first test with the bridge near the remote to minimize possible interferences or bad signal level.
Then I just chose the message that has the same information in the 'data' part as many of the other messages. And then select the 'right' bit buckets (the ones with similar hex values in most of the sniffed messages). I think that with more than 3 bit buckets it will be difficult to determine which are the good ones. Anyway, @Portisch is the expert in this field!

Portisch, 2018-10-08:
> Try this:
> 
> ```
> AA	uart sync init
> A8	transmit RF data
> 0F	data len
> 7F	protocol identifier 0x7F
> 0B68	SYNC_HIGH
> 1BE4	SYNC_LOW
> 041F	BIT_HIGH_TIME
> 42	BIT_HIGH_DUTY (66%)
> 01F4	BIT_LOW_TIME
> 21	BIT_LOW_DUTY (33%)
> 18	BIT_COUNT (24)
> B6E05C	RF data to send
> 55	uart sync end
> ```
> Result to a RfRaw string:
> `AA A8 0F 7F 0B68 1BE4 041F 42 01F4 21 18 B6E05C 55`
> 
> Or maybe this:
> `AA A8 0F 7F 1BE4 0B68 041F 42 01F4 21 18 B6E05C 55`

Please try again with the newest master: 71e7134d7873a4da81d5f1c5551fc68dbfccffd6

Portisch, 2018-10-09:
> Please try again with the newest master: [71e7134](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/71e7134d7873a4da81d5f1c5551fc68dbfccffd6)

Use the RfRaw strings **WITHOUT** spaces!
Like:

> RfRaw AAA80F7F0B681BE4041F4201F42118B6E05C55

iamsven, 2018-10-10:
Thank you for the quick reply @Portisch 

I tried both the codes (`AA A8 0F 7F 0B68 1BE4 041F 42 01F4 21 18 B6E05C 55` and `AA A8 0F 7F 1BE4 0B68 041F 42 01F4 21 18 B6E05C 55`) you provided, without spaces on the Bridge with the old firmware version and with the [71e7134](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/71e7134d7873a4da81d5f1c5551fc68dbfccffd6) version.
Unfortunately, none of them work on both firmware versions. :/

I now tried sniffing (with the new firmware) pressing the "B" button, which is paired with another/different socket. (RF Bridge and remote were 5cm apart) - I had no success again...
[Sniffing the "B" button](https://gist.github.com/iamsven/c8a04e7e0ef65d28d8d045f09ce98546)

Do you have any idea would I could be doing wrong?

haegar33, 2018-10-15:
I have given up on Brennestuhl devices. They simply work with an unknown protocol.

Portisch, 2018-10-15:
May you try to vary the timing. Longer/Shorter when resending...

sroettgermann, 2018-10-15:
> They simply work with an unknown protocol.

Maybe that is not completely true. In my opinion there might be a problem with the code here or the bridge itself, because when I'm working with a simply sender/receiver unit attached to a RaspberryPi, it is working fine.
With the RF-Bridge it is still not working for me, too.

Portisch, 2018-10-15:
> Maybe that is not completely true. In my opinion there might be a problem with the code here or the bridge itself, because when I'm working with a simply sender/receiver unit attached to a [...]

Can you "sniff" the buckets with the RaspberryPi? Than I can compare it with the sniffed data from the bridge. Or any chance to record the raw RF data? Do you have an oscilloscope?

Jason2866, 2018-10-21:
I think newer Brennenstuhl devices uses Intertechno V3 Protokoll
Could be solved with #41

Portisch, 2018-11-05:
Rolling code, not enough space on the device to implement.

closed 2018-11-05 by Portisch