Portisch/RF-Bridge-EFM8BB1 #35: How to learn after flashing?

Trippersgarage, 2018-09-28:
Hi. Excuse the dumb question, I'm a noob in waaaaay over his head.

I got the rf bridge flashed to tasmota and have gotten Portisch flashed as well....after days of head scratching and googling. Am now completely stuck on how to get the bridge to learn a fan remote I have. 

The remote has 8 buttons. I used the rfraw AAB155 command to sniff and got the following for a few random buttons;

16:41:02 CMD: rfraw AAB155
16:41:02 RSL: RESULT = {"RfRaw":"ON"}
16:41:02 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
16:41:13 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 0064 02FA 0177 0150 2274 22121212112211221122121212121122121212112122112214 55"}}
16:41:13 RSL: RESULT = {"RfRaw":{"Data":"AA B1 05 02DF 033D 0184 01D6 299A 22020202002200220022020202020022020202002022002204 55"}}
16:41:13 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0178 02F5 2968 00101010110011001100101010101100101010110100110012 55"}}

As I understand it, I shroud use the bitbucketconvertor tool so I use it on;

{"RfRaw":{"Data":"AA B1 03 017B 02F1 295E 00101010110011001100101010101100101010110100110012 55"}}

and the result I get is;

AA A8 22 7F 00101010110011001100101010101100101010110100110012 017B 295E 5d 02F1 6 0  55

Question is what do I do then? Where do I go from here? 

Would appreciate any help on just one of these buttons, I will work out the rest. I'm sure I've goofed up already and would appreciate any guidance. And yeah, I'm an idiot so if you could possibly please help explain to me what I should do like I'm a 5 year old. :))))

Thank you!!!!!

ErikAndren, 2018-09-29:
In the bridge web console type RfRaw and the generated string you have above without any intra whitespace
I. E. RfRaw AAA8...55

gerardovf, 2018-09-29:
@Trippersgarage take a look at this: https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/31

Trippersgarage, 2018-09-30:
Thank you for the help. :)

Been trying but I'm still getting stuck. 

13:53:38 CMD: rfraw AAB155
13:53:38 RSL: RESULT = {"RfRaw":"ON"}
13:53:38 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
13:53:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0302 02BC 016B 2F94 22020202002200220022020202020022020202020200220023 55"}}
13:56:04 CMD: RfRaw AAA80B7F2F940302016B2202BC41055   (<--- The generated string from bitbucketconverter, taking out all the whitespace as per ErikAndren's suggestion)
13:56:04 RSL: RESULT = {"RfRaw":"ON"}

And then where do I go from here? I tried 'learning' the buttons;

13:56:48 RSL: RESULT = {"RfKey1":"Start learning"}
13:56:48 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
13:57:19 RSL: RESULT = {"RfRaw":{"Data":"AAA255"}}

But it doesn't seem to work, it just sits there.

I tried EricAndren suggestion, I'm sure thats correct but I don't know whats the next step? 

Gerardovf; I read that many times before I posted and after you suggested it. I'm trying to follow and understand but I'm getting lost.. 

Thank you all for your help and patience, it really is very much appreciated.

Portisch, 2018-09-30:
You can't convert this B1 data to an A8 command. You have some '22' and '00' as bits in your data. This can't be bit 0 or bit 1.

Just use the tool from @gerardovf to convert it to a B0 data. Then send the B0 command.
Learning for unknown protocols isn't possible.

Trippersgarage, 2018-09-30:
I did use the tool, that was the result. Perhaps I'm doing something stupid with the tool. 

As a test, I tried using the example string in the readme file;

"In the command line give the 'B1' message string and the retries value (in decimal): i.e. In the command line give the 'B1' message string and the retries value (in decimal):<br/>
 i.e. BitBucketConverter.py "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20
Command Line : "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20<br/>

I input that sting into the tool and the result I get is 'AA A8 0B 7F 3EBC 07EB 00FD 2a 0157 39 0  55'
Shouldn't I get AAB01C041407EB015700FD3EBC01010101010110100101010110101010010355 as a result in this test?

Trippersgarage, 2018-10-01:
@gerardovf I wasn't before, but I've tried it now. Still having issues.

I've been using the tool on my Mac as well as on ubuntu. Regardless of which version I use or which platform, its not working for me. 

It just goes back to command line when using the  [Gist Link](https://gist.github.com/gerardovf/15109405e8a53dd075bbe650d70fafc6) latest version as suggested, it just goes back to the command line. This is what I get;

buntu@ubuntu:~/Desktop$ python3 ./BitBucketConverter.py
Enter B1 line: "AA B1 03 0178 02F5 2968 00101010110011001100101010101100101010110100110012 55"
ubuntu@ubuntu:~/Desktop$ python3 ./BitBucketConverter.py
Enter B1 line: "AA B1 04 07EB 0157 00FD 3EBC 010101010101101001010101101010100103 55" 20
ubuntu@ubuntu:~/Desktop$ 

On OSX, same issue.

I'm sure its user error, basically me doing something stupidly obviously wrong.

gerardovf, 2018-10-01:
@Trippersgarage The new python3 script is designed to enter the whole information you are getting in the console.
Try to enter something like this:
`17:56:30 MQT: gvf/cega/bridge1/tele/RESULT = {"RfRaw":{"Data":"AA B1 03 0151 0293 2D5A 00110010101010110011001100 55"}}`
I'll try to modify the tool to give some hint if the entered text does not follow the rules.

Portisch, 2018-10-01:
I took again a look and it looks like "normal" protocol. I have taken my first look with my phone ;).

Your original sniffed data:
`AA B1 05 02DF 033D 0184 01D6 299A 22020202002200220022020202020022020202002022002204 55`

Put the sync in front:
`42202020200220022002202020202002202020200202200220`

Split:
```
sync	42
data	20 20 20 20 02 20 02 20 02 20 20 20 20 20 02 20 20 20 20 02 02 20 02 20
binary	0  0  0  0  1  0  1  0  1  0  0  0  0  0  1  0  0  0  0  1  1  0  1  0
hex	0A821A
```

(The data 22121212112211221122121212121122121212112122112214 or 00101010110011001100101010101100101010110100110012 brings the same result in hex.)

But this isn't a PT2260 protocol so it can't be sniffed by default.

Try this:
```
AA	uart sync init
A8	transmit RF data
0F	data len
7F	protocol identifier 0x7F
299A	SYNC_HIGH
0184	SYNC_LOW
02DF	BIT_HIGH_TIME
41	BIT_HIGH_DUTY (65%)
0184	BIT_LOW_TIME
23	BIT_LOW_DUTY (35%)
18	BIT_COUNT (24)
0A821A	RF data to send
55	uart sync end
```
Result to a RfRaw string:
`AA A8 0F 7F 299A 0184 02DF 41 0184 23 18 0A821A 55`

@Trippersgarage: If it is working I can define a new protocol for easier usage.

gerardovf, 2018-10-01:
@Portisch I'm trying to improve the tool. The 'A8' calcs are currently 'hardwired'.
Can you help me with the rules to calculate SYNC_HIGH, SYNC_LOW, BIT_HIGH_TIME, BIT_HIGH_DUTY, 
BIT_LOW_TIME, BIT_LOW_DUTY, BIT_COUNT, SYNC_BIT_COUNT?

**Command Line : -d -v

Creating B0...
AA B1 05 02DF 033D 0184 01D6 299A 22020202002200220022020202020022020202002022002204 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 5
Bucket 0: 02DF (hex), 735 (dec)
Bucket 1: 033D (hex), 829 (dec)
Bucket 2: 0184 (hex), 388 (dec)
Bucket 3: 01D6 (hex), 470 (dec)
Bucket 4: 299A (hex), 10650 (dec)
szOutAux: 'AAB0xx051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'
iLength: 25 (hex), 37 (dec)
szOutFinal2Send: 'AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'

Creating A8...
szDataStr: '22020202002200220022020202020022020202002022002204'
Sync: 42    Data: 20 20 20 20 02 20 02 20 02 20 20 20 20 20 02 20 20 20 20 02 02 20 02 20 
strFinalBits: '              0  0  0  0  1  0  1  0  1  0  0  0  0  0  1  0  0  0  0  1  1  0  1  0  '
0000 (0-0h)
1010 (10-Ah)
1000 (8-8h)
0010 (2-2h)
0001 (1-1h)
1010 (10-Ah)
Hex: 0A821A
iLength: 0F (hex), 15 (dec)

The data for command 0xA8 will be:
SYNC_HIGH: bucket 4: 299A
SYNC_LOW: bucket 2: 02DF
BIT_HIGH_TIME: bucket 2: 0184
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x1f (68%)
BIT_LOW_TIME: bucket 1: 033D
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x44 (68%)
BIT_COUNT + SYNC_BIT_COUNT: 0x18 (24, SYNC_BIT_COUNT = 0)
Data: 0A821A
7F299A02DF01841f033D44180A821A
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 299A 02DF 0184 1f 033D 44 18 0A821A' == 0x0F

'AA A8 0F 7F 299A 02DF 0184 1f 033D 44 18 0A821A 55'**

Portisch, 2018-10-01:
This sniffed data is quit hard to decode per automatic script.

There are 5 buckets in the list but only 3 buckets are used in the data - so just ignore the other ones.
This is  possible chitter of bad hardware...

So bucket 0, 2 and 4 are left.

Sync bucket (longest bucket) is everytime the last nibble of the data string: bucket index 4.
A sync have a high and low time, normaly. So the first nibble at the data is the sync_low. In this case bucket 0. This is because the data is starting recording after the longest bucket.

Attention: this will fit most of the known sniffed data. If you got it working take a look here: #34 
But lets do first this kind of data!

So two buckets 0 and 2 are left. These have to be the bit buckets.

A bit 1 get defined by "long bucket" + "short bucket".
A bit 0 get defined by "short bucket" + "long bucket".

The longer bucket is bucket 0.
This means a nibble data "02" is bit 1 and "20" is bit 0.
With this knownledge the data can be converted to hex.

Trippersgarage, 2018-10-02:
@Portisch Tried the string you suggested, doesn't seem to work;

11:36:30 CMD: RFRaw AAA80F7F299A018402DF41018423180A821A55
11:36:30 RSL: RESULT = {"RfRaw":"ON"}
11:36:30 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:36:37 CMD: rfkey 2 1
11:36:37 RSL: RESULT = {"RfKey1":"Start learning"}
11:36:37 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:37:08 RSL: RESULT = {"RfRaw":{"Data":"AAA255"}}
11:37:17 RSL: RESULT = {"RfKey1":"Learning active"}
11:37:18 RSL: RESULT = {"RfKey1":"Learning active"}

gerardovf, 2018-10-02:
So, regarding the 'A8' calc lets make the tool more flexible.

[BitBucketConverter181001.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2437295/BitBucketConverter181001.txt)

The script will ask for 'SYNC_LOW bucket number' and 'BIT_HIGH_TIME bucket number'
Starting without options in the command line you get:
>>> 

Creating B0...

Creating A8...

A8 command: 'AA A8 0F 7F 299A 0184 02DF 41 0184 22 18 0A821A 55

B0 command: 'AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'

>>> 
Starting with debug and verbose options (-d -v) in the command line you get:
>>> 
Command Line : -d -v

Creating B0...
AA B1 05 02DF 033D 0184 01D6 299A 22020202002200220022020202020022020202002022002204 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 5
Bucket 0: 02DF (hex), 735 (dec)
Bucket 1: 033D (hex), 829 (dec)
Bucket 2: 0184 (hex), 388 (dec)
Bucket 3: 01D6 (hex), 470 (dec)
Bucket 4: 299A (hex), 10650 (dec)
szOutAux: 'AAB0xx051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'
iLength: 25 (hex), 37 (dec)
szOutFinal2Send: 'AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'

Creating A8...
szDataStr: '22020202002200220022020202020022020202002022002204'
Sync: 42    Data: 20 20 20 20 02 20 02 20 02 20 20 20 20 20 02 20 20 20 20 02 02 20 02 20 
strFinalBits: '              0  0  0  0  1  0  1  0  1  0  0  0  0  0  1  0  0  0  0  1  1  0  1  0  '
0000 (0-0h)
1010 (10-Ah)
1000 (8-8h)
0010 (2-2h)
0001 (1-1h)
1010 (10-Ah)
Hex: 0A821A
iLength: 0F (hex), 15 (dec)

The data for command 0xA8 will be:
SYNC_HIGH: bucket 4: 299A
SYNC_LOW: bucket 2: 0184
BIT_HIGH_TIME: bucket 0: 02DF
BIT_HIGH_DUTY: (100% / (bucket 4 + bucket 0)) * bucket 4: 0x41 (65%)
BIT_LOW_TIME: bucket 1: 0184
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x22 (34%)
BIT_COUNT + SYNC_BIT_COUNT: 0x18 (24, SYNC_BIT_COUNT = 0)
Data: 0A821A
7F299A018402DF41018422180A821A
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 299A 0184 02DF 41 0184 22 18 0A821A' == 0x0F

A8 command: 'AA A8 0F 7F 299A 0184 02DF 41 0184 22 18 0A821A 55

B0 command: 'AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455'

>>>

Portisch, 2018-10-02:
@Trippersgarage : does it work with the 0xB0 command?:
`AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455`
This is only the recorded button you sniffed. So I don't know if this will turn on/off the device!?

@gerardovf :looks good, but:

> BIT_HIGH_TIME: bucket 0: 02DF
> BIT_HIGH_DUTY: (100% / (bucket 4 + bucket 0)) * bucket 4: 0x41 (65%)
> BIT_LOW_TIME: bucket 1: 0184
> BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x22 (34%)

have to be:
BIT_HIGH_TIME: bucket 0: 02DF
BIT_HIGH_DUTY: (100% / (bucket **2** + bucket 0)) * bucket **0**: 0x41 (65%)
BIT_LOW_TIME: bucket **2**: 0184
BIT_LOW_DUTY: (100% / (bucket 2 + bucket **0**)) * bucket **2**: 0x22 (34%)

And the special of this sniffed data is:
bucket 0 & 1 and bucket 2 & 3 are the same. I should be just jitter of the hardware.
So the best should be:
bucket long = (bucket 0 + 1) / 2 == 782
bucket short = (bucket 2 + 3) / 2 == 429

@Trippersgarage : try this new timing also:
`AA A8 0F 7F 299A 01AD 030E 41 01AD 22 18 0A821A 55`

Trippersgarage, 2018-10-03:
@Portisch 
Tried it with the following;

5:55:51 CMD: RfRaw AAB025051402DF033D018401D6299A2202020200220022002202020202002202020200202200220455
15:55:51 RSL: RESULT = {"RfRaw":"ON"}
15:55:52 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
15:56:10 CMD: RfKey 2 1
15:56:10 RSL: RESULT = {"RfKey1":"Start learning"}
15:56:10 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
15:56:42 RSL: RESULT = `{"RfRaw":{"Data":"AAA255"}}

No difference. :(

Then tried it with the other one you suggested;

CMD: RfRaw AAA80F7F299A01AD030E4101AD22 180A821A55
16:05:04 RSL: RESULT = {"RfRaw":"ON"}
16:05:18 CMD: RfKey 2 1
16:05:18 RSL: RESULT = {"RfKey1":"Start learning"}
16:05:18 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
16:05:49 RSL: RESULT = `{"RfRaw":{"Data":"AAA255"}}

Something different happened with this; when I input the string `AAA80F7F299A01AD030E4101AD22 180A821A55` I made a mistake and left a white space and the sonoff beeped. But no other result as you can see. When I input it with no white space, no beep but same as all the previous attempts. :(

Am I even doing this correctly?

I also tried sniffing the remote again and this is what I got;

16:13:27 CMD: RfRaw AAB155
16:13:27 RSL: RESULT = {"RfRaw":"ON"}
16:13:27 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
16:13:32 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017F 02EC 299A 00101010110011001100101010101100101010101011010012 55"}}
16:14:00 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0183 02EA 2954 00101010110011001100101010101100101010101011001012 55"}}
16:14:04 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 016D 0300 299A 00101010110011001100101010101100101010101010110102 55"}}
16:14:06 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0180 02EC 2954 00101010110011001100101010101100101010101010110012 55"}}
16:14:13 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0173 0136 02F5 294A 00202020220022002200202020202200202020220200220023 55"}}
16:14:18 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 016C 02FC 29A4 00101010110011001100101010101100101010101011001102 55"}}
16:14:21 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F6 2954 00101010110011001100101010101100101010101011001102 55"}}
16:14:27 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0187 02E5 299A 00101010110011001100101010101100101010110100110012 55"}}
16:14:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0173 02F9 299A 00101010110011001100101010101100101010101011001102 55"}}

gerardovf, 2018-10-03:
For me it's strange how different the sniffed codes are. It looks like you're not keeping the remote button pressed all the time. Take a look at my tests. https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#b1-to-b0-helping-tool You will notice that every received message is just 1 second apart from the previous one. THIS IS IMPORTANT. It was my fault too after I followed @Portisch guidelines.

Trippersgarage, 2018-10-03:
@gerardovf 

I'm sure the reason why I'm having issues is my own user errors. :))

Ok, tried that. And wow.. (This is fan speed 1 button)

17:25:42 CMD: RfRaw AAB155
17:25:42 RSL: RESULT = {"RfRaw":"ON"}
17:25:42 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
17:25:45 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0182 02E7 29A4 00101010110011001100101010101100101010101011010012 55"}}
17:25:45 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0179 02F2 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:45 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0176 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0179 02F4 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0176 0140 02F4 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0173 02FA 2968 00101010110011001100101010101100101010101011010012 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017A 02F2 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0176 0140 02F7 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0175 02F8 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0175 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017A 02F2 2968 00101010110011001100101010101100101010101011010012 55"}}
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0178 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0177 02F6 2954 00101010110011001100101010101100101010101011010012 55"}}
17:25:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0177 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:48 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017A 02F3 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:49 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 017A 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:49 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0175 02F3 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:49 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0178 0140 02F4 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:49 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F8 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:50 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0175 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:50 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0175 02F8 2968 00101010110011001100101010101100101010101011010012 55"}}
17:25:50 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 017A 0140 02F3 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:50 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F8 2968 00101010110011001100101010101100101010101011010012 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0176 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F7 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0176 0140 02F3 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017A 02F1 2968 00101010110011001100101010101100101010101011010012 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 017A 0140 02F1 2954 00202020220022002200202020202200202020202022020023 55"}}
17:25:51 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0177 02F1 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:52 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0177 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:52 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0179 02F2 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:52 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0179 0140 02F2 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:52 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0177 02F6 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 017A 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0178 02F4 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0175 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0175 02F8 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0175 0140 02F3 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:53 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0177 02F1 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:54 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0175 0140 02F8 295E 00202020220022002200202020202200202020202022020023 55"}}
17:25:54 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0179 02F4 295E 00101010110011001100101010101100101010101011010012 55"}}
17:25:54 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0177 0140 02F1 295E 00202020220022002200202020202200202020202022020023 55"}}
[... cut ...]
17:29:36 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0171 02FB 295E 00101010110011001100101010101100101010101010101012 55"}}
17:29:37 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F5 295E 00101010110011001100101010101100101010101010101012 55"}}
17:29:38 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0171 02FA 2968 00101010110011001100101010101100101010101010101012 55"}}
17:29:38 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0176 02F5 2968 00101010110011001100101010101100101010101010101012 55"}}

Should I try these with the tool?

Again, thank you both for helping with this, I really owe you guys a beer!!

gerardovf, 2018-10-03:
That's perfect! I've used the tool for you.

Try:
```
RfRaw AAB0210314017102FB295E0010101011001100110010101010110010101010101010101255

RfRaw AAB0210314017A02F2295E0010101011001100110010101010110010101010101101001255
```

Good luck!

=======================================================
17:25:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 017A 02F2 295E 00101010110011001100101010101100101010101011010012 55"}}
>>> 
Command Line : -d -v
AA B1 03 017A 02F2 295E 00101010110011001100101010101100101010101011010012 55

Creating B0...
AA B1 03 017A 02F2 295E 00101010110011001100101010101100101010101011010012 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 3
Bucket 0: 017A (hex), 378 (dec)
Bucket 1: 02F2 (hex), 754 (dec)
Bucket 2: 295E (hex), 10590 (dec)
szOutAux: 'AAB0xx0314017A02F2295E0010101011001100110010101010110010101010101101001255'
iLength: 21 (hex), 33 (dec)
szOutFinal2Send: 'AAB0210314017A02F2295E0010101011001100110010101010110010101010101101001255'

B0 command: 'AAB0210314017A02F2295E0010101011001100110010101010110010101010101101001255'

>>> 

17:29:36 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 0171 02FB 295E 00101010110011001100101010101100101010101010101012 55"}}

Command Line : -d -v
AA B1 03 0171 02FB 295E 00101010110011001100101010101100101010101010101012 55

Creating B0...
AA B1 03 0171 02FB 295E 00101010110011001100101010101100101010101010101012 55
Repeat: 20 ('14' hex)
iNbrOfBuckets: 3
Bucket 0: 0171 (hex), 369 (dec)
Bucket 1: 02FB (hex), 763 (dec)
Bucket 2: 295E (hex), 10590 (dec)
szOutAux: 'AAB0xx0314017102FB295E0010101011001100110010101010110010101010101010101255'
iLength: 21 (hex), 33 (dec)
szOutFinal2Send: 'AAB0210314017102FB295E0010101011001100110010101010110010101010101010101255'

B0 command: 'AAB0210314017102FB295E0010101011001100110010101010110010101010101010101255'

Trippersgarage, 2018-10-04:
@gerardovf 

:(

11:03:37 CMD: RfRaw AAB0210314017102FB295E0010101011001100110010101010110010101010101010101255
11:03:37 RSL: RESULT = {"RfRaw":"ON"}
11:03:37 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:03:42 CMD: Rfkey 2 1
11:03:42 RSL: RESULT = {"RfKey1":"Start learning"}
11:03:42 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:04:13 RSL: RESULT = {"RfRaw":{"Data":"AAA255"}}

11:05:45 CMD: RfRaw AAB0210314017A02F2295E0010101011001100110010101010110010101010101101001255
11:05:45 RSL: RESULT = {"RfRaw":"ON"}
11:05:46 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:05:55 CMD: Rfkey 2 1
11:05:55 RSL: RESULT = {"RfKey1":"Start learning"}
11:05:55 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
11:06:26 RSL: RESULT = {"RfRaw":{"Data":"AAA255"}}

I'm beginning to think that this remote just isn't going to work...

gerardovf, 2018-10-04:
That's how I felt when everything was not working for me.

@Trippersgarage I'd recommend you to start with something that works, even if you have to borrow a remote/device from someone else. Just to find confidence with @Portisch code and Tasmota. :-)

Anyway. Why are you using "Rfkey 2 1"? It's not needed, at least in these early stages.

Trippersgarage, 2018-10-04:
@gerardovf 

You mean try sniffing a remote that already works? 

Regarding RfKey 2 1; I must be missing something. I figured after putting in the string you gave seeing if that button works. So what I’m doing is imputing the string you gave and then try using RfKey 2 1 to see if I can assign that remote button. Figured if I can get one button to work, the rest will? Am i doing this correctly?

gerardovf, 2018-10-04:
@Trippersgarage I mean using a different remote & device to see if you're more lucky...

And regarding "RfKey 2 1", it's just that I've never used that command, I don't know if I can assign a 'B0' sequence to a button. But anyway your problem is not with Tasmota but with @Portisch firmware so I understand "RfKey 2 1" doesn't help here.

Jason2866, 2018-10-05:
@Trippersgarage  You cant save Rfraw codes to a RfKey. RfKey is only for original ITEAD firmware functions.

Trippersgarage, 2018-10-12:
Just an update; finally got it working!!! YAY!! Thank you everyone for the valuable info and help. Turns out I had to wipe the chip on the bridge, reinstall tasmota and portisch. Seems to work now, thou am still figuring out how to integrate it into my setup (Pi-Homebridge-Siri-json madnesss).

Seriously thank you all!!!

closed 2018-10-12 by Portisch