Portisch/RF-Bridge-EFM8BB1 #36: Sniffing and sending to CoTech remote sockets

b1g1an, 2018-10-06:
I have some RF sockets that have all previously been working fine with an RMPro through Home Assistant. The RMPro died and I have replaced it with an RF Bridge.

The remote codes for some Status sockets I have were picked up fine using 'SetOption28 1' and worked perfectly with the freshly Tasmota'd Bridge but some CoTech ones aren't. It sniffed codes but the sockets don't do anything when sent back.

I was advised to flash this firmware onto the Bridge which I did and using RfRaw AAB155 I sniffed codes for two pairs of on and off buttons but after running these through BitBucketConverter.py, with the kind help of @gerardovf, none of them would trigger the sockets.

I did find an article online that talks about successfully setting up the same sockets, albeit I have the UK version, up for use with an Arduino if this gives any clues how to get them working with the Bridge and then HA...

https://www.instructables.com/id/Control-CoTech-Remote-Switch-With-Arduino-433Mhz/

I'm not going to pretend I really understand what's going on with this but am perfectly happy just to be told what to do and take the rest on faith so any help would be most appreciated thank you.

I've attached the codes I sniffed but failed to convert to anything useful if they give any clues.

[RF raw codes.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2453210/RF.raw.codes.txt)

gerardovf, 2018-10-07:
@b1g1an as the codes include 4 bitbuckets we need @Portisch knowledge and expertise...

Portisch, 2018-10-08:
First: I think I found a bug about the 0xA8 command.
Second: Each button do send 4 codes. So I don't know which one will work. Please try with the newest binary from the master branch!

Switch your plug A off and try these codes and check which one switch the plug on.

### A On 1:
```
AA B1 04 01E6 0428 0BE0 1C66 01101001011010011001010101101010011010011010010123 55

AA B1 04 01E6 0428 0BE0 1C66
3 (SYNC_LOW)
01 10 10 01 01 10 10 01 10 01 01 01 01 10 10 10 01 10 10 01 10 10 01 01
2 (SYNC_HIGH)
55
```
AA A8 0F 7F 0BE0 1C66 0428 44 01E6 20 18 66876C 55

### A On 2:
```
AA B1 04 01EA 0422 0BEA 1C7A 01101001010101100110010110101001010101101010010123 55

AA B1 04 01EA 0422 0BEA 1C7A 
3 (SYNC_LOW)
01 10 10 01 01 01 01 10 01 10 01 01 10 10 10 01 01 01 01 10 10 10 01 01
2 (SYNC_HIGH)
55
```
AA A8 0F 7F 0BEA 1C7A 0422 44 01EA 20 18 614E1C 55

### A On 3:
```
AA B1 04 01E5 042D 0BE0 1C84 01101001011001010110101010101010011001101010010123 55

AA B1 04 01E5 042D 0BE0 1C84
3 (SYNC_LOW)
01 10 10 01 01 10 01 01 01 10 10 10 10 10 10 10 01 10 01 10 10 10 01 01
2 (SYNC_HIGH)
55
```
AA A8 0F 7F 0BE0 1C84 042D 44 01E5 20 18 647F5C 55

### A On 4:
```
AA B1 04 01DF 042D 0BE0 1C7A 01101001010110010101100110011010100110011010010123 55

AA B1 04 01DF 042D 0BE0 1C7A
3 (SYNC_LOW)
01 10 10 01 01 01 10 01 01 01 10 01 10 01 10 10 10 01 10 01 10 10 01 01
2 (SYNC_HIGH)
55
```
AA A8 0F 7F 0BE0 1C7A 042D 44 01DF 20 18 622BAC 55

b1g1an, 2018-10-08:
@Portisch many thanks for looking at this, unfortunately none of those gave a response from the plug.

In the article I linked it says...
"testing show that not all the codes are needed. But at least 2 per button"
...and his test does send two at the same time?

Portisch, 2018-10-08:
> But at least 2 per button

Ok, I didn't check the article.
Then try first sending it by the 0xB0 command. There is no other way to send two different commands in one transmit.

> AA B0 3C 04 14 01E1 042F 0BE0 1C70 2301101001011010011001010101101010011010011010010123011010010101011001100101101010010101011010100101 55

This is one command line, send it without line breaks.
It will send "A On 1" and "A On 2" in "one shot" and repeat it 20 times.
Please test this 0xB0 command if plug A get switched on when it was off.

b1g1an, 2018-10-08:
Sorry still nothing :) I tried learning the code to the plug too as they can hold several different ones at the same time but also no luck.

Just to confirm, I do just put 'RfRaw' in front of the code don't I? Don't know if it's relevant but the RF light doesn't flash on the Bridge like it does with 'normal' codes.

Portisch, 2018-10-08:
> Just to confirm, I do just put 'RfRaw' in front of the code don't I? Don't know if it's relevant but the RF light doesn't flash on the Bridge like it does with 'normal' codes.

I don't know - I never used Tasmota firmware. But the led have to be flashing when transmit RF data. Also you should receive a acknowledge after a transmit like AAA055.

b1g1an, 2018-10-08:
Ahh, might be onto something then, if I send it with RfRaw, in the console I get...

```
13:24:03 CMD: RfRaw AA B0 3C 04 14 01E1 042F 0BE0 1C70 2301101001011010011001010101101010011010011010010123011010010101011001100101101010010101011010100101 55
13:24:03 MQT: stat/RF_Bridge/RESULT = {"RfRaw":"ON"}
```

If I send with RfCode, which is how I send the ones the Bridge could use by default, I get...

```
13:24:55 CMD: RfCode AA B0 3C 04 14 01E1 042F 0BE0 1C70 2301101001011010011001010101101010011010011010010123011010010101011001100101101010010101011010100101 55
13:24:55 MQT: stat/RF_Bridge/RESULT = {"RfCode":6345260}
```

Neither cause the led to flash.

When sending one of the working ones the led flashes and I get...

```
12:53:34 MQT: stat/RF_Bridge/RESULT = {"RfCode":6719340}
12:53:34 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

These are the commands available in Tasmota...

```
Command | Payload | Description
-- | -- | --
SetOption28 |   | Enable hex (0) or decimal (1) RF signal receive data representation
RfCode |   | Show last send 24-bit user code
RfCode | 1..8388607 | Send 24-bit user code using RfSync, RfLow and RfHigh timing
RfCode | #1..#7FFFFF | Send 24-bit hexadecimal user code using RfSync, RfLow and RfHigh timing
RfHigh |   | Show high pulse time in microseconds
RfHigh | 1 | Reset high pulse time to 840 microseconds
RfHigh | 2..32767 | Set high pulse time in microseconds
RfHigh | #2..#7FFF | Set high pulse time in hexadecimal microseconds
RfHost |   | Show 16-bit host part of user code
RfHost | 1 | Reset 16-bit host part of user code to 11802 (#2E1A)
RfHost | 2..32767 | Set 16-bit host part of user code
RfHost | #2..7FFF | Set 16-bit host part of user code in hexadecimal
RfKey<x> |   | Send learned or default RF data for RfKey1 to RfKey16
RfKey<x> | 1 | Send default RF data for RfKey1 to RfKey16 using RfSync, RfLow, RfHigh and RfHost parameters
RfKey<x> | 2 | Learn RF data for RfKey1 to RfKey16
RfKey<x> | 3 | Unlearn RF data for RfKey1 to RfKey16
RfKey<x> | 4 | Save RF data using RfSync, RfLow, RfHigh and last RfCode parameters
RfKey<x> | 5 | Show default or learnt RF data
RfLow |   | Show low pulse time in microseconds
RfLow | 1 | Reset low pulse time to 270 microseconds
RfLow | 2..32767 | Set low pulse time in microseconds
RfLow | #2..#7FFF | Set low pulse time in hexadecimal microseconds
RfSync |   | Show start sync pulse time in microseconds
RfSync | 1 | Reset start sync pulse time to 8470 microseconds
RfSync | 2..32767 | Set start sync pulse time in microseconds
RfSync | #2..#7FFF | Set start sync pulse time in hexadecimal microseconds

With EFM8BB1 RF chip firmware update support (released 20180604) the following commands become useful when the firmware has been updated by the provided Portisch firmware in folder tools/fw_efm8bb1.

Command | Payload | Description
-- | -- | --
RfRaw |   | Show current state. Off for iTead default firmware, On for Portisch firmware support
RfRaw | 0 | (Default on restart) Set iTead default firmware support and messages
RfRaw | 1 | Set Portisch firmware support and messages
RfRaw | 166 or AAA655 | (Portisch) Start sniffing/reading RF signals disabling iTead default RF handling
RfRaw | 167 or AAA755 | (Portisch) Stop sniffing/reading RF signals enabling iTead default RF handling
RfRaw | 168 or AAA855 | (Portisch) Transmiting iTead default RF protocols
RfRaw | 169 or AAA955 | (Portisch) Start sniffing and learning predefined protocols
RfRaw | 176 or AAB055 | (Portisch) Bucket Transmitting using command 0xB0
RfRaw | 177 or AAB155 | (Portisch) Start Bucket sniffing using command 0xB1
RfRaw | 192 or AAC000C055 | (Portisch) Beep - 00C0 is the length of the sound
RfRaw | 255 or AAFF55 | (Portisch) Show Rf firmware version
RfRaw | <Hexadecimal> | (Portisch) Data to be send to RF chip
```

Portisch, 2018-10-09:
What binary I do need of the Tasmota firmware to download a RF chip firmware. And where to find this function in the webinterface?

Jason2866, 2018-10-09:
@Portisch 
Wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433
Use latest Master Release or Dev build.

Portisch, 2018-10-09:
Program Version | 6.2.1
-- | --
Build Date & Time | 2018-09-09T16:48:36

But I can only upload a Tasmota firmware? Or choose just a RF chip binary?

Jason2866, 2018-10-09:
Yes just use your RF binary. It detects it :-)

Portisch, 2018-10-09:
> Upload Failed Magic byte is not 0xE9

Jason2866, 2018-10-09:
Hmm, we can chat now at https://discord.gg/Ks2Kzd4

Portisch, 2018-10-09:
@b1g1an Please try again WITHOUT spaces in the RfRaw data!

Like:

> RfRaw AAB03C041401E1042F0BE01C70230110100101101001100101010110101001101001101001012301101001010101100110010110101001010101101010010155

I receive than a AAA055 acknowledge.

Portisch, 2018-10-09:
> RfRaw AAB055041401E604280BE01C6601101001011010011001010101101010011010011010010123011010010101011001100101101010010101011010100101230110100101100101011010101010101001100110101001012355

This are now 3 of 4. I think 4 of 4 arn't possible because of too long uart data.

ErikAndren, 2018-10-09:
Is the plug in the correct state? e. g. commands turns on plug, plug must be off before.

b1g1an, 2018-10-09:
@Portisch still nothing, tried learning it too just in case but no luck there either

Would the changes you made to the binary have any effect on sniffing?

@ErikAndren yes, plug is definitely off

Jason2866, 2018-10-21:
@b1g1an Do you have tried this nice helper tool?https://gist.github.com/gerardovf/15109405e8a53dd075bbe650d70fafc6

b1g1an, 2018-10-21:
@Jason2866 I did on the original 'sniffs' before starting this issue but haven't on these, will give it a go but there is the issue that these sockets supposedly require two of the four rolling codes sent to them as above.

Portisch, 2018-10-22:
Try to combine the last 0xB1 sniffing (ON button).
This is one cycle exists of 4 commands (4 & 5 bucket codes):

> 15:09:26 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0224 03FB 0BF4 1CAC 01101001100101011010100110010101101010010110011023 55"}}
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0190 047D 0140 0A92 1C84 01101001100110101001100101100110010101010110011034 55"}}
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 021C 03EF 0BEA 1C70 01101001100101101010101010100101010110100110011023 55"}}
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 01D8 046A 013B 0A8B 1C70 01101001101010100110011010010110101010100110011034 55"}}

As you can see there is no bucket 2 in the 5 bucket codes. So It is the same like the 4 bucket codes:
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0190 047D 0140 0A92 1C84 01101001100110101001100101100110010101010110011034 55"}}

-> no bucket 2 in the data, delete it:
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0190 047D 0A92 1C84 01101001100110101001100101100110010101010110011023 55"}}

Combine the 4 bucket codes (1 & 3):

> 15:09:26 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0224 03FB 0BF4 1CAC 01101001100101011010100110010101101010010110011023 55"}}
> 15:09:27 MQT: tele/RF_Bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 021C 03EF 0BEA 1C70 01101001100101101010101010100101010110100110011023 55"}}

Should result to:

> AA B0 3C 04 14 0224 03FB 0BF4 1CAC 01101001100101011010100110010101101010010110011023 01101001100101101010101010100101010110100110011023 55

Portisch, 2018-10-22:
Try this 0xB0 command:

> AA B0 3C 04 14 0224 03FB 0BF4 1CAC 01101001100101011010100110010101101010010110011023 01101001100101101010101010100101010110100110011023 55

Portisch, 2018-11-05:
Rolling code, not enough space on the device to implement.

closed 2018-11-05 by Portisch