Portisch/RF-Bridge-EFM8BB1 #38: WFDC879 Doorman doorbell

malkstar, 2018-10-14:
Hey there,

I've been trying to add a new protocol but it doesn't seem to work with the current types of protocol.

Here is the product: https://www.responseelectronics.com/shop/door-bells-and-intercoms/wireless-door-bells/doorman-replacement-wirefree-bell-push-71199204/

Here is the bucket data I've collected and have been working against

```
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E2 0176 1B4E 01010011001010101100101012 55"}}
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0177 00E2 1B6C 10101100110101010011010102 55"}}
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0192 00CC 01C7 1B58 10101100110101010011010103 55"}}
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A6 00B1 00F0 1B44 10101100110101010011010103 55"}}
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A9 01CC 1B3A 10101100110101010011010103 55"}}
09:19:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AE 00F2 009F 1B30 10101200120202020012020203 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AA 00AA 01CC 1B58 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01A9 00A0 01CC 0104 1B4E 10103100110101010011010104 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AC 00A5 05B4 1B4E 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 00B3 01C2 1B44 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 01A4 00A9 00FB 10F4 09BA 1B4E 20101100110101010011010105 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A2 00B5 0096 1B44 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A7 01C7 1B4E 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A5 00AD 00DC 1B3A 10101100110101010011010103 55"}}
09:19:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B4 00A2 00FA 1B4E 10201100210101010011010103 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 00B0 01C2 1B44 10101100110101010011010103 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A3 00F0 1B44 20102100110101010011010103 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B1 00A7 01C2 1B4E 10101100110101010011010103 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B2 009E 01B8 0104 1B4E 10103100310101010011010104 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01BB 00AB 1AD6 10101100110101010011010102 55"}}
09:19:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A5 00B9 00FA 1B4E 10101100110101010011010103 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0196 00B8 01C3 1B3A 10101100110101010011010103 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AC 00B5 0082 1B44 10011101010101020011010103 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A5 00E6 1B12 10101100110101010011010103 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C0 009E 01C2 0109 1B58 10101100110101010011010104 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B1 0098 00FA 1B3A 10202100210101010011010103 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B7 0190 00A0 00FA 1874 30202200320202020022020204 55"}}
09:19:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01AB 00A4 1B62 10101100110101010011010102 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 038E 01B1 00AB 01C2 0096 1B4E 21212211221212121122121215 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01BA 00A6 0082 1B62 10101100110101010011010103 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 00B6 01C2 1B4E 10101100110101010011010103 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 019C 00C1 01CC 1B44 10101100110101010011010103 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01AF 009B 01CC 010E 1B58 30303100310101010011010104 55"}}
09:19:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 019C 00AC 00FA 1B44 10101100110101010011010103 55"}}
09:19:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A2 00D2 1B76 10101100110101010011010103 55"}}
09:19:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01AC 01A1 00EC 00A0 1A0E 20202300230303030022020204 55"}}
09:19:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 019D 00B7 00FA 1B44 10101100110101010011010103 55"}}
09:19:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A5 00FA 1B4E 10101100110101010011010103 55"}}
09:19:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B7 009C 00F5 1B44 20201100210101010011010103 55"}}
09:19:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AD 00AA 00FA 1B44 10101100110101010011010103 55"}}
09:19:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01BE 008F 01CE 00FA 1B44 30303100310101010031010104 55"}}
09:19:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01A5 00B2 1B4E 10101100110101010011010102 55"}}
18:15:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01C3 0090 01EA 012C 1B1C 30303100310101010031010104 55"}}
18:15:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A5 00AD 01CC 1B58 10101100110101010011010103 55"}}
18:15:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DD 017A 1B80 01010011001010101100101012 55"}}
18:15:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01A5 00B0 1B62 10101100110101010011010102 55"}}
18:15:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A5 01C2 1B4E 10101100110101010011010103 55"}}
18:15:24 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AC 00A9 01C4 1B58 10101100110101010011010103 55"}}
18:15:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AA 00B2 01C2 1B58 10101100110101010011010103 55"}}
18:15:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D5 0183 1B8A 01010011001010101100101012 55"}}
18:15:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01AF 00AD 1B58 10101100110101010011010102 55"}}
18:15:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AD 00A9 01C2 1B58 10101100110101010011010103 55"}}
18:15:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B1 00A9 01C4 1B58 10101100110101010011010103 55"}}
18:15:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B5 00A4 01C2 1B4E 10101100110101010011010103 55"}}
18:15:27 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B0 00AC 01CC 1B4E 10101100110101010011010103 55"}}
18:15:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AD 00A9 01C2 1B58 10101100110101010011010103 55"}}
18:15:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B1 00A8 01CC 1B4E 10101100110101010011010103 55"}}
18:15:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A5 01C2 1B58 10101100110101010011010103 55"}}
18:15:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00F0 00A2 1B4E 10101200120202020012020203 55"}}
18:15:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A4 0104 1B58 10101100110101010011010103 55"}}
18:15:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 02B2 01B4 009A 01D6 00FA 1996 41414211221212121122121215 55"}}
[... 84 lines cut ...]
18:21:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A7 01CC 1B4E 10101100110101010011010103 55"}}
18:21:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B5 00A3 00FA 1B4E 20201100110101010011010103 55"}}
18:21:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B0 00AC 01C2 1B58 10101100110101010011010103 55"}}
```

The protocol I tried adding

```
SI_SEGMENT_VARIABLE(DOORMAN, static DUTY_CYLCE_PROTOCOL_DATA, SI_SEG_CODE) =
	{
		250,			// SYNC_HIGH
		430,		// SYNC_LOW
		0,			// SYNC_BIT_COUNT
		460,		// BIT_HIGH_TIME
		162,		// BIT_LOW_TIME
		73,			// BIT_HIGH_DUTY
		26,			// BIT_LOW_DUTY
		0,			// BIT_COUNT
		0			// REPEAT_DELAY
	};
```

Any help in the right direction would be appreciated, I think I need to read up on RF protocol types more.

Cheers

malkstar, 2018-10-15:
Thank you, building that code didn't seem to make a difference, but I ran `rfraw AAA80E7F01901B5801904200C8210CE84055` and the data seems to come out with a more predictable result. I'm not sure whether the build I was on made a difference but will test that.

```
12:21:06 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C01010055"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C20201155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C20202155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C20101155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C30301155"}}
12:21:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C20101155"}}
12:21:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
12:21:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B5800A0012C10101155"}}
```

This is probably all I need as I am just hoping to listen for the button press and I should be able to capture that from MQTT using this data. Am I missing something as to why this wouldn't show up like `RFReceived` as normal though?

Portisch, 2018-10-15:
You **can't** use command 0xA4!
You have only 12 bit of data, not 40!

You can use 0xA8 with identifier 0x7F (universal transmit) or you define the protocol in the protocols.h and use this identifier.

malkstar, 2018-10-15:
I tested on the main release and the steps I took using your help still produce the same result, so I am guessing my `rf_protocols.h` change did not work. Just trying to work out if that is OK now.

I'm not really getting how after running the A8 command it gives better result but it does. As I say, I'm not interested in retransmitting / mimicing, I just want to receive the signal.

Thanks again for your help

malkstar, 2018-10-15:
Similar result, need to run `rfraw 177`, then the A8 command to get a result. I changed the A8 command slightly to use the new protocol this time and that worked too.

```
15:40:51 CMD: rfraw 177
15:40:51 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
15:40:52 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
15:40:58 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B9 009B 01E0 0104 1B44 30303100310101010011010104 55"}}
15:40:58 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AD 00AC 01E0 1B76 10101100110101010011010103 55"}}
15:40:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0191 0190 00CB 1B8A 20202200220202020022020203 55"}}
15:40:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D0 018E 1BA8 01010011001010101100101012 55"}}
15:40:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 0195 00B6 1B8A 20202200220202020022020203 55"}}
15:40:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AB 00B2 01C2 1B76 10101100110101010011010103 55"}}
15:40:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A7 00FA 1B6C 10101100110101010011010103 55"}}
15:41:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00AB 00FA 1B76 10101100110101010011010103 55"}}
15:41:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B5 009F 01CC 00FA 1B6C 10303100110101010011010104 55"}}
15:41:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B4 009F 01CC 00FA 1B76 30301100110101010011010104 55"}}
15:41:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00AA 01CC 1B76 10101100110101010011010103 55"}}
15:41:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A7 00FA 1B8A 10101100110101010011010103 55"}}
15:41:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B4 00A9 00FA 1B6C 10101100110101010011010103 55"}}
15:41:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B5 009F 01CC 00FA 1B76 30303100110101010011010104 55"}}
15:41:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A7 00FA 1B6C 10101100110101010011010103 55"}}
15:41:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A8 01CC 1B76 10101100110101010011010103 55"}}
15:41:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AE 00AC 00FA 1B6C 10101100110101010011010103 55"}}
15:41:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00AA 01CC 1B76 10101100110101010011010103 55"}}
15:41:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B8 00A7 00FA 1B6C 10101100110101010011010103 55"}}
15:41:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 01B5 009F 01C2 00FA 1B76 10301100110101010011010104 55"}}
15:41:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B4 00A9 01CC 1B6C 10101100110101010011010103 55"}}
15:41:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00AB 01CC 1B76 10101100110101010011010103 55"}}
15:41:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B5 00A4 01CC 1B6C 10101100110101010011010103 55"}}
15:41:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AE 01B9 00AD 1B76 20202200220202020022020203 55"}}
15:41:05 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B4 00A9 01CC 1B6C 10101100110101010011010103 55"}}
15:41:10 CMD: rfraw AAA80E0701901B5801904200C8210CE84055
15:41:10 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
15:41:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006430103155"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006401010055"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006420202155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006430103155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006420202155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410201155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410101155"}}
15:41:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA41B58001E006410201155"}}
```

Portisch, 2018-10-15:
To sniff defined protocols you have to switch to the new sniffing: 0xA6

malkstar, 2018-10-15:
Nothing appears in the console after doing `rfraw 166` and pressing the button.

I just noticed that the red led on the bridge is lighting up after doing that on the branch though.

```
16:49:45 CMD: rfraw 166
16:49:45 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
16:49:45 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

malkstar, 2018-10-15:
yep, `On branch WFDC879`, same with `AAA655`. RF led flashes again too

```18:04:03 CMD: rfraw AAA655
18:04:03 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
18:04:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

Portisch, 2018-10-16:
Looks like another bug in Tasmota.

malkstar, 2018-10-16:
Thank you @Porstich. Your support is very appreciated!

malkstar, 2018-10-16:
Tried `WFDC879` branch and `development` on Tasmota but still no luck using `rfraw 166`. Will try `rfraw AAA655` now instead. RF LED is still flashing when `0xA6` sniffing and pressing button.

Portisch, 2018-10-16:
Try this Tasmota firmware:
[sonoff_6.2.1.16_2018-10-16T125135_aec24ac9.zip](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2482694/sonoff_6.2.1.16_2018-10-16T125135_aec24ac9.zip)

I can receive 0xA6 sniffed data with this firmware:

> 00:00:20 CMD: rfraw aaa655
> 00:00:20 RSL: RESULT = {"RfRaw":"ON"}
> 00:00:20 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:02:41 RSL: RESULT = {"RfRaw":{"Data":"AAA60602D0F932111E55"}}
> 00:02:44 RSL: RESULT = {"RfRaw":{"Data":"AAA60602D0F932115555"}}

If still nothing get received I have to check what exactly happens by using 12 bit of data.
Now the software is designed to use a bit count like 8, 16, 24, 32, 40,.....
On my first look I didn't found any problem...

Portisch, 2018-10-16:
@malkstar forget this time about the Tasmota firmware! Too many bugs.
I pushed a new version: 4e76bf2ea1e700f6a83089f26b0e27553baf8453
> 00:00:15 CMD: rfraw aaa655
> 00:00:15 RSL: RESULT = {"RfRaw":"ON"}
> 00:00:15 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:00:20 RSL: RESULT = {"RfRaw":{"Data":"AAA6030717B055"}}
> 00:00:23 RSL: RESULT = {"RfRaw":{"Data":"AAA6030717B055"}}

Don't try the command 0xB1 on the console! Just try command 0xA6!

malkstar, 2018-10-16:
On master Tasmota still nothing in console.

On development tasmota and latest branch I get the following results:

```
15:40:02 CMD: rfraw aaa655
15:40:02 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
15:40:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
15:40:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60307FFF055"}}
15:40:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60307FFF055"}}
15:40:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:38 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603077FF055"}}
15:40:40 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:45 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:40:52 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60307BFF055"}}
15:41:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA60307BFF055"}}
15:41:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
15:41:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
```

I tried pressing button every one second, it appears only 1 in 5 are coming through. Result though, thank you.

Jason2866, 2018-10-16:
Rfraw 177 is fixed in Tasmota. 
Problem was introduced with commit deleting white space characters

malkstar, 2018-10-17:
Thank you, I won't be able to try it until later today but will feedback when I get chance.

It might be worth noting that last night I managed to get the event to come back more reliably by upping the TOLERANCE number from 70 - 85. I upped this by 5 each time until 9/10 clicks were coming through @ 85. Upping it to 100 got way more clicks but also a lot more repeats.

Thank you for your perseverance @Portisch!

Portisch, 2018-10-18:
Please try again to sniff with 0xB1 the doorbell. I have done some improvement on this command:
ff807c36997d39f2f7c42e99ea8ca7876993e49d

malkstar, 2018-10-18:
> @malkstar please try the new [88b3fa1](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/88b3fa12b6663d4b663e301456cf5e17d24c6add)
> @Jason2866 I am not offend but tired to fix problems what are ESP releated, not EFM8. But it is working right now! Good work! I will keep this version as backup for the future.
> 
> New Tasmota firmware:
> [6.2.1.16_2018-10-17T091249_aec24ac9.zip](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2486254/6.2.1.16_2018-10-17T091249_aec24ac9.zip)

This didn't work for me. No result in console or RF LED flashes

> Please try again to sniff with 0xB1 the doorbell. I have done some improvement on this command:
> [ff807c3](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/ff807c36997d39f2f7c42e99ea8ca7876993e49d)

Result when 0xB1 sniffing:

```
12:28:14 CMD: rfraw 177
12:28:14 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
12:28:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:28:18 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01C8 0085 010E 1B1C 22222102210101010021010103 55"}}
12:28:18 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01B6 00A4 1B4E 10101100110101010011010102 55"}}
12:28:18 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01B4 00A4 1B4E 10101100110101010011010102 55"}}
12:28:19 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01AE 00AA 00FA 1B4E 10101100110101010011010103 55"}}
12:28:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01BF 0090 0104 1B3A 20202100210101010021010103 55"}}
12:28:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01AA 00B2 1B58 10101100110101010011010102 55"}}
12:28:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D7 0180 1B76 01010011001010101100101012 55"}}
12:28:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 019C 00BB 1B6C 10101100110101010011010102 55"}}
12:28:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01B4 00A4 1B4E 10101100110101010011010102 55"}}
12:28:26 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A4 00FA 1B58 20101100110101010011010103 55"}}
12:28:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01BF 0090 0104 1B26 20202100210101010021010103 55"}}
12:28:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 00A9 1B44 10101100110101010011010102 55"}}
12:28:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E0 017C 1B80 01010011001010101100101012 55"}}
12:28:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01A1 00BB 1B6C 10101100110101010011010102 55"}}
12:28:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A4 00FA 1B4E 20101100110101010011010103 55"}}
12:28:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01B4 00A4 1B4E 10101100110101010011010102 55"}}
12:28:40 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01C8 0086 010E 1B1C 22222100210101010021010103 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B3 00A4 00FA 1B4E 10101100110101010011010103 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DA 017F 1B6C 01010011001010101100101012 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 019D 00BF 1B6C 10101100110101010011010102 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B2 00A7 00FA 1B58 10101100110101010011010103 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B1 00A2 00FA 1B58 10102100110101010011010103 55"}}
12:28:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B5 00A5 00FA 1B4E 10101100110101010011010103 55"}}
```

Portisch, 2018-10-18:
Try this new firmware:
051bfbbdc92579359b9503fbb4c70c093dfe1218

Test if data can be decoded by command 0xA6. I can't simulate it here right now because of missing hardware. The Ardunio UNO isn't able to produce a "constant" timing.

malkstar, 2018-10-21:
Nothing comes through on new branch unfortunately. Still works a lot better in 0xA6 sniffing when the tolerance is upped from 50 to 100 though, but data comes through twice as well.

This is 5 distinct button clicks:

```
12:29:25 CMD: rfraw 166
12:29:25 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
12:29:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:30:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:38 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:39 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:47 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:51 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
12:30:52 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA603073FF055"}}
```

Bucket sniffing appears to be coming out more reliably too incase it helps (same again, 5 button clicks):

```
12:32:29 CMD: rfraw 177
12:32:29 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
12:32:29 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:32:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 01A2 00B5 1B58 10101100110101010011010102 55"}}
12:32:34 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00CD 018B 1B80 01010011001010101100101012 55"}}
12:32:40 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00CC 018F 1B76 01010011001010101100101012 55"}}
12:32:46 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00C9 0190 1B6C 01010011001010101100101012 55"}}
12:32:47 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 011C 009F 01BC 1B4E 00000120012121212001212123 55"}}
12:32:51 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00CF 0189 1B76 01010011001010101100101012 55"}}
12:32:56 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01B4 009C 00FA 1B4E 20202100210101010021010103 55"}}
12:32:56 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00C7 0193 1B62 01010011001010101100101012 55"}}

Portisch, 2018-10-31:
Please try the newest release: https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x02
Try to use command 0xA6 and/or 0xB1. The firmware is now supporting inverted signals like yours.
But maybe the pulse timing have to be adjusted:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/89c66ec3882116459a9deec39f2bd32182870f9c/inc/RF_Protocols.h#L229

malkstar, 2018-11-06:
@Portisch thank you for this. The new firmware is way more reliable in 0xB1 output:

```
12:55:22 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
12:55:22 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0078 1B3A 00000100010101010001010102 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DC 017C 1B76 01010011001010101100101012 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0096 01C2 1B58 00000120012121212001212123 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0096 01C2 1B4E 00000120012121212001212123 55"}}
12:56:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 008C 01CC 1B62 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 008C 01D6 1B30 00000120012121012001010103 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00D2 0190 1B8A 01010011001010101100101012 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0096 01C2 1B4E 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 008C 01CC 1B62 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:05 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:06 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 008C 01D6 1B44 00000120012121212001212123 55"}}
12:56:06 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 0172 1B76 01010011001010101100101012 55"}}
12:56:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0096 01C2 1B4E 00000120012121212001212123 55"}}
12:56:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:07 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 0096 01C2 1B4E 00000120012121212001212123 55"}}
12:56:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0082 1B26 00000100010101010001010102 55"}}
12:56:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00FA 01C2 008C 1B58 00000210021212121002121213 55"}}
12:56:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 0172 1B76 01010011001010101100101012 55"}}
12:56:08 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:09 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:10 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E6 0172 1B76 01010011001010101100101012 55"}}
12:56:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01C2 1B58 00000120012121212001212123 55"}}
12:56:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01CC 1B58 00000120012121212001212123 55"}}
12:56:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0104 008C 01C2 1B58 00000120012121212001212123 55"}}
```

0xA6 occasionally works but less reliable than previous branch and high tolerance

Portisch, 2019-01-30:
Try sniffing again with the latest master:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch