Portisch/RF-Bridge-EFM8BB1 #41: Intertechno V3 Protocol

tcj-lev, 2018-10-17:
Hi,

is there any way to include the Intertechno V3 Protocol?
I've got eight plugs using this protocol and currently I'm switching them via a CUL.
But i would like to replace the CUL by a Sonoff RF Bridge.
The RF Bridge is already flashed with Tasmota and the current version of this RF Bridge firmware.
What should I do?
Who can help?
What do you/we need?

Portisch, 2018-10-17:
Sniffing by command 0xA6.
If no results than sniffing by 0xB1 command.
Also a link to a datasheet would be nice.

tcj-lev, 2018-10-17:
Sniffing by 0xB1 via the Tasmota console
All of the following should has been received while sending follwing 32bits by my CUL and openhab
10011010001010001101010000 00 0010
26 bit ID
2 bit command (00 = on)
4 bit channel (0-15)
`22:05:26 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0168 0A3C 00D0 04D9 00D0 2882 010302020302030302030202030302020302030203030202030302020302030003030003020203030202030302000302030203020302030302000302030300020300 55"}}
22:05:26 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 014A 0A00 0140 04E3 00B6 283C 010304040304030304030404030304040304030403030404030304040300030403030403040403030404030304040304030403040304030304000304030304040300 55"}}
22:06:12 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0163 0A32 00D2 04D7 00CF 286E 010302000302030300030202030302020302030203030202030302020302030003030003020203030202030302000302030203020302030302000302030300020300 55"}}
22:06:13 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0148 0A00 0145 04E1 00B9 2850 010304040304030304030404030304040304030403030404030304040300030003030403040403030404030304040304030403040304030304000304030304040300 55"}}
22:07:09 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0164 0A28 00D0 04DB 00C9 2878 010302000302030300030202030302020302030203030202030302020302030003030003020203030202030302000302030203020302030302000302030300020300 55"}}
22:07:10 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0146 0A00 0125 04EA 00B8 2846 010304040304030304030404030304000304030403030404030304040300030403030403040403030404030304040304030403040304030304000304030304040300 55"}}
22:07:32 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 014D 0A32 00CE 04D6 00CF 2882 010302000302030302030202030302000302030203030202030302020302030203030003020203030202030300020302030203020302030302000302030302020300 55"}}
22:07:32 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 014B 09B0 010F 04C9 00B9 2896 010304040304030304030404030304040304030403030404030304040304030403030403040403030400030300040304030403040304030300040302030304040300 55"}}
22:08:11 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0166 0A32 00CE 04DA 00CB 2878 010302020302030300030202030302020300030203030202030302020302030003030203020203030202030302000302030203020302030203020300030300020300 55"}}
22:08:12 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 015C 09D8 00CC 04CE 00CD 28C8 010302000302030302030202030302000302030203030202030302000302030203030203020203030202030300020302030203020302030302020302030302020300 55"}}
22:08:12 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0149 09B0 0159 04C8 00B8 2850 010300040304030304030404030304040304030403030404030304040304030403030403040403030404030300040304030403040304030300040304030304040300 55"}}
`
does this help?

Portisch, 2018-10-18:
All data what got recorded are 42 bits!?
I think I have to rework the 0xB1 bucket sniffing first to get correct results.

b1g1an, 2018-10-18:
@Portisch Is it worth me trying it for my problem with the Cotech plugs?

Portisch, 2018-10-18:
Every can try the newest firmware for his self.

tcj-lev, 2018-10-19:
new firmware
nearly same command - I've turned the plug on (00) and off (01) several times 

`16:28:27 CMD: rfraw 177
16:28:27 MQT: tasmota/sonoff_rf_bridge_1/stat/RESULT = {"RfRaw":"ON"}
16:28:27 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AAA055"}}
16:28:33 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0171 0A14 013F 04C3 00B4 285A 010304020304030302030404030304040304030403030204030304040304030003030003040403030404030304000304030403040304030304000304030304040305 55"}}
16:28:33 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0142 0A14 013E 04DE 00BC 285A 010304040304030304030404030304040304030403030404030304040304030403030403040403030404030300040304030403040304030304000304030304040305 55"}}
16:28:40 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0244 09C4 0171 04B9 00B4 285A 012324242324232324232424232324242324232423232424232324242324232423232223242423232424232322242324232423242324232423222324232324244325 55"}}
16:28:41 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0149 09BA 011E 04DD 00B6 28A0 010300040304030304030404030304000304030403030404030304000304030403030403040403030404030300040304030403040304030403040304030304040305 55"}}
16:28:48 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 016A 0A1E 0125 04CB 00B4 285A 010304020304030304030404030304020304030403030404030304040304030403030003040403030404030304040304030403020302030304000302030302020305 55"}}
16:28:48 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0144 09BA 0118 04D2 00B9 28A0 010304040304030304030404030304040304030403030404030304040304030403030403040403030404030300040304030403040304030304000304030304040305 55"}}
16:29:02 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0153 0A14 014A 04C4 00B5 285A 010304040304030300030404030304040304030403030004030304040304030003030003040403030404030304000304030403040304030403000300030300040305 55"}}
16:29:09 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 017C 0A14 014A 04C1 00B4 2864 010304020304030302030404030304040304030403030204030304040304030003030003040403030404030304000304030403040304030304000304030320040305 55"}}
16:29:09 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0142 0A14 013A 04F1 00B4 2850 010304040304030304030404030304040304030403030404030304040304030403030403040403030404030304040304030403040304030304000304030304040305 55"}}
16:29:10 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 017B 0A0A 04C4 00B4 2850 010203030203020203020303020203030203020302020303020203030203020302020302030302020303020203030203020302030203020302000203020203030204 55"}}
16:29:10 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 016C 0A0A 0127 04C3 00B5 2850 010304020304030304030404030304040304030403030404030304040304030403030203040403030404030304000304030403040304030403020304030304040305 55"}}
16:29:12 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 016D 0A14 013C 04CC 00BB 285A 010304020304030304030404030304020304030403030004030304040304030403030003040403030404030304000304030403040304030304000304030304040305 55"}}
16:29:12 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 024E 09BA 014A 04D0 00B6 2850 012324242324232324232424232324242324232423232424232324242324232423232423242423232424232322242324232423242324232324222324232324242325 55"}}
16:29:21 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 017B 0A14 013C 04C6 00B4 285A 010304020304030302030404030304040302030403030204030304040304030203030003040403030404030304000304030403040304030403000300030320242305 55"}}
16:29:21 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0141 0A0A 0127 04F3 00B9 2850 010304000304030304030404030304000304030403030404030304040304030403030003040403030404030304000304030403040304030403000304030304040305 55"}}
16:29:30 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 016F 0A1E 0125 04C4 00BA 285A 010304020304030304030404030304020304030403030404030304040304030403030003040403030404030300040304030403040304030302000302030302020305 55"}}
16:29:30 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0143 09BA 013A 04D2 00B7 28A0 010304040304030304030404030304040304030403030404030304000304030403030403040403030404030300040304030403040304030304040304030304040305 55"}}
16:29:38 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 017B 0A14 014A 04C3 00B4 2864 010304020304030302030404030304040304030403030204030304040304030003030003040403030404030304000304030403040304030403000300230300042305 55"}}
16:29:38 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0149 0A0A 0127 04EB 00B4 2850 010304000304030304030404030304000304030403030404030304040304030403030003040403030404030304000304030403040304030403000304030304040305 55"}}
16:29:45 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 015C 0A1E 013D 04C3 00B4 285A 010304020304030300030404030304040304030403030004030304040304030003030003040403030404030304000304030403040304030304000304030304040305 55"}}
16:29:45 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 05 0149 0A0A 04D7 00B9 2850 01020303020302020302030302020303020302030202030302020303020302030202030203030202030302020303020302030203020 [... cut ...]
16:31:17 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 016F 0A14 0122 04C4 00B9 285A 010304020304030304030404030304020304030403030404030304040304030403030003040403030404030300040304030403040304030403000302030302020305 55"}}
16:31:18 MQT: tasmota/sonoff_rf_bridge_1/tele/RESULT = {"RfRaw":{"Data":"AA B1 06 0141 09BA 011F 04E3 00BA 28AA 010304040304030304030404030304000304030403030404030304000304030403030403040403030404030300040304030403040304030403040304030304040305 55"}}`

joecool99, 2018-10-21:
I've found this description about the Intertechno Protocoll:
https://www.sweetpi.de/blog/329/ein-ueberblick-ueber-433mhz-funksteckdosen-und-deren-protokolle
Please see Section "Protokoll 3".

Maybe this is helpfull.

Portisch, 2018-10-22:
This protocol isn't supported yet. Only protocol 2 is it right now. Protocol 1 would be possible but for protocol 3 there is a lot more modification necessary. Maybe in future. I will close this issue.

closed 2018-10-22 by Portisch

tcj-lev, 2018-10-22:
sniff