Portisch/RF-Bridge-EFM8BB1 #42: Action Powerplug (Impuls)

12christiaan, 2018-10-21:
Who can help me with this device? They are powerplug sold by 'action'. 
I used to operate them by RFlink, but want to not make them work via the RF bridge Sonoff.
The remote is not recognized by the official software (no red light when I press the remote)
I flashed the Portisch/RF-Bridge-EFM8BB1 into the module.

When I put de device ‘Bucket sniffing using command 0xB1’
Via the command (Tasmota) RfRaw 177 
I get the following output. (I pressed the same remote button several times).

17:23:52 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0249 00CF 1734 01010101010101011001100110101001100110011010100110 55"}}
17:23:52 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0247 00F6 0098 17A2 01010101010101012001100120201001100110012010100110 55"}}
17:24:09 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D3 173E 01010101010101011001100110101001100110011001101010 55"}}
17:24:09 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0257 00BA 0082 1770 01010101010101011001100110101001100110011001101010 55"}}
17:24:17 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 024C 00CA 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:17 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0245 00FF 0094 1798 01010101010101012001200120102001200120012020200120 55"}}
17:24:21 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023E 00D6 1734 01010101010101011001100110101001100110011001101010 55"}}
17:24:21 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0260 0104 009B 17A2 01010101010101012001200120202001200120012001202020 55"}}
17:24:24 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0246 00D2 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:24 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0242 00FD 0097 1798 01010101010101012001200120202001200120012010200120 55"}}
17:24:27 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023D 00D6 172A 01010101010101011001100110101001100110011001101010 55"}}
17:24:27 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 025F 0103 0097 17A2 01010101010101012001200120202001200120012001202020 55"}}
17:24:29 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 024C 00CB 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:29 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0245 00C2 03DE 175C 01010101010101011001100110101001100110011010100110 55"}}
17:24:32 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023A 00D2 173E 01010101010101011001100110101001100110011001101010 55"}}
17:24:34 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 024D 00CC 173E 01010101010101011001100110101001100110011010100110 55"}}
17:24:34 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0246 00FE 0098 17A2 01010101010101012001200120201001200110012010100120 55"}}
17:24:37 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0239 00DB 1734 01010101010101011001100110101001100110011001101010 55"}}
17:24:37 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0266 00FD 0099 17A2 01010101010101012001200120202001200120012001202010 55"}}
17:24:39 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0249 00CB 173E 01010101010101011001100110101001100110011010100110 55"}}
17:24:39 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0243 00F9 009B 17A2 01010101010101012001200120202001200120012010200120 55"}}
17:24:42 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D6 173E 01010101010101011001100110101001100110011001101010 55"}}
17:24:42 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 00FA 009F 1798 01010101010101012001200120202001200120012001202020 55"}}
17:24:45 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0247 00D1 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:45 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0249 00F6 0094 17A2 01010101010101012001200120101001100120012010100110 55"}}
17:24:47 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0239 00D8 1734 01010101010101011001100110101001100110011001101010 55"}}
17:24:49 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 024B 00CB 173E 01010101010101011001100110101001100110011010100110 55"}}
17:24:49 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0244 00F7 0098 17A2 01010101010101012001200120202001200120012010100110 55"}}
17:24:51 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D3 172A 01010101010101011001100110101001100110011001101010 55"}}
17:24:51 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0266 00FF 0099 178E 01010101010101012001200120202001200120012001201020 55"}}
17:24:54 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0249 00CF 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:54 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0244 00FB 0099 17A2 01010101010101012001200120202001200120011010200110 55"}}
17:24:56 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D7 1734 01010101010101011001100110101001100110011001101010 55"}}
17:24:56 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 00FF 0098 17A2 01010101010101012001200120202001200120012001202020 55"}}
17:24:59 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0247 00D0 1734 01010101010101011001100110101001100110011010100110 55"}}
17:24:59 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 02F1 00FE 0099 178E 01010101010101012001200120102001200120012020200120 55"}}
17:25:02 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023B 00D7 173E 01010101010101011001100110101001100110011001101010 55"}}
17:25:02 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0266 00F8 0096 17A2 01010101010101012001200120201001200120012001201020 55"}}
17:25:05 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 023E 00FE 0099 1798 01010101010101012001200120202001200120012020100120 55"}}

Using the method described in several topics, I try to use the ‘BitBucketConverter181002.py’. which gave me result, but they did not work.

Then I took the old RFlink device. And send the command “10;TriState;855a2a;1;ON”
The powerplug goes on (so seam to work).
While sniffing I did the same command again (9 times) with the following result:

17:36:18 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023F 00D0 173E 01010101010101011001100110101001100110011001101010 55"}}
17:36:18 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0261 0103 009B 17A2 01010101010101012001200120202001200120012001202020 55"}}

17:36:28 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023F 00D2 172A 01010101010101011001100110101001100110011001101010 55"}}
17:36:28 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0246 00C7 1766 01010101010101011001100110101001100110011001101010 55"}}

17:36:33 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D3 1734 01010101010101011001100110101001100110011001101010 55"}}
17:36:33 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0262 0107 0097 17A2 01010101010101012001200120202001200120012001202020 55"}}

17:36:36 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023A 00D7 1734 01010101010101011001100110101001100110011001101010 55"}}
17:36:36 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0105 0264 0096 1798 10101010101010102110211021212110211021102110212120 55"}}

17:36:38 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00D4 1734 01010101010101011001100110101001100110011001101010 55"}}
17:36:38 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0247 00C5 1766 01010101010101011001100110101001100110011001101010 55"}}

17:36:40 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023B 00D6 1734 01010101010101011001100110101001100110011001101010 55"}}
17:36:40 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 04 0266 0103 0094 17A2 01010101010101012001200120202001200120012001202020 55"}}

17:36:50 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 0237 00D5 1734 01010101010101011001100110101001100110011001101010 55"}}
[... cut ...]
17:36:53 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 05 0266 0106 0097 058C 178E 01010101010101012001200120202001200120012001202020 55"}}

17:36:56 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 03 023E 00D6 1734 01010101010101011001100110101001100110011001101010 55"}}
17:36:56 MQT: tele/sonoff_RF1_B2B117/RESULT = {"RfRaw":{"Data":"AA B1 05 0269 00FD 0099 01AE 1AA4 01010101010101012001200120202001200120012001202010 55"}}

As you can see. Each time I send the command, I get 2 result in the log.

How to continue from here?

Portisch, 2018-10-22:
SYNC_HIGH: 572µs
SYNC_LOW: 5950µs
24 bit of data: 0xFF5154

Please try to send this. I don't know if this in ON or OFF. This is just the converted data from your recorded once.
> AA B0 21 03 14 023F 00D2 172A 12010101010101010110011001101010011001100110011010 55

Portisch, 2018-10-22:
Also you arn't using the latest binary firmware from the master branch! Upgrade first!
When I simulate your data I can sniff the data by the default 0xA4 command:
> 00:30:42 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00B40230FF515155"}}
> 00:31:33 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00BE0244FF515155"}}
> 00:31:41 RSL: RESULT = {"RfRaw":{"Data":"AAA4173400BE0244FF515155"}}
> 00:31:56 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00BE0244FF515155"}}
> 00:32:01 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00BE023AFF515155"}}
> 00:32:33 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00BE023AFF515155"}}
> 00:32:44 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00B4023AFF515155"}}

Sniffing by 0xB1:
> 00:33:05 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00BE 173E 01010101010101011001100110101001100110011010100112 55"}}
> 00:33:06 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0237 00F5 0087 1798 01010101010101012001200120201001200120011020100113 55"}}
> 00:33:18 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00C1 173E 01010101010101011001100110101001100110011010100112 55"}}
> 00:33:18 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0239 00F2 008A 178E 01010101010101012001200120201001200110011020100113 55"}}

Portisch, 2018-10-22:
You arn't using the latest binary from the master branch. Update first!

Default sniffing your protocol:
> 00:30:42 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00B40230FF515155"}}
> 00:31:33 RSL: RESULT = {"RfRaw":{"Data":"AAA4173E00BE0244FF515155"}}
> 00:31:41 RSL: RESULT = [...]

Sniffing by 0xB1:

> 00:32:57 CMD: rfraw aab155
> 00:32:57 RSL: RESULT = {"RfRaw":"ON"}
> 00:32:57 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
> 00:33:05 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00BE 173E 01010101010101011001100110101001100110011010100112 55"}}
> 00:33:06 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0237 00F5 0087 1798 01010101010101012001200120201001200120011020100113 55"}}
> 00:33:18 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 023C 00C1 173E 01010101010101011001100110101001100110011010100112 55"}}
> 00:33:18 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 0239 00F2 008A 178E 01010101010101012001200120201001200110011020100113 55"}}

closed 2018-10-22 by Portisch