Portisch/RF-Bridge-EFM8BB1 #43: Possible to add support for HT-12E?

andrethomas, 2018-10-26:
Hi

I got some of these chips today and was wondering if this firmware would be able to receive data from it.

https://www.farnell.com/datasheets/1899539.pdf

From logic analyzer the output looks pretty straight forward.

The 1's are indicated by short pulses whereas the one bit represented here is pulled low, resulting in a slightly longer impulse.

[image]

Perhaps its too simplistic to identify, I don't know which is why I am asking ;)

andrethomas, 2018-10-26:
Right, I figured there was a start or stop condition because there are only 12 pins that can be used for input.

I think something which may be problematic (not sure as I have never dug in your code at all) is the fact that the chip is externally biased by a resistor determined by user and this has an impact on the timing (obviously) so I am not sure how one would compensate for that in terms of decoding but to know the first post used 1M resistor and these below are 2 x 1M in parallel... i.e. ~500k?

[image]

Will attempt rfraw 0xb1 on the sonoff bridge and see what happens....

andrethomas, 2018-10-26:
Using 1mOhm resistor seems it may lead to something useful ?

```
20:02:01 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022E 0157 290E 10101010101010101010101012 55"}}
20:02:01 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022D 0157 2904 10101010101010101010101012 55"}}
20:02:02 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0227 015D 2904 10101010101010101010101012 55"}}
20:02:02 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022A 015A 2904 10101010101010101010101012 55"}}
20:02:03 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022C 0159 290E 10101010101010101010101012 55"}}
20:02:03 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0228 015D 2904 10101010101010101010101012 55"}}
20:02:04 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0229 015C 28FA 10101010101010101010101012 55"}}
20:02:04 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022D 0159 290E 10101010101010101010101012 55"}}
20:02:05 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022B 015C 2904 10101010101010101010101012 55"}}
20:02:05 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022C 0158 290E 10101010101010101010101012 55"}}
20:02:06 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022A 015E 2904 10101010101010101010101012 55"}}
20:02:06 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022B 015A 290E 10101010101010101010101012 55"}}
20:02:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022B 015C 2904 10101010101010101010101012 55"}}
20:02:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022D 015C 290E 10101010101010101010101012 55"}}
20:02:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022F 015A 2904 10101010101010101010101012 55"}}
20:02:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022D 0159 290E 10101010101010101010101012 55"}}
20:02:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 022A 015B 290E 10101010101010101010101012 55"}}
20:02:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0223 015F 290E 10101010101010101010101012 55"}}
```

Portisch, 2018-10-26:
[image]
You are right. But now the firmware make a shot of the high period and low period. And then check if bit 0 or 1.

On your chip it have to be: make a shot of low period and high period. And then check if bit 0 or 1.

I have to think about it how this can be done...

andrethomas, 2018-10-26:
If its not something you think other people will use then its no sweat... i just figured if it was easy to do why not ;) Its just as easy for me to just use a tiny45 and rcswitch lib.. its not as if this is in a device I need.. just picked up 5ps o the chips because they looked interesting.

Portisch, 2018-10-26:
I think this will solve also this issue: #38

andrethomas, 2018-10-26:
Ok, tell me which scenario's you want and I'll give you the logic analyzer's output for those?

Portisch, 2018-10-26:
Would be nice to have a plot of 0xAAA and 0x555. And measure each high and low time. Then I should be able to simulate the signal and check what's wrong with 0xB1 sniffing.

andrethomas, 2018-10-26:
Sure, give me some time I'll do it

andrethomas, 2018-10-26:
[image]

So the first one is obviously a start condition.

Then it seems it sends MSB first so 0xAAA results in so
So the logic sent is

`1 0 1 0 1 0 1 0 1 0 1 0`

```
20:56:06 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0121 028A 2918 10110011001100110011001103 55"}}
20:56:06 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 0294 2918 10110011001100110011001103 55"}}
20:56:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0262 0124 0294 2918 10110011001100110011001103 55"}}
20:56:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 0294 2918 10110011001100110011001103 55"}}
20:56:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0262 0124 0294 2922 10110011001100110011001103 55"}}
20:56:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 026A 0121 028A 2922 10110011001100110011001103 55"}}
20:56:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0267 0122 0294 2918 10110011001100110011001103 55"}}
20:56:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0260 0124 0294 2918 10110011001100110011001103 55"}}
20:56:10 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0121 028A 2918 10110011001100110011001103 55"}}
20:56:10 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0121 0294 2918 10110011001100110011001103 55"}}
20:56:11 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0261 0125 028A 290E 10110011001100110011001103 55"}}
20:56:11 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0269 0121 028A 290E 10110011001100110011001103 55"}}
20:56:12 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0268 0121 028A 2918 10110011001100110011001103 55"}}
20:56:12 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0126 028A 2918 10110011001100110011001103 55"}}
20:56:13 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0122 028A 2918 10110011001100110011001103 55"}}
20:56:13 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 028A 2918 10110011001100110011001103 55"}}
20:56:14 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0121 028A 290E 10110011001100110011001103 55"}}
20:56:15 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 025E 0127 028A 2918 10110011001100110011001103 55"}}
20:56:15 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0121 0294 2918 10110011001100110011001103 55"}}
20:56:16 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 028A 2918 10110011001100110011001103 55"}}
20:56:16 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0122 028A 290E 10110011001100110011001103 55"}}
20:56:17 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 028A 2918 10110011001100110011001103 55"}}
20:56:17 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0269 0121 028A 2918 10110011001100110011001103 55"}}
20:56:18 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 028A 2918 10110011001100110011001103 55"}}
20:56:18 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0122 028A 2918 10110011001100110011001103 55"}}
20:56:19 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0263 0126 028A 2918 10110011001100110011001103 55"}}
20:56:19 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0264 0122 0294 2922 10110011001100110011001103 55"}}
20:56:20 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0261 0128 0294 2918 10110011001100110011001103 55"}}
```

Now for the inversion we do 0x555 (0b010101010101) and again we see that the first one is some form of start marker and then short pulse = logic high, long pulse = logic low if one considers MSBit is sent first so data out would be

`0 1 0 1 0 1 0 1 0 1 0 1`

[image]
```

21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0251 0135 028A 2918 11001100110011001100110013 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 012D 00FA 0257 2922 00220022002200220022002203 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0250 0134 0285 2922 11001100110011001100110013 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0134 00FA 02 55 "}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0256 0132 028A 2918 11001100110011001100110013 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0133 00FA 0251 2922 00220022002200220022002203 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0130 0254 292C 00110011001100110011001102 55"}}
21:04:46 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0137 00FA 0252 2918 00220022002200220022002203 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0132 0256 2922 00110011001100110011001102 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0131 00FA 0253 2918 00220022002200220022002203 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 012E 02 55 "}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0253 0132 0280 2918 11001100110011001100110013 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0132 0252 2918 00110011001100110011001102 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0251 0133 0280 2918 11001100110011001100110013 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0132 00F0 0256 2922 00220022002200220022002203 55"}}
21:04:47 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0135 024F 2922 00110011001100110011001102 55"}}
[... cut ...]
21:04:50 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0137 00FA 0252 2918 00220022002200220022002203 55"}}
21:04:50 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0253 0132 0280 2918 11001100110011001100110013 55"}}
21:04:50 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0132 00FA 0257 2918 00220022002200220022002203 55"}}
21:04:50 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 024E 0137 0285 2918 11001100110011001100110013 55"}}
21:04:50 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 0253 0136 028A 2918 11001100110011001100110013 55"}}
```

andrethomas, 2018-10-26:
Nope, LSB sent first!

Sending 0b000011111111 results in

[image]

Portisch, 2018-10-26:
You decreased the speed? Now it is a low time of 2.25ms. Before it was 0.6054ms as I can read on your first screen!?
Now the 0xB1 sniffing looks ok. Maybe I have to speed up the sniffing somehow.

Also I need an idea how to detect an inverted bit diagram. "Normally" it is high followed by low. On yours its low followed by high.

andrethomas, 2018-10-26:
I think it was from 500k resistor to 1 mega ohm resistor - the highest value I have on hand.

I suppose it could be simple enough to invert the signal using a transistor but that would cause the DATA pin on the radio to be high by default which is no good for low power use.

andrethomas, 2018-10-26:
So it seems for a logic 0 the state goes high earlier than for logic 1

[image]

andrethomas, 2018-10-26:
Here's an export if you want to plot it in excel or something similar.
[export.txt](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2521016/export.txt)

andrethomas, 2018-10-26:
[image]

Could use 2 x 1M in series to get it to lower some more then maybe you won't need to increase sample time on your side.

andrethomas, 2018-10-26:
Using 2 x 1 Mega Ohm = 2M

[image]

Also much more consistent data now

```
23:16:59 CMD: rfraw 177
23:16:59 MQT: stat/bridge1/RESULT = {"RfRaw":"ON"}
23:16:59 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AAA055"}}
23:16:59 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0274 0475 5064 00101100110101010101010102 55"}}
23:17:00 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 047C 506E 00101100110101010101010102 55"}}
23:17:00 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0479 5064 00101100110101010101010102 55"}}
23:17:01 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026D 047C 506E 00101100110101010101010102 55"}}
23:17:01 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0273 0476 5064 00101100110101010101010102 55"}}
23:17:02 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0479 506E 00101100110101010101010102 55"}}
23:17:02 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0479 506E 00101100110101010101010102 55"}}
23:17:03 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:03 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:04 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:04 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 047D 5078 00101100110101010101010102 55"}}
23:17:05 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:05 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026E 047B 5078 00101100110101010101010102 55"}}
23:17:06 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0475 5078 00101100110101010101010102 55"}}
23:17:07 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:08 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:09 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:10 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0273 0475 5078 00101100110101010101010102 55"}}
23:17:10 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0475 5078 00101100110101010101010102 55"}}
23:17:11 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0271 0478 5078 00101100110101010101010102 55"}}
23:17:11 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0273 0478 5078 00101100110101010101010102 55"}}
23:17:12 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:12 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:13 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:13 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:14 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0274 0475 506E 00101100110101010101010102 55"}}
23:17:14 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 047D 506E 00101100110101010101010102 55"}}
23:17:15 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:15 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0474 5064 00101100110101010101010102 55"}}
23:17:16 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 047C 5064 00101100110101010101010102 55"}}
23:17:16 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:17 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0474 5078 00101100110101010101010102 55"}}
23:17:17 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:18 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026E 0479 506E 00101100110101010101010102 55"}}
23:17:18 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0273 0479 506E 00101100110101010101010102 55"}}
23:17:19 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5064 00101100110101010101010102 55"}}
23:17:19 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5064 00101100110101010101010102 55"}}
23:17:20 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0473 5064 00101100110101010101010102 55"}}
23:17:20 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0272 0476 506E 00101100110101010101010102 55"}}
23:17:21 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:21 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0276 0475 5078 00101100110101010101010102 55"}}
23:17:22 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 5078 00101100110101010101010102 55"}}
23:17:22 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0275 0474 506E 00101100110101010101010102 55"}}
23:17:23 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0270 0478 5078 00101100110101010101010102 55"}}
23:17:23 MQT: tele/bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 026D 047C 506E 00101100110101010101010102 55"}}
```

Portisch, 2018-10-27:
Thx for all the data! I will need to increase the sniffing because of issue #38. There is no chance to change the speed.
My first idea is to implement a new type of protocol what will handled by a negative edge. The current protocols are handled right now on a positive edge.

andrethomas, 2018-10-27:
You're welcome. I also note that I am running an older version of the firmware so Jason tells me you update some stuff recently so I guess I should reflash the EFM chip.

andrethomas, 2018-10-31:
```
17:25:36 CMD: rfraw 177
17:25:36 MQT: stat/bridge2/RESULT = {"RfRaw":"ON"}
17:25:36 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:25:36 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5000 00101101010101010101010102 55"}}
17:25:36 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 029E 04E2 501E 00101101010101010101010102 55"}}
17:25:37 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5000 00101101010101010101010102 55"}}
17:25:37 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5014 00101101010101010101010102 55"}}
17:25:38 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04D8 500A 00101101010101010101010102 55"}}
17:25:38 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04D8 500A 00101101010101010101010102 55"}}
17:25:39 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04D8 500A 00101101010101010101010102 55"}}
17:25:39 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04D8 5000 00101101010101010101010102 55"}}
17:25:40 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 029E 04E2 500A 00101101010101010101010102 55"}}
17:25:40 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 501E 00101101010101010101010102 55"}}
17:25:41 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5014 00101101010101010101010102 55"}}
17:25:41 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 501E 00101101010101010101010102 55"}}
17:25:41 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5014 00101101010101010101010102 55"}}
17:25:42 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5014 00101101010101010101010102 55"}}
17:25:42 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5014 00101101010101010101010102 55"}}
17:25:43 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 029E 04E2 501E 00101101010101010101010102 55"}}
17:25:43 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04EC 501E 00101101010101010101010102 55"}}
```

andrethomas, 2018-10-31:
Much more consistent results with 1 x 1Mohm resistor

```
17:39:07 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 00D2 02A8 28E6 01212000000000000000000003 55"}}
17:39:07 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:07 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 02A8 28DC 01212000000000000000000003 55"}}
17:39:08 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:08 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:08 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28E6 01212000000000000000000003 55"}}
17:39:08 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 02A8 28DC 01212000000000000000000003 55"}}
17:39:09 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 02A8 28DC 01212000000000000000000003 55"}}
17:39:09 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:09 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:09 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00D2 02A8 28DC 01212000000000000000000003 55"}}
17:39:10 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00D2 02A8 28E6 01212000000000000000000003 55"}}
17:39:10 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:10 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00D2 02A8 28DC 01212000000000000000000003 55"}}
17:39:11 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:11 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00DC 029E 28DC 01212000000000000000000003 55"}}
17:39:12 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 04 0186 00D2 02A8 28DC 01212000000000000000000003 55"}}
```

Portisch, 2018-10-31:
You have to find a matching value for your used frequency to use the 0xA6 decode.

`17:25:36 MQT: tele/bridge2/RESULT = {"RfRaw":{"Data":"AA B1 03 02A8 04E2 5000 00101101010101010101010102 55"}}`

The shortest bucket (0x02A8) is the pulse time. So set [pulse time](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/89c66ec3882116459a9deec39f2bd32182870f9c/inc/RF_Protocols.h#L229) to `680` and 0xA6 should work.
Maybe... Then the protocols says: pulse_time * 36 is sync low: 0x02A8 * 36 = 0x5FA0 what is much bigger than 0x5000.
I guess a pulse time between 300-400 would be the best. So adjust your osc to run with ~1666-1250Hz.

closed 2019-01-07 by Portisch