Portisch/RF-Bridge-EFM8BB1 #44: Problems with Superswitch sockets

badabing2005, 2018-10-27:
Hi,

I have this socket set named Superswitch (Based on EV1527 OTP Encoder)
[image]

Using [rc-switch](https://github.com/sui77/rc-switch) library with arduino sketch I can capture the codes, and re-transmit them without any issues (it works).
Switch 2 code for example is:
code: 15879412
protocol: 1
bits: 24
delay: 228

I want to achieve the same using Sonoff RF Bridge 433.
I have flashed the Sonoff with Tasmota v 6.2.1 and flashed EFMBB1 with the [latest](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/download/0x01/RF-Bridge-EFM8BB1.zip)

I tried to capture it with B1
```
14:31:12 CMD: RfRaw AA A9 55
14:31:12 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
14:31:30 CMD: RfRaw AAA955
14:31:30 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
14:31:51 CMD: RfRaw 177
14:31:52 MQT: stat/sonoff/RESULT = {"RfRaw":"ON"}
14:32:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAAA55"}}
14:32:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
14:32:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00F9 02A5 1B76 01101001100101101001100110100101101010100110010102 55"}}
14:32:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DF 02B2 1BA8 01101001100101101001100110100101101010100110010102 55"}}
14:32:11 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00E8 02AB 00AA 1B9E 01101001100101101001100110100101101010100110010103 55"}}
14:32:12 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DF 02B3 1BB2 01101001100101101001100110100101101010100110010102 55"}}
14:32:12 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 02AD 00C8 1B94 01101001100101101001100110100101101010100110010103 55"}}
14:32:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00ED 02A4 00AA 1B94 01121001100101101001100110100101101010100110010103 55"}}
14:32:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DE 02C2 1BB2 01101001100101101001100110100101101010100110010102 55"}}
14:32:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DB 02B7 1BC6 01101001100101101001100110100101101010100110010102 55"}}
14:32:13 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00DB 02B6 00B4 1BBC 01101001100101101001100110100101101010100110010103 55"}}
14:32:14 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00EC 02AC 00B4 1BB2 01101001100101101001100110100101101010100110010103 55"}}
14:32:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00F6 029C 1BA8 01101001100101101001100110100101101010100110010102 55"}}
14:32:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00FA 02A4 1A18 01101001100101101001100110100101101010100110010102 55"}}
14:32:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00DC 02BD 1B9E 01101001100101101001100110100101101010100110010102 55"}}
14:32:15 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 00E3 02AC 1B94 01101001100101101001100110100101101010100110010102 55"}}
14:32:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00EF 02A9 0078 1B80 01101001100101101001100110100101101010100110010103 55"}}
14:32:16 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 00D8 02BC 00D7 1BBC 01101001100101101001100110100101101010100110010103 55"}}
```

Selected the following as best fit
`"AA B1 03 00DE 02C2 1BB2 01101001100101101001100110100101101010100110010102 55"`

Converted to B0 code using [BitBucketConverter.py](https://github.com/Portisch/RF-Bridge-EFM8BB1/files/2374903/BitBucketConverter.txt)
``` python BitBucketConverter.py "AA B1 03 00DE 02C2 1BB2 01101001100101101001100110100101101010100110010102 55" 20
'RfRaw AAB021031400DE02C21BB20110100110010110100110011010010110101010011001010255'
```

Issuing `RfRaw AAB021031400DE02C21BB20110100110010110100110011010010110101010011001010255` doesn't do anything.

Looking at the code portion `0110100110010110100110011010010110101010011001010`
IT translates to binary: `100101100101001100001011`
```01 10 10 01 10 01 01 10 10 01 10 01 10 10 01 01 10 10 10 10 01 10 01 01
1  0  0  1  0  1  1  0  0  1  0  1  0  0  1  1  0  0  0  0  1  0  1  1
```

Which would be Decimal `9851659` which is no where close to what rc-switch decodes `15879412`.

It's worth noting that to make Sonoff RF bridge detect any signal from the remote I have to bring the remote right next to the bridge and stay pressed on the button several seconds before I see any code.

Using SDR the captured signal matches exactly why rc-switch decodes to.
This is a screenshot from another set I have.
[image]

Is this is timing issue? do I need to define a protocol in RF_Protocols.h ? or am I doing something wrong?

I've hit a wall and would appreciate some guidance.

Thanks in advance.
BB

Portisch, 2018-10-27:
Like in your screen it should be the buckets:
0110101001100...
Instead of the record ones:
0110100110010...

As I saw the same problem on issue #43 I think the 0xB1 is too slow. I will check how to fix this.

badabing2005, 2018-10-27:
Thank you @Portisch  for a prompt response.
I look forward to your findings / resolution.

Thanks
BB

Jason2866, 2018-10-28:
@badabing2005 
Does standard decode not work? For me Protocol 1527 works reliable with RfRaw 0
14:53:47 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":9420,"Low":310,"High":840,"Data":"A00A1F","RfKey":"None"}}

Portisch, 2018-10-28:
0xDE << 310

badabing2005, 2018-10-28:
@Jason2866 

```10:32:23 CMD: RfRaw 0
10:32:23 MQT: stat/sonoff/RESULT = {"RfRaw":"OFF"}```

Nope, Nada, the RF Led doesn't even blink
I have few other different types of 1527 based devices, none of which are detected.
They all work fine with RCSwitch

Thanks for the suggestion.

Jason2866, 2018-10-28:
Have you tried the orig. EFM Firmware too?
Because the ITEAD Pir are 1527 devices too.
For me none 1527 device has ever failed not with ITEAD or Portisch firmware.
Maybe your Bridge is faulty?

badabing2005, 2018-10-28:
I have a CT60 PIR it gets detected without issues.
I initially started with the original EFM firmware, when that didn't detect my other devices, I flashed Protisch firmware at first the one shipped with Tasmota, which is an older version, that didn't work either so I flashed the latest from here.

I won't discount my Bridge being faulty, but that doesn't explain why it works with CT60 PIR.

Portisch, 2018-10-28:
@badabing2005 can you post what you sniffed with RC Switch and what you have adjusted!?

Portisch, 2018-10-29:
No problems here:

```
00:01:06 RSL: RESULT = {"RfReceived":{"Sync":6430,"Low":220,"High":630,"Data":"747612","RfKey":"None"}}
00:01:09 RSL: RESULT = {"RfReceived":{"Sync":5930,"Low":210,"High":590,"Data":"747612","RfKey":"None"}}
00:01:21 RSL: RESULT = {"RfReceived":{"Sync":5950,"Low":200,"High":590,"Data":"747612","RfKey":"None"}}
00:01:32 RSL: RESULT = {"RfReceived":{"Sync":5920,"Low":200,"High":580,"Data":"747612","RfKey":"None"}}
00:01:43 RSL: RESULT = {"RfReceived":{"Sync":5930,"Low":200,"High":590,"Data":"747612","RfKey":"None"}
```

Or with 0xB1:
```
00:03:30 CMD: RfRaw 177
00:03:30 RSL: RESULT = {"RfRaw":"ON"}
00:03:30 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
00:03:35 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 00C7 0244 1720 01101010011001010110101001101001010101100101100102 55"}}
00:03:35 RSL: RESULT = {"RfRaw":{"Data":"AA B1 04 00F5 025C 0093 173E 01101010211021212110101021101021212121102121102123 55"}}
00:03:35 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 00BF 0248 1752 01101010011001010110101001101001010101100101100102 55"}}
00:03:46 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 00C9 0240 172A 01101010011001010110101001101001010101100101100102 55"}}
00:03:47 RSL: RESULT = {"RfRaw":{"Data":"AA B1 03 00B6 0253 1770 01101010011001010110101001101001010101100101100102 55"}}
```

```
Data:						 
Sync:
011010100110010101101010011010010101011001011001 02

Data to binary:
011010100110010101101010011010010101011001011001
0 1 1 1 0 1 0 0 0 1 1 1 0 1 1 0 0 0 0 1 0 0 1 0
Hex:
747612
```

Or 0xA6:
```
00:08:50 RSL: RESULT = {"RfRaw":"ON"}
00:08:51 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
00:09:01 RSL: RESULT = {"RfRaw":{"Data":"AAA6040174761255"}}
00:09:13 RSL: RESULT = {"RfRaw":{"Data":"AAA6040174761255"}}
```
Protocol identifier 0x01: PT2260_IDENTIFIER

[image]

Solved from my side. Feel free to reopen the issue.

closed 2018-10-29 by Portisch

Jason2866, 2018-10-29:
Your Bridge is defect because of this:

> It's worth noting that to make Sonoff RF bridge detect any signal from the remote I have to bring the remote right next to the bridge and stay pressed on the button several seconds before I see any code.

badabing2005, 2018-10-29:
Thanks @Portisch  and @Jason2866 
I appreciate all the feedback and help, I will order another RF Bridge and check with that.
Although part of me suspects a defective unit, another part says there is nothing wrong with my unit.

CT60 PIR works flawlessly, RF Bridge in the basement, the PIR two floors up the opposite side and RF Bridge has no issues detecting it consistently.

@Jason2866  My statement above was only for my Superswitch remote and other similar remotes that I have.
I also thought their might be interference, because when I put the device in B1 mode, the RF Led flickers slightly.
I moved the bridge all over the house and it does the same everywhere.

@Portisch 
Here's my code using RCSwitch, it is for [Moteino](https://lowpowerlab.com/guide/moteino/) (Arduino with RF), ignore the RFM69 part, 
As you can see it can receive a code in the following format
c:7734602,p:1,b:24,d:240
Code, Protocol, Bits, PulseLength
To transmit
Or it can decode the signal from the remote and output in the above format,
```
// **********************************************************************************
// RFMote sketch works with Moteinos equipped with RFM69W/RFM69HW/RFM69CW/RFM69HCW
// to operate remote radio controlled devices.
// This works with popular low cost power outlet sockets.
// **********************************************************************************
// Copyright BB 2016,
// **********************************************************************************
// License
// **********************************************************************************
// This program is free software; you can redistribute it
// and/or modify it under the terms of the GNU General
// Public License as published by the Free Software
// Foundation; either version 3 of the License, or
// (at your option) any later version.
//
// This program is distributed in the hope that it will
// be useful, but WITHOUT ANY WARRANTY; without even the
// implied warranty of MERCHANTABILITY or FITNESS FOR A
// PARTICULAR PURPOSE. See the GNU General Public
// License for more details.
//
// Licence can be viewed at
// http://www.gnu.org/licenses/gpl-3.0.txt
//
// Please maintain this license information along with authorship
// and copyright notices in any redistribution of this code
// **********************************************************************************
#include <RCSwitch.h>             // get it here: https://github.com/sui77/rc-switch
#include <RFM69.h>                // get it here: https://github.com/lowpowerlab/RFM69
#include <RFM69_ATC.h>            // get it here: https://github.com/lowpowerlab/RFM69
#include <RFM69_OTA.h>            // get it here: https://github.com/lowpowerlab/RFM69
#include <SPIFlash.h>             // get it here: https://github.com/lowpowerlab/spiflash
#include <SPI.h>                  // comes with Arduino
//****************************************************************************************************************
//**** IMPORTANT RADIO SETTINGS - YOU MUST CHANGE/CONFIGURE TO MATCH YOUR HARDWARE TRANSCEIVER CONFIGURATION! ****
//****************************************************************************************************************
#define GATEWAYID     2
#define NODEID        125
#define NETWORKID     150
//Match frequency to the hardware version of the radio on your Moteino (uncomment one):
//#define FREQUENCY   RF69_433MHZ
//#define FREQUENCY   RF69_868MHZ
#define FREQUENCY     RF69_915MHZ
#define IS_RFM69HW                // uncomment only for RFM69HW! Leave out if you have RFM69W!
#define ENCRYPTKEY    "MyEncryptionKey"
#define ACK_TIME      30          // max # of ms to wait for an ack
#define RFTX          6           // RF Transmitter pin = digital pin 6
#define RFRX          1           // RF Receiver on interrupt 1 => that is pin #3
#define LED           9           // Onboard LED = digital pin 9
#define MAXDECODETIME 30000       // Max time listener is on waiting to receive a signal (in millseconds)
#define SERIAL_BAUD   115200
#define SERIAL_EN                 // comment out if you don't want any serial output
[... 297 lines cut ...]
        pulseLength = mySwitchRx.getReceivedDelay();
        char buff[80];
        sprintf(buff, "RFMote:Rx_c_%lu,p_%d,b_%d,d_%d", receivedVal, protocol, bits, pulseLength);
        reportToGW(buff);
        DEBUG("  Decoded Code:"); DEBUG(receivedVal); DEBUG(" Protocol:"); DEBUG(protocol); DEBUG(" Bits: "); DEBUG(bits); DEBUG(" Delay: "); DEBUGln(pulseLength);
      }
      mySwitchRx.resetAvailable();
    }
  }
  mySwitchRx.disableReceive();
  //DEBUGln("  Exiting Listener.\n");
  digitalWrite(LED,LOW);
}

unsigned long timeSince(unsigned long timestamp)
{
  unsigned long now = millis();
  if (now >= timestamp)
    return now - timestamp;
  else return ((unsigned long)(-1)) - timestamp + now;
}
```

Thanks
BB