Portisch/RF-Bridge-EFM8BB1 #45: DSC Sensors

maxdamage2122, 2018-11-04:
This is my first time using RF sensors and I am trying to get 2 DSC sensors working. The 2 sensors I have are [EV-DW4975](http://www.dsc.com/alarm-security-products/EV-DW4975%20-%20Vanishing%20Contact/88) and [WS4904P](http://www.dsc.com/alarm-security-products/WS4904P%20-%20Wireless%20Pet-Immune%20Passive%20Infrared%20Detector/108). I have the latest tasmota on my sonoff rf bridge and successfully flashed this firmware. I haven't been able to sniff out any information from either of these devices using the RfRaw commands. Is it not possible to use these devices or am I just doing something wrong?

Portisch, 2018-11-04:
You used the latest master binary? Tried 0xB1 sniffing too?
As these are security products they are maybe not sniff able.

maxdamage2122, 2018-11-04:
I used RF-Bridge-EFM8BB1-20181102.hex from the tasmota github. I just tried RfRaw AAB155 and RfRaw AAB055 and did not receive anything from these devices.

Portisch, 2018-11-04:
Use the latest hex file from this repository and try again.
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

maxdamage2122, 2018-11-04:
Flashed that hex file and same results. I am receiving random data with 0xB1 just nothing from either of these devices.

Portisch, 2018-11-04:
You know that the devices do not send every time a signal? Check this blog: http://rctnotes.blogspot.com/2015/08/rtl433-now-decodes-dsc-433-mhz-securty.html?m=1

Also the RF bridge isn't supporting these kind of protocol. You only can use the 0xB0 and 0xB1 commands.

closed 2018-11-12 by Portisch