Portisch/RF-Bridge-EFM8BB1 #46: SilverCrest plug Rolling code

murdemon, 2018-11-04:
Hello, i scan rolling code for A-button SilverCrest plugs
it have 4 roling sequinces. Then using B1 to B0 tool. And trying send one by one. 4-5 times. And only one code for 5-8 time. Anyway no success to ON plug. What can i try more for success?

11:15:28 {"RfRaw":{"Data":"AA B1 04 03C0 01B8 0BC2 1C70 01011010101001010110101010100110011001011010101023 55"}}
11:15:28 {"RfRaw":{"Data":"AA B1 04 03D4 01CC 0924 1C52 01011010101001101010011001101001101001011010101023 55"}}
11:15:33 {"RfRaw":{"Data":"AA B1 04 0460 01A4 00D2 091A 01012020012020010120012001012001200120202020202023 55"}}
11:15:33 {"RfRaw":{"Data":"AA B1 04 03D4 01C2 0BCC 1C66 01011010011010010110011001011001100110101010101023 55"}}
11:15:37 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010011010100110100110101001100101011010101023 55"}}
11:15:38 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010101001010110101010100110011001011010101023 55"}}
11:15:42 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010101001101010011001101001101001011010101023 55"}}
11:15:44 {"RfRaw":{"Data":"AA B1 04 03CA 01B8 0BB8 1C66 01011010011010010110011001011001100110101010101023 55"}}
11:15:45 {"RfRaw":{"Data":"AA B1 04 03CA 01B8 0BC2 1C66 01011010011010100110100110101001100101011010101023 55"}}
11:15:48 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 07D0 1C66 01011010101001010110101010100110011001011010101023 55"}}
11:15:51 {"RfRaw":{"Data":"AA B1 04 03D4 01C2 0BCC 1C5C 01011010101001101010011001101001101001011010101023 55"}}
11:15:53 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010011010010110011001011001100110101010101023 55"}}
11:15:55 {"RfRaw":{"Data":"AA B1 04 03CA 01B8 0BC2 1C66 01011010011010100110100110101001100101011010101023 55"}}
11:15:57 {"RfRaw":{"Data":"AA B1 04 03C0 01C2 0BC2 1C70 01011010101001010110101010100110011001011010101023 55"}}
11:15:59 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010101001101010011001101001101001011010101023 55"}}
11:16:01 {"RfRaw":{"Data":"AA B1 04 03CA 01B8 0BC2 1C66 01011010011010010110011001011001100110101010101023 55"}}

murdemon, 2018-11-04:
i scan with SDRSharp
here what i have on RF Bridge console. 
{"RfRaw":{"Data":"AA B1 04 03DE 01CC 0BCC 1C66 01011010101001010110101010100110011001011010101023 55"}}
here what i send from RF Bridge
RfRaw AAB023041403DE01CC0BCC1C660101101010100101011010101010011001100101101010102355

in linl wav file for record in SDRSharp for signal from remote control. And for signal from RF Bridge
[https://www.dropbox.com/s/y71iuq2fel1gzrh/ButA.wav?dl=0](url)
[https://www.dropbox.com/s/9m170b4avniab16/ButA_froSNF.wav?dl=0](url)

RF data received - is same , but in RemoteContorl wav we have some extra data and then RF data repeated 6 time, in RF Bridge we have only RD data repeated 21 time.

What can be that extra data , and how can i send this too . with RF Bridge?
[https://www.dropbox.com/s/925iygq2n7dxacu/ExtraHeaderFrom%20Remote.jpg?dl=0](url)
[https://www.dropbox.com/s/1pu9aorbijk71st/RFData.jpg?dl=0](url)

murdemon, 2018-11-05:
What dose mean this times ? on thos picture can show in paint please? [https://www.dropbox.com/s/vs8q4knnibyw1ao/signalRFBridge.jpg?dl=0](url) 
AAB023040703DE01CC0BCC1C660101101010100101011010101010011001100101101010102355
0x01-0x20: Bucket 0 length: 288µs
0x01-0xC0: Bucket 1 length: 448µs
0x03-0x00: Bucket 2 length: 768µs
0x27-0x10: Bucket 3 length: 10ms (sync)
and
0x20: data len: 32 bytes
0x04: number of buckets: 4

Portisch, 2018-11-05:
If you find a single signal what will switch the plug I can implement the timing. But if more than one signal is needed it is unsupported by this firmware. There is not enough space on the device to implement a feature like this.

Here is some data from your sniffing for A on:

> 11:15:28 {"RfRaw":{"Data":"AA B1 04 03C0 01B8 0BC2 1C70 01011010101001010110101010100110011001011010101023 55"}}
> 11:15:28 {"RfRaw":{"Data":"AA B1 04 03D4 01CC 0924 1C52 01011010101001101010011001101001101001011010101023 55"}}
> 11:15:33 {"RfRaw":{"Data":"AA B1 04 03D4 01C2 0BCC 1C66 01011010011010010110011001011001100110101010101023 55"}}
> 11:15:37 {"RfRaw":{"Data":"AA B1 04 03CA 01C2 0BC2 1C66 01011010011010100110100110101001100101011010101023 55"}}

```
010110101010010101101010101001100110010110101010 23
1 1 0 0 0 0 1 1 1 0 0 0 0 0 1 0 1 0 1 1 0 0 0 0
C382B0
AAB023040A03D401CC09241C522301011010101001010110101010100110011001011010101055
```

```
010110101010011010100110011010011010010110101010 23
1 1 0 0 0 0 1 0 0 0 1 0 1 0 0 1 0 0 1 1 0 0 0 0
C22930
AAB023040A03D401CC09241C522301011010101001101010011001101001101001011010101055
```

```
010110100110100101100110010110011001101010101010 23
1 1 0 0 1 0 0 1 1 0 1 0 1 1 0 1 0 1 0 0 0 0 0 0
C9AD40
AAB023040A03D401CC09241C522301011010011010010110011001011001100110101010101055
```

```
010110100110101001101001101010011001010110101010 23
1 1 0 0 1 0 0 0 1 0 0 1 0 0 0 1 0 1 1 1 0 0 0 0
C89170
AAB023040A03D401CC09241C522301011010011010100110100110101001100101011010101055
```

Or combine: C382B0 + C22930:
`AAB03C040A03D401CC09241C52230101101010100101011010101010011001100101101010102301011010101001101010011001101001101001011010101055`

closed 2018-11-05 by Portisch