Portisch/RF-Bridge-EFM8BB1 #48: Is it possible to sniff codes from an AC123-01 transmitter ?

fabito, 2018-11-06:
Hi, 

First of all I'd like to thank for the great work you've been doing with this alternative firmware.
I have 3 RF controlled blinds here at home and I'm trying to control
 them via my Sonoff RF bridge (flashed with the latest tasmota and portisch firmwares).

The problem is that I can't sniff the transmitted codes. I tried standard, advanced and bucket sniffing but nothing is logged in the console.

The transmitters are labeled **AC123-01** (more details [here](http://www.cn-allbest.com/e_productshow/?87-AC123-01-roller-shutter-window-Channel-Insertion-Transmitter-87.html) and in the pictures below). 

[image]

[image]

Does anyone know how can I find out whether these type of transmitters are compatible with the bridge 
 or not ? Is it just a matter of [adding a new protocol](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h) ? 

Any help is appreciated  (you might have noticed that I'm a total newbie on the subject!!)

Portisch, 2018-11-06:
It should be possible to sniff by [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) like here: https://github.com/sui77/rc-switch/issues/237#issuecomment-399487895

You pushed the button a long time?

fabito, 2018-11-06:
I probably wasn't pushing/holding long enough.
After holding the `up button` for ~5 sec I got my first output :

```18:43:38 CMD: RfRaw AAB155
18:43:38 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"ON"}
18:43:38 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
18:44:47 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101010110011010011010011001101012F 55"}}```

But that was the ONLY output I got so far. I tried multiple times and for longer periods and still nothing. Is that a normal/expected behavior?

Portisch, 2018-11-06:
Normally when long pushing the button every second should be a 0xB1 received. Depends on the RF Transmitter. Please try if you can record some more samples.

fabito, 2018-11-06:
Hi @Portisch, 

After insisting a lot - multiple long and short pushes - I manage to get a few more samples (all from the up button). It was very hard to get them, still don't get how frequent the signals are being transmitted (is there a pattern at all ?)

```
19:12:17 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 023A 1360 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101100101100101010110010112F 55"}}
19:12:38 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 024E 00FA 13BA 10101010101010102F 55"}}
19:12:51 MQT: tele/home/office/rf-bridge/STATE = {"Time":"2018-11-06T19:12:51","Uptime":"0T01:55:38","Vcc":3.112,"Wifi":{"AP":1,"SSId":"toka2-office","RSSI":100,"APMac":"F8:1A:67:C5:40:5A"}}
19:13:06 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 0118 1388 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F 55"}}
19:13:07 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 026C 0118 05C8 137E 0011001101010010110100101010101100101100101100110100110101001010110101010101020110101010101010101010101001100101100101100110010103 55"}}
19:13:24 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0276 0118 137E 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F 55"}}
19:13:25 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 0104 1392 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F 55"}}
19:14:24 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0262 010E 1392 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F 55"}}
19:15:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0258 00F0 13A6 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F 55"}}
19:15:49 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101010110011010011010011001101012F 55"}}
19:17:51 MQT: tele/home/office/rf-bridge/STATE = {"Time":"2018-11-06T19:17:51","Uptime":"0T02:00:38","Vcc":3.112,"Wifi":{"AP":1,"SSId":"toka2-office","RSSI":98,"APMac":"F8:1A:67:C5:40:5A"}}
```

fabito, 2018-11-07:
I manage to find the right commands for `up`:
```
Enter B1 line: 00:37:13 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 010E 1374 00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002F 55"}}
Filtered 0xB1 data: AAB103026C010E137400110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002F55
AAB103026C010E137400110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002F55
Repeat: 20
Number of buckets: 3
Second sync bucket is in front
Sync buckets: F0
New data: F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002
Failed to decode buckets...
http://192.168.0.107/ax?c2=191&c1=RfRaw%20AA B0 4A 03 14 026C 010E 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002 55
Sending command to bridge
Resulting 0xB0 data: AA B0 4A 03 14 026C 010E 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002 55
```
And `down`:
```
Enter B1 line: 02:19:44 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 0104 1374 00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002F 55"}}
Filtered 0xB1 data: AAB103026C0104137400110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002F55
AAB103026C0104137400110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002F55
Repeat: 20
Number of buckets: 3
Second sync bucket is in front
Sync buckets: F0
New data: F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002
Failed to decode buckets...
http://192.168.0.107/ax?c2=191&c1=RfRaw%20AA B0 4A 03 14 026C 0104 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002 55
Sending command to bridge
Resulting 0xB0 data: AA B0 4A 03 14 026C 0104 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002 55
```
Still haven't find a good candidate for the `stop` command.

One thing that makes me intrigued is how difficult is to get the samples in the console.
I have to keep pushing the buttons multiple times (hundreds) and very fast to get just a handful of samples.
Has anyone seen that before ? Is there anyway to workaround it ?

Portisch, 2018-11-07:
I have done some research for this protocol. It look like it have only one sync pulse - not two. That is the reason of the "F" in the sniffed RF data. I never had some examples about a protocol like this and the firmware needs some adjustment.

You have the RF data:
`00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102F`
The 'F' on the end is only a place holder.

When the sync is only one bucket than there this data is left:  
`0011001101010010110100101010101100101100101100110100110101001010110101010101010011010101010101010101010100110010110010110011001010 <-> 2`

Converted to bit-bucket groups:  
`00 11 00 11 01 01 ...`

But this is not possible because of bucket rows '00' and '11'. Only '01' or '10' are possible.  
So move the first nibble '0' to the sync bit:  
`011001101010010110100101010101100101100101100110100110101001010110101010101010011010101010101010101010100110010110010110011001010 <-> 20`

Converted to bit-bucket groups: 
`01 10 01 10 10 10 01 01 10 10 01 01 01 01 01 10 01 01 10 01 01 10 01 10 10 01 10 10 10 01 01 01 10 10 10 10 10 10 10 01 10 10 10 10 10 10 10 10 10 10 10 10 01 10 01 01 10 01 01 10 01 10 01 01 0`

But than one bucket on the end is too less/much...

Please do a 0xB1 sniffing again with this firmware. It will mark each bucket if it was a high or low bucket.  
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/bucket_marking/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex
Maybe this will help to bring some needed information..

Portisch, 2018-11-07:
I am not sure why sniffing is os hard for this remote. I don't have aRF Bridge right now to test.
Maybe when a protocol is implemented the sniffing would be better...  
But for this I will need the new sniffed data.

fabito, 2018-11-07:
A few `up` samples collected with the **bucket_marking** firmware:

```
00:49:52 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
00:50:12 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 023A 136A 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
00:50:15 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0276 0118 1388 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:50:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 0104 137E 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:50:28 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
00:51:02 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
00:51:22 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0244 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181908181908181909090818190819092F 55"}}
00:51:25 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0212 00BE 13D8 9090909090909090AF 55"}}
00:51:27 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 026C 010E 1374 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:51:49 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 023A 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181908181908181909090818190819092F 55"}}
00:52:50 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0230 00D2 13C4 9090909090909090AF 55"}}
00:53:07 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 024E 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
00:53:34 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0226 00D2 13C4 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:54:05 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0262 0104 1388 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:54:10 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0226 00C8 13C4 9090909090909090AF 55"}}
00:54:11 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1360 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181A 55"}}
00:54:12 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1356 190819081818190908181909090909081909081909090818190819081818181818181818181A 55"}}
00:54:13 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0262 0104 137E 08190819090908181909081818181819081819081818190908190819090818190909090909090908190909090909090909090909081908181818190908181909082F 55"}}
00:54:15 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1356 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181908181908181909090818190819092F 55"}}
00:54:45 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0244 136A 190819081818190908181909090909081909081909090818190818181818181818181A 55"}}
00:54:53 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0244 00E6 13A6 9090909090909090AF 55"}}
00:55:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0082 0294 137E A1A1A1A1A1A1A1A1BF 55"}}
00:55:38 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0244 1356 19081908181819090818190909090908190908190909081819081908181909081818181818181819081818181818181818181818190819090909081819090818192F 55"}}
```

Portisch, 2018-11-08:
The 8 and 9 are 0 and 1
Just the highest bit of the byte nibble is set to 1 and this indicate a high bucket.  

8: 1000 & 0x07 == 0
9: 1001 & 0x07 == 1

Please try this firmware again:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/bucket_marking/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

Please also try the sniffing [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6) command. Maybe it can be already decoded with the last commit.
I found some more screens for this protocol here:
https://www.mikrocontroller.net/topic/451288

fabito, 2018-11-08:
Hi @Portisch , this last firmware seems to be broken. Have you tested it yourself ?

Portisch, 2018-11-08:
No, I do not have a bridge right now so I can not test it. I will check the changes...

fabito, 2018-11-08:
I went back to previous firmware and was able to sniff.
I can also retry (re flash) the new one just to be sure.

Portisch, 2018-11-08:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/5e4ca2302fc3822b953ae4099887606cc4dbd089/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

I reverted one commit, please try the new hex file again if [0xA6](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xA6) and [0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1) is working!

fabito, 2018-11-08:
HI @Portisch ,

There is something wrong with this firmware too.
Neither commands: `RfRaw AAB155` or `Rfraw AAA555` seems to be working.
Usually when I send this command I get this msg back: `{"RfRaw":{"Data":"AAA055"}}`. But with this firmware nothing is happening. Even my other RF devices don't work either.

Portisch, 2018-11-08:
Then you have to wait until my new RF ridge will arrive.

fabito, 2018-11-08:
@Portisch , thanks a lot for all your help.
I managed to sniff and convert the working codes to B0 and set up homessistant to send `RfRaw` commands for open, close and stop.
Let me know when you have a new working firmware so I can help testing this new protocol.

Portisch, 2018-11-08:
Please post your working B0 codes!

fabito, 2018-11-08:
@Portisch ,

See below the working B0 codes:

```yaml
cover:
  - platform: mqtt
    name: "Blinds 1"
    command_topic: "cmnd/home/office/rf-bridge/Backlog"
    qos: 0
    retain: true
    payload_open: "RfRaw 1; RfRaw AA B0 4A 03 14 026C 010E 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010101001100101010110100101101002 55; RfRaw 0"
    payload_close: "RfRaw 1; RfRaw AA B0 4A 03 14 026C 0104 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101001101010100101101010101001101002 55; RfRaw 0"
    payload_stop: "RfRaw 1; RfRaw AA B0 4A 03 14 0262 00F0 1374 F00110011010100101101001010101011001011001010110100110011010010110101010101010100110101010101010101010011010100101010101101001101002 55; RfRaw 0"
    optimistic: true
  - platform: mqtt
    name: "Blinds 2"
    command_topic: "cmnd/home/office/rf-bridge/Backlog"
    qos: 0
    retain: true
    payload_open: "RfRaw 1; RfRaw AA B0 4A 03 14 026C 010E 1374 F00110011010100101101001010101011001011001010110101010011010010101101010101010100110101010101010101010101001100101100110100101100102 55; RfRaw 0"
    payload_close: "RfRaw 1; RfRaw AA B0 4A 03 14 026C 010E 1374 F00110011010100101101001010101011001011001010110101010011010010101101010101010100110101010101010101001101010100101011010101001100102 55; RfRaw 0"
    payload_stop: "RfRaw 1; RfRaw AA B0 4A 03 14 0276 0110 1374 F00110011010100101101001010101011001011001010110101010011010010101101010101010100110101010101010101010011010100101100101101001100102 55; RfRaw 0"
    optimistic: true
  - platform: mqtt
    name: "Blinds 3"
    command_topic: "cmnd/home/office/rf-bridge/Backlog"
    qos: 0
    retain: true
    payload_open:  "RfRaw 1; RfRaw AA B0 4A 03 14 0276 010E 137E F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010101001100101100101100110010102 55; RfRaw 0"
    payload_close: "RfRaw 1; RfRaw AA B0 4A 03 14 026C 010E 137E F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101001101010100101011001101010010102 55; RfRaw 0"
    payload_stop: "RfRaw 1; RfRaw AA B0 4A 03 14 DIDNT FIND YET 55; RfRaw 0"
    optimistic: true
```
The only one missing is the Blinds 3  `stop` command.

fabito, 2018-11-08:
I also found this instructions. You can trigger different options by holding the buttons for 3 and 5 seconds:

[image]

Portisch, 2018-11-09:
I am still confused how this can work!? You used the last HEX file from the master branch for transmit?
Or what firmware you are using for transmit?  

Try this for Blind 3:
`payload_stop: "AA B0 4A 03 14 0276 0110 1374 F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010010102 55"`

fabito, 2018-11-09:
I'm using the latest release: [0x02](https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x02)

fabito, 2018-11-09:
The `stop` code you sent worked !  Can you explain me how you figured it out ?
I'll re flash my bridge with the master's firmware.

Portisch, 2018-11-09:
>>> The stop code you sent worked
What firmware??

```
13:50:53 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0258 137E 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:02:48 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:03:18 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 024E 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:04:15 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:04:15 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 1374 1F 55"}}
14:04:17 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:04:18 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:04:18 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 137E 1F 55"}}
14:04:18 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
14:04:19 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
20:50:43 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
20:50:43 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 1374 1F 55"}}
20:51:20 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
20:51:20 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 136A 1F 55"}}
20:51:21 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0280 00D2 1202 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010020103F 55"}}
20:51:29 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0258 137E 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
20:51:29 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 1374 1F 55"}}
20:51:34 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0258 137E 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
21:01:54 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0258 1388 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
21:02:02 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 136A 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
21:02:02 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 01 136A 1F 55"}}
21:02:14 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 023A 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
21:06:02 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 0244 1374 11001100101011010010110101010100110100110100110010110010101101010010101010101011001010101010101010101100101011010100101010101101012F 55"}}
```

These codes are all the same. Only one is inverted (buckets 0 <-> 1):
```
20:51:21 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0280 00D2 1202 00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010020103F 55"}}
```
This is matching to your other bucket mapping from the other working codes.
Remove the wrong bucket number 3:
```
00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010020102F
```
Fix the wrong bucket number in the data:
```
00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010010102F
```
Move the 'F' to the front:
```
F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010010102
```
Merge another command and this data:
```
payload_stop: "AA B0 4A 03 14 0262 00F0 1374
```
+
```
F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010010102
```
+
```
55
```
Result:
```
AA B0 4A 03 14 0262 00F0 1374 F00110011010100101101001010101011001011001011001101001101010010101101010101010100110101010101010101010011010100101011010101010010102 55
```

I will think about how to implement this as protocol. There is only a single sync bucket. I never was seeing this before.

fabito, 2018-11-09:
`master` firmware isn't working either. It's behaving similarly to the other ones. Seems like RF crashes after RfRaw is turned on.
I'll stick with 0x02 firmware for now.

BrassyPanache, 2019-01-01:
I have two devices which I wish to control. One device (fan) is captured correctly with version 2 of the firmware. It has two sync buckets. The other device (blind) only appears very rarely in the B1 capture.  I see the same behaviour with the dangling 'F' and assume it has a single sync bucket.  I can issue a command for the fan which works as expected. I cannot reproduce a correct command for the blinds.

When I use the master release I see invalid captures for both devices with 'A' and 'B' hex values in the data capture even though there are only 5 buckets.

I am using the latest Tasmota release on the Sonoff RF Bridge.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

fabito, 2019-01-31:
Hi @Portisch ,

Thanks for the new firmware.
See below some data sniffed with the new firmware:

```
03:26:07 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1374 0276 0118 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:26:10 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 0122 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:08 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 02C6 02DA 281818 55"}}
03:30:08 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 010E 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 02BC 02DA 281818 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1374 0276 010E 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 010E 023A 1360 024E 38192A 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0104 0258 0258 281818 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 023A 1360 024E 38192A 55"}}
03:30:09 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0104 024E 0262 281818 55"}}
03:30:10 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 0118 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:12 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 0122 137E 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1929292A1A192A1929 55"}}
03:30:13 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 02C6 02EE 281818 55"}}
03:30:13 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:14 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 0122 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 02C6 02E4 281818 55"}}
03:30:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:16 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 02BC 02D0 281818 55"}}
03:30:17 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 026C 010E 1374 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1929292A1A192A1929 55"}}
03:30:18 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0280 0118 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:19 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00AA 02B2 02C6 281818 55"}}
03:30:19 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1929292A1A192A1929 55"}}
03:30:21 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0280 0122 1388 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:21 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 008C 02C6 02E4 281818 55"}}
03:30:22 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A192A1A192A1A1929292A1A192A1929 55"}}
03:30:23 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 02D0 02D0 281818 55"}}
03:30:23 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:24 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00C8 0294 029E 281818 55"}}
03:30:24 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 1360 0276 010E 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:25 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0096 02BC 02E4 281818 55"}}
[... cut ...]
03:30:30 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0276 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}
03:30:32 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 02A8 02C6 281818 55"}}
03:30:32 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 136A 0280 0118 136A 38192A192A1A1A19292A1A19292929292A19292A1929292A1A192A192A1A19292A1A1A1A1A1A1A1A192A1A1A1A1A1A1A1A1A1A1A1A192A192929292A1A19292A1A19 55"}}```

fabito, 2019-01-31:
Hi @Portisch ,

Just tested all of my remotes. I was able to easily sniff, convert and send all signals.
Thanks a lot!

[offtopic] Do you know how can I enable `RfRaw 1` mode by default every time the sonoff reboots ?
I tried a rule : 

```Rule1 on system#boot do RfRaw 1 endon```

But something is setting `RfRaw 0` after the rules run. Check the init log:

```00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on /home/office/rf-bridge-3518.local with IP address 192.168.0.106
00:00:04 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/home/office/rf-bridge/LWT = Online (retained)
00:00:06 MQT: cmnd/home/office/rf-bridge/POWER = 
00:00:06 MQT: tele/home/office/rf-bridge/INFO1 = {"Module":"Sonoff Bridge","Version":"6.3.0","FallbackTopic":"DVES_B2ADBE","GroupTopic":"sonoffs"}
13:34:04 MQT: tele/home/office/rf-bridge/INFO2 = {"WebServerMode":"Admin","Hostname":"/home/office/rf-bridge-3518","IPAddress":"192.168.0.106"}
13:34:04 MQT: tele/home/office/rf-bridge/INFO3 = {"RestartReason":"Software/System restart"}
13:34:04 RUL: SYSTEM#BOOT performs "RfRaw 1"
13:34:04 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"ON"}
13:34:04 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"ON"}
13:34:04 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"ON"}
13:34:04 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"ON"}
13:34:04 MQT: tele/home/office/rf-bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
13:34:04 MQT: stat/home/office/rf-bridge/RESULT = {"RfRaw":"OFF"}```

allouchem, 2019-11-19:
Hi,

Hoping someone with success here can give me a hand. I followed this thread and managed to get the one of the rooms working; however when I tried the other rooms, I have hit a brick wall. 

Working Codes for room one ares:
`cover:
  - platform: mqtt
    name: "Office Small Shutters"
    command_topic: "cmnd/sonoffbridge/Backlog/RfRaw"
    qos: 0
    retain: true
    payload_open: "AA B0 4A 03 08 0244 00BE 1400 A08190819090908181909081819081908181819090819081909081819090818190909090909090819090909090909090909090909081908181908181819090819080 55"
    payload_close: "AA B0 4A 03 08 0244 00BE 1400 A08190819090908181909081819081908181819090819081909081819090818190909090909090819090909090909090909081909090908181819081908190819080 55"
    payload_stop: "AA B0 4A 03 08 0244 00BE 1400 A08190819090908181909081819081908181819090819081909081819090818190909090909090819090909090909090909090819090908181819090908190819080 55"
    optimistic: true`

Sniffed Data for Shutter Down in one of the other rooms below. Assuming if i can get down to work, the rest will be easy to figure out

`19T12:59:39","RfRaw":{"Data":"AA B1 03 023A 00B4 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:41 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:41","RfRaw":{"Data":"AA B1 03 0230 00D2 0258 28190819 55"}}
12:59:41 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:41","RfRaw":{"Data":"AA B1 03 023A 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:43 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:43","RfRaw":{"Data":"AA B1 03 0226 00C8 024E 28190819 55"}}
12:59:43 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:43","RfRaw":{"Data":"AA B1 03 0244 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:44 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:44","RfRaw":{"Data":"AA B1 03 0230 00D2 0258 28190819 55"}}
12:59:44 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:44","RfRaw":{"Data":"AA B1 03 023A 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:46 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:46","RfRaw":{"Data":"AA B1 03 023A 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:48 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:48","RfRaw":{"Data":"AA B1 03 0230 00D2 0258 28190819 55"}}
12:59:48 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:48","RfRaw":{"Data":"AA B1 03 0244 00BE 1414 A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:50 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:50","RfRaw":{"Data":"AA B1 03 0230 00D2 024E 28190819 55"}}
12:59:50 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:50","RfRaw":{"Data":"AA B1 03 023A 00B4 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:53 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:53","RfRaw":{"Data":"AA B1 03 023A 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:55 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:55","RfRaw":{"Data":"AA B1 03 0230 00D2 024E 28190819 55"}}
12:59:55 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:55","RfRaw":{"Data":"AA B1 03 023A 00BE 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}
12:59:57 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:57","RfRaw":{"Data":"AA B1 03 0230 00D2 0258 28190819 55"}}
12:59:57 MQT: tele/sonoffbridge/RESULT = {"Time":"2019-11-19T12:59:57","RfRaw":{"Data":"AA B1 03 023A 00B4 140A A08190819090908181909081819081908181819090819081909081819090818190909090909090908190909090909090909090819090819090819090908190819080 55"}}`

xxpigeonkillerxx, 2020-03-22:
Hi @Portisch 

I have the same type of remote AC123 and cannot sniff any signals. 

I am using the RF Bridge with the latest Tasmota and Portisch firmware. 

I am able to sniff signals from other remotes, that I guess send more basic codes. 

I'm trying to figure out how to continue troubleshooting this but have run out of options. 

Any advice/guidance would be much appreciated!

reesericci, 2020-08-02:
@Portisch it seems @Snipercaine is having some issues as well

peteakalad, 2020-08-02:
@reesericci was fixed with latest 0.4 release..

reesericci, 2020-08-02:
@peteakalad ik, we used the release from this GH and it worked. It seems the one from tasmota and it started working!

carryonrewardless, 2020-08-13:
Another +1 for the work of contributors above.  I too have a keen interest in this remote controller and will be flashing to my Sonoff RF Bridge. Thanks all.

bricefrench, 2021-05-13:
I have 5 ac123-01 remotes for roller shutters and I managed to copy the codes with the broadlink rm 4 pro easily.

carryonrewardless, 2021-05-14:
@bricefrench Did the Broadlink require no modification?

bricefrench, 2021-05-24:
Sorry for the response time.
No, no modification the broadlink rm 4 pro learns the remote control without problem.

bricefrench, 2021-05-24:
1) Broadlink rm 4 is connected to my wifi.
2) I downloaded the broadlink app from the playstore.
 I followed this tutorial: https: //www.youtube.com/watch? v = LMQP5uMrcEY.
 I managed to do different scenes. for example closing of all roller shutters at nightfall.

carryonrewardless, 2021-05-24:
I have been able to do this also.  
Less success with integrating into Home Assistant though.  It creates entities for cover, switch and remote but does not seem to communicate commands.
Nice to have the blinds controlled by the phone app at least.
Thanks

pjoterd, 2024-12-12:
Hey, I have an AC123 remote control and would like my rxb6 to receive signals from it. Is it possible to program the remote control to your own needs ? For some simple codes ? I need just such a remote control because it fits well in my hand but I can not receive the code with rxb6 
[image]
[image]
[image]