Portisch/RF-Bridge-EFM8BB1 #49: LED Control

conubo, 2018-11-06:
Hi,
will you add to control the LED / LEDs to the protocoll so i can switch the LED on and off ?
I think the Blue LED can controled by the efm8.

Portisch, 2018-11-06:
No, only the red one. The LED will be switched by RF signals.

conubo, 2018-11-06:
in this firmware hex file is a demo to flash the blue LED.
maybe that helps?

https://github.com/christophe94700/efm8-arduino-programmer/blob/master/Firmware/Blinky_Led.hex

Portisch, 2018-11-06:
Only a hardware hack will help:. 
https://www.itead.cc/wiki/images/3/3e/Sonoff_RF_Bridge_433_Schematic.SCH.pdf
The green one is power, the blue one controled by the ESP and the red one by the EFM8BB1.

conubo, 2018-11-06:
sorry, my mistake. Many thank you for the information.
With this information I can control the blue LED from the esp via GPIO13.
Like:
   pinMode (13, OUTPUT_OPEN_DRAIN);
   digitalWrite (13, LOW);
   delay (1000);
   digitalWrite (13, HIGH);
   delay (1000);

thank you

closed 2018-11-06 by conubo

Rjevski, 2022-09-24:
Am I correct in saying that the red LED is indeed controlled by this firmware? And if so, would you be open to a PR implementing this in the form of a command from the host? I might give it a try.

I've got my RF Bridge in a visible location (needs to be there for signal reception purposes) and with a hardware hack I've managed to control the green Sonoff logo LED (connecting it to GPIO12 via a 1k resistor), the blue LED is already controllable and defaults to off (unless there's a warning in ESPHome) so now only the RF LED remains.