Portisch/RF-Bridge-EFM8BB1 #50: Intertechno Code is not recognized

BigDi, 2018-11-08:
I have some IT sockets. The old one are working. But I have some new devices that are not recognized also with B1 settings. Sometimes after pressing a couple times then I get one "hit". But it is not reproduceable. At the moment I use FHEM and the SIGNALduino and this combination showes the received code and I also can send to code to my socktes. Do you know SIGNALduino? It is only a small Arduino with a receiver. The Arduino make a rudimentary decoding and real decoding works in FHEM.

Portisch, 2018-11-08:
There is something unknown:  
https://github.com/athombv/homey/issues/52
Post a oscilloscope/logic analyser screen for more information.
If you own a Arduino try this firmware: https://github.com/sui77/SimpleRcScanner or https://github.com/sui77/rc-switch

closed 2018-11-08 by Portisch

BigDi, 2018-11-08:
But SIGNALduino recognized this control.

Portisch, 2018-11-08:
I do not have the hardware.
I do not have any sniffed data.
I do not have any debug data from RCswitch or other sniffing software.
I do not have any screen of the waveform.

What should I do?

BigDi, 2018-11-08:
Is it not possible to port some parts of the code of the SIGNALduino? This one capture the code and identify it as IT Code.
https://github.com/RFD-FHEM/SIGNALDuino/blob/master/RF_Receiver/RF_Receiver.ino

BigDi, 2018-11-09:
It seems that youe code clean broke some function...

Portisch, 2018-11-11:
I know, use release 02. Use https://gist.github.com/Portisch/6dd5d6d20465521d76538a2907ba5e55 to convert to B0. Only bucket transfer can handle this protocol.