Portisch/RF-Bridge-EFM8BB1 #51: Livolo wall switch

murdemon, 2018-11-08:
Livolo switch not reading with RwRaw AAB155 ..  in link wav file from SDRSharp. It is possible to  control this switch with new firmware? 
[https://www.dropbox.com/s/tby2army7fmcceq/14-39-56%20Unknown%20433%2C950%20MHz.wav?dl=0](url)

Portisch, 2018-11-08:
I can do nothing with "SDRSharp".  
https://forum.pimatic.org/topic/781/request-protocol-debug-for-livolo-wall-switches
https://github.com/iondulgheru/CHIP_livolo/blob/master/livolo.py

When I enter the off =  '1242424352424342424242424242425342524342' in the bitconverterscript:
[image]

Sure this will not be sniffed because the longest sync is only 550µs. And the minimum is right now 1300µs.

This is the python code converted to 0xB1:
`AA B1 03 0226 006E 012F 0111111221111211111111111111112211211211 55`
And the 0xB0 command:
`AA B0 1C 03 14 0226 006E 012F 0111111221111211111111111111112211211211 55`

You can try if this will turn off your plug.

This is the on command:
`AA B1 03 0226 006E 012F 012121222121222121212121212121212122212121 55`
And the 0xB0 command:
`AA B0 1D 03 14 0226 006E 012F 012121222121222121212121212121212122212121 55`

No idea if this will work...

robertcsakany, 2018-11-11:
I've made a little python program to calculate RfRaw program for given remote and keycode. 

The gist is  [Here](https://gist.github.com/robertcsakany/3b5bc554d740237dd0f6888fea32bcc7)

Portisch, 2018-11-12:
:thumbsup:

closed 2018-11-12 by Portisch

murdemon, 2019-10-22:
Livolo 433 wall switch is working with https://github.com/robertcsakany this soft and Portisch firmware

systemiii, 2019-10-27:
When I store the second scene key in wall switch, the first scene stored doesn't work anymore.
Can you suggest me a solution?
I'm using the Csakany's python program.

robertcsakany, 2019-10-27:
@systemiii 
As I rememember there was some serious limitation in the Livolo's scene handling in the Liviolo's own firmware. Thats a reason I do not use scene I'm using 4 key knob virtual remote controller with different ID for switches.

systemiii, 2019-10-27:
Thank you, i didn't know about scenes limitation. I need to use scenes because it's the unique way to turn on (not toggle) the switch, as i know.

robertcsakany, 2019-10-27:
I think I know what you mean. I hade the very same problem with toggle, my solution was that There are code which ALL OFF  - I use that to achive OFF state and if I would like to Switch ON,  I send ALL OFF command before, after a TOGGLE.
For my Livolo clone Welaik switch I use code and as I mentioned I make dedicated RemoteID for all of the gang.

 8 - TOGGLE 
40 - ALL OFF

murdemon, 2019-11-01:
I use same with Livolo 4 but know , all off for off command from domoticz,
and toggle,  for ON command. Working more or less fine.

helenclarko, 2019-11-11:
Hey all, 

I don't mean to ask dumb questions but I'm desperately trying to get this to work.

Can anyone guide me through this?

I have the small 4 button remote and switch programmed to button A.

shalwia, 2020-05-03:
> For my Livolo clone Welaik switch I use code and as I mentioned I make dedicated RemoteID for all of the gang.
> 
> 8 - TOGGLE
> 40 - ALL OFF
Are you using some concrete Remote IDs to operate on Welaik switches or they're the same to Livolo ones?

alamakot, 2021-07-23:
Hi, how to generate livolo codes for 4 buttons remote using https://github.com/robertcsakan LivoloCodeCalc.py ? What should I type in remote id? Thanks