Portisch/RF-Bridge-EFM8BB1 #52: 0xA6 sniffing not working

Daskshadow, 2018-11-09:
Hi,
I have a RF Bridge running with Tasmota 6.3. The EFM8BB1 is running with the firmware RF-Bridge-EFM8BB1-20181102.hex from [here](https://github.com/arendst/Sonoff-Tasmota/tree/development/tools/fw_efm8bb1).
So now I wan't to detect the Protocol for HT6P20X chips. I put the RF Chip into the 0xA6 sniffing mode (since this protocol is already in the RF_Protocols.h): 
```
16:54:32 CMD: RfRaw AAA655
16:54:32 RSL: RESULT = {"RfRaw":"ON"}
16:54:32 RSL: RESULT = {"RfRaw":{"Data":"AAA055"}}
```

So now I emulated an HT6P20X remote with RCSwitch on a Digispark:
```cpp
#include <RCSwitch.h>

#define LED 1
RCSwitch mySwitch = RCSwitch();

void setup(){
  mySwitch.setProtocol(6);
  mySwitch.enableTransmit(4);
  
  pinMode(LED, OUTPUT);
  
  for(int i=0;i<3;i++){
    digitalWrite(LED, HIGH);
    delay(1000);              
    digitalWrite(LED, LOW);
    delay(1000);
  }
}

void loop(){
  mySwitch.send("100101010100101010100101");
  delay(2000);
  mySwitch.send(9783973,24);
  delay(2000);
}
```

for my understanding I should receive something like AA A6 04 06 95 4A A5 55 but instead I receive nothing. 
In 0xB1 sniffing mode I receive this: 
```
AA B1 03 01C2 03DE 2A08 01001011001100110011001011001100110011001011001102 55 
```
witch looks like the correct protocol. So why is it not recognized by the 0xA6 sniffing mode?

Portisch, 2018-11-09:
You use a old firmware. You this one: https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x02

Daskshadow, 2018-11-09:
Works, thank you!

closed 2018-11-09 by Daskshadow