Portisch/RF-Bridge-EFM8BB1 #53: Possible to integrate a Generic Protokoll from Pimatic?

hansmerkel, 2018-11-17:
I have some self designed devices for room temperature and humdity. It works for almost 2 years very fine.
But the receiver is a pimatic-homeduno based solution and it seems that pimatic will be not supported in short time.
I would like to use the Sonoff RF Bridge with tasmota as receiver but I can´t get the codes from the room devices visible in the console of tasmota on the RF Bridge.

The protocol I use is 
[https://github.com/pimatic/rfcontroljs/blob/master/lib/protocols/generic.js](url)

Is this perhaps enough information for integrating?

Portisch, 2018-11-17:
I don't care about any protocol. Only about the RF pulse timing. What RF hardware you are using? Timings?

hansmerkel, 2018-11-17:
Ok, so is it possible to receive the messages from the devices?
If I understand the wiki, I will have to send the code three times in a row to be decoded?

Portisch, 2018-11-17:
Minimum of 3 repeats, just do it repeat 10-20 times. The STX882 do not handle any protocol, it's only done by the microcontroller. Why you do not use a simpler protocol like PT2262?.
Protocols like this can be easily decoded by this firmware. If you need more than 24bits just adjust the sync high/low pattern to whatever you like and you have you own protocol:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/3165fb435250d3fe85750c21c268bb5d10816680/inc/RF_Protocols.h#L52-L84

closed 2018-11-17 by Portisch