Portisch/RF-Bridge-EFM8BB1 #54: Bridge stopped receiving codes?

petruskevicius, 2018-11-19:
Hi! After flashing my Portishch firmware, my bridge stopped showing received codes in the console. Did I do something wrong? Do I have to enable codes receiving functionality after flashing the bridge? Thanks!

Portisch, 2018-11-19:
The firmware should receive as normal the default codes: https://github.com/Portisch/RF-Bridge-EFM8BB1/releases/tag/0x02

jamman9000, 2019-01-23:
I'm having this issue. I attempted to flash using the wiki written instructions but it failed each time until I reversed the GPIO4 and 5 wires (so IO5 <-> C2CK, IO4 <-> C2D), this resulted in a successful flash of the current "generic-bucket" branch version of RF-Bridge-EFM8BB1.hex as recommended by the front Portisch information page. No operation of the S2 switch was noted in the wiki, I set it to OFF for this flash and returned it to ON after.

After reboot the RF Bridge no longer received codes from my sensors as it did before I flashed.

I found this issue and @Portisch 's link to a release that was older than the file in the generic-bucket, so I used the same wire configuration, this time did not operate the S2 switch, and successfully flashed the 10-31-2018 hex file. After reboot, still no codes received. The RF light flashes randomly, I assume receiving something but nothing on the console. I haven't watched the board as I triggered a sensor so not sure if the RF light flashes when I do that, I was in another room triggering the sensor.

I think I'll have to go back to the original firmware until you receive your new device, continue development, and this is resolved.

jamman9000, 2019-01-25:
I tried to flash back to the original firmware, here are my results based on which hex file I used:

- (portisch provided original fw) original_FW.bin ... flash fail, "Magic byte is not 0xE9"
- (portisch provided original fw) original_FW_INTEL8.bin ... flash success, no codes
- (portisch provided original fw) original_FW_INTEL16.bin ... flash success, no codes
- (generic-branch) RF-Bridge-EFM8BB1.hex ... flash success, no codes
- (master branch) RF-Bridge-EFM8BB1.hex ... flash success, no codes
- (tasmota github master branch) RF_Bridge_iTead_Original.hex
- (tasmota github...) RF_Bridge_EFM8BB1-20181023.hex ... flash fail, "Magic byte is not 0xE9"
- (tasmota github...) RF_Bridge_EFM8BB1-20181102.hex ... flash fail, "Magic byte is not 0xE9"
- (tasmota github...) RF_Bridge_EFM8BB1-20181127.hex ... flash fail, "Magic byte is not 0xE9"

So, maybe I'm bricked. I have to either send this one back for a replacement or wait for help to fix what I have here.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.
The "Magic byte is not 0xE9" is a Tasmota message. This get reported if the file isn't in Intel-Hex format.
Try to flash this new master again and try the command "AAFF55". You should receive "AA0355" and you are good!

ethologus, 2019-01-30:
> I tried to flash back to the original firmware, here are my results based on which hex file I used:
> 
> * (portisch provided original fw) original_FW.bin ... flash fail, "Magic byte is not 0xE9"
> * (portisch provided original fw) original_FW_INTEL8.bin ... flash success, no codes
> * (portisch provided original fw) original_FW_INTEL16.bin ... flash success, no codes
> * (generic-branch) RF-Bridge-EFM8BB1.hex ... flash success, no codes
> * (master branch) RF-Bridge-EFM8BB1.hex ... flash success, no codes
> * (tasmota github master branch) RF_Bridge_iTead_Original.hex ... flash fail, "Magic byte is not 0xE9"
> * (tasmota github...) RF_Bridge_EFM8BB1-20181023.hex ... flash fail, "Magic byte is not 0xE9"
> * (tasmota github...) RF_Bridge_EFM8BB1-20181102.hex ... flash fail, "Magic byte is not 0xE9"
> * (tasmota github...) RF_Bridge_EFM8BB1-20181127.hex ... flash fail, "Magic byte is not 0xE9"
> 
> So, maybe I'm bricked. I have to either send this one back for a replacement or wait for help to fix what I have here.

Hey! Most likely you haven't bricked it or somehow damaged. Actually I have been in your shoes few months ago when I flashed Portisch firmware and all of the sudden bridge stopped receiving codes. So I ordered another one and I flashed it. And guess what, it stopped receiving the codes again. I cured it by flashing Tasmota from scratch. Portisch firmware was untouched. Hope that helps.

jamman9000, 2019-01-31:
@Portisch I now understand why I received those "magic byte" errors, I right clicked on the github file links and selected "Save As" instead of copying the raw HEX data and pasting into a HEX file. I now have good files.

I saved the latest master hex file and flashed it. Triggered my sensor many times, no console logs, no codes. I then tried flashing a few older versions, no change. I then flashed tasmota again, no change. I then flashed the original iTead RF chip firmware, no change. I would love to just go back to before I flashed the EFM chip when it worked fine, but I don't think it's possible.

I don't know how to perform the sniffing process. I type 0xB1 into the console and it says it's an unknown command. I type AAFF55 into the console and it says unknown command. Pretty certain there's some information that is assumed I should know here but don't.

Portisch, 2019-01-31:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

jamman9000, 2019-01-31:
12:31:40 CMD: rfraw 1
12:31:40 MQT: RFBridge/stat/RESULT = {"RfRaw":"ON"}
12:31:51 CMD: rfraw AAFF55
12:31:51 MQT: RFBridge/stat/RESULT = {"RfRaw":"ON"}
12:32:15 CMD: rfraw 166
12:32:15 MQT: RFBridge/stat/RESULT = {"RfRaw":"ON"}
12:32:46 CMD: rfraw 255
12:32:46 MQT: RFBridge/stat/RESULT = {"RfRaw":"ON"}
12:34:03 CMD: rfcode
12:34:03 MQT: RFBridge/stat/RESULT = {"RfCode":0}
12:35:12 CMD: setoption28 0
12:35:12 MQT: RFBridge/stat/RESULT = {"SetOption28":"OFF"}
12:35:22 CMD: rfraw AAA655
12:35:22 MQT: RFBridge/stat/RESULT = {"RfRaw":"ON"}

Portisch, 2019-01-31:
Looks like you do not run the alternative firmware! Try command AAA155, if you receive some AA..55 the chip is working. If no response is coming from AAFF55 it means the alternative firmware is not running.

jamman9000, 2019-01-31:
I'm frustrated with myself. I've flashed the new firmware so many times, I don't know how it could not be on there. In my frustration I went to swap the jumpers between IO4/5 and C2CK/C2D and try again - and accidentally contacted something else. I may have pulled IO4 or 5 high, or IO4 or 5 low depending on whether I touched the 3v3 or GND. I don't know. It beeped (rebooted) and before I did any other damage I just pulled the cable out of my computer to cut power (using FTDI chip).

Now when I power it up I get two blinks of the blue LED light 1 second each, a pause, then 1 flash a second for eternity. I can't get to the tasmota Web UI, can't ping it, I have flashed tasmota 2-3 times, and tried different ways of doing it. If I hold the reset button upon boot the light won't flash at all - with C2 ON I get memory failure when I try to flash (expected), with C2 OFF I get successful flash but after it reboots it goes back to constant blinking.

I have to see if Banggood will take this back for a replacement....

jamman9000, 2019-02-12:
While trying to make a video of my "quality problem" device to send back to banggood, I found I hadn't flashed it back to the original and I found that it connected to my wifi. Ran "rfraw 177" and received back:

"RfRaw":"ON"
"RfRaw":{"Data":"AAA055"}

Tried looking at the procedure again, I started triggering my door sensors open/closed but nothing showed up in the console at the "kept pushing one of the buttons on the remote" step (bottom of tasmota rf bridge wiki).

@ethologus I've flashed it probably 3 times since, might have to connect to that UART header to log the firmware, just need a how-to

Portisch, 2019-02-19:
Waiting for replacement...

closed 2019-02-19 by Portisch

dharvind, 2022-07-01:
> While trying to make a video of my "quality problem" device to send back to banggood, I found I hadn't flashed it back to the original and I found that it connected to my wifi. Ran "rfraw 177" and [...]

hi, were you able to finally make it work?

kind of the same issue here and not able to make it work

I'm on Tasmota 8.4.0 (tested latest 12 also but when did not work tried with older versions), RF-Bridge-EFM8BB1-20190220 on a Sonoff RF Bridge V2. The responses to AAA155 and AAFF55 seem to be good, but 177 getting adhoc entries much shorter than the examples:

15:15:55 CMD: rfraw AAFF55
15:15:55 RSL: stat/tasmota_01BE6C/RESULT = {"RfRaw":"ON"}
15:15:56 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:15:56","RfRaw":{"Data":"AA0355"}}
15:15:56 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:15:56","RfRaw":{"Data":"AAA055"}}
15:16:38 CMD: rfraw AAA155
15:16:38 RSL: stat/tasmota_01BE6C/RESULT = {"RfRaw":"ON"}
15:16:38 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:16:38","RfRaw":{"Data":"AAA055"}}
15:17:09 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:17:09","RfRaw":{"Data":"AAA255"}}
15:17:25 CMD: rfraw 177
15:17:25 RSL: stat/tasmota_01BE6C/RESULT = {"RfRaw":"ON"}
15:17:25 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:17:25","RfRaw":{"Data":"AAA055"}}
15:18:01 RSL: tele/tasmota_01BE6C/RESULT = {"Time":"2022-07-01T15:18:01","RfRaw":{"Data":"AA B1 03 0078 03A2 032A 281818 55"}} //this does not appear immediately after pressing remote control button though

plantroon, 2022-08-25:
I am having this same problem on my Sonoff Bridge (R2 v1). I am testing with a wireless doorbell button.

Basically after I press the button after writing rfraw 177, the long messages either appear in ~10 seconds or more or never at all.

ESP firmware is Tasmota 12.1.1, Portisch firmware RF-Bridge-EFM8BB1-20190220.hex. 

Any chance you figured it out @dharvind ?

dharvind, 2022-08-25:
No luck :(
Tried on 2 devices, multiple tasmota versions, multiple portisch versions, still no luck. Someone experienced with portisch in a forum told me that there might be some issue within the firmware itself not allowing to capture codes. Actually the behavior above is not capturing the actual signal as this should happen near immediately, it's capturing part of some distant rf or something like this. 
On the side I am looking into other devices like broadlink rm to test. If you are able to progress do post here 👍

plantroon, 2022-08-25:
> No luck :( Tried on 2 devices, multiple tasmota versions, multiple portisch versions, still no luck. Someone experienced with portisch in a forum told me that there might be some issue within the [...]

Yea exactly this here as well. Those are not my signals most likely. Well, I ran out of things to try and besides the doorbell I don't have any other 433 MHz devices.

plantroon, 2022-08-25:
So I tried with some random old battery powered doorbell and it works flawlessly as expected. So the issue is with a battery-less doorbell which I am using currently and am trying to make work with this, but can't. Maybe its signals are "weaker" and are therefore not picked up? Even if I tried directly next to the Sonoff Bridge, there's nothing. The battery-powered doorbell works fine.

@dharvind what devices (the transmitters) did you try if I can ask?

dharvind, 2022-08-25:
I don't think it's this reason for my use case. The device I wanted to work is a curtain remote control RF433 (has a crc323 battery). But then I tried the TV remote, the AC remote (AAA batteries) but none worked.