Portisch/RF-Bridge-EFM8BB1 #55: Version 2 now crashes with some B0 raw data

haegar33, 2018-11-19:
After updating to latest V2 some of the elaborated raw codes from V0 does not work anymore and, even worse, make the RF chip hang.

Example B0 Code:
`rfraw AAB04E0514013B0A50051200D828AA01020302030203030203020302020302030302020303020302020303020302030202030203030203020203020303020302020303020302020302030203020303020455`

Only a switch off-on of the RF Bridge resolves the issue. In version 0 this code triggered the plug,

Furtermore now the B1 sniffing gives different results. Inserting the sniffed B1 lines in the python script results:

`Enter B1 line: 14:24:52 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0514 0102 55"}}
Filtered 0xB1 data: AAB103015400BE0514010255
AAB103015400BE0514010255
Repeat: 20
Number of buckets: 3
Something is wrong with the RF data!
DrawImage, try to draw a image of: 0102
Resulting 0xB0 data: AA B0 0A 03 14 0154 00BE 0514 0102 55`

Note: For this powerplug the old sniffing always resulted in 5-6 buckets, now its 3-4...
Also learning codes with A6 sniffing does not give any defined protocols.

Portisch, 2018-11-19:
I don't have any hardware to test it right now. 
Did you use the v0x02 from the releases page? When yes try this version: [increase RF sync detection](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/c01edb26ebe43c9ad1b89a8eb08cef05f1561f06)

If this isn't working better for 0xB1 try this version:
[increase speed of 0xB1 bucket sniffing](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/a37d6a98b5591d783b18af3f4060b08fb265e939)

Please post some 0xB1 data what is getting sniffed with an earlier version than v0x02.

haegar33, 2018-11-19:
Ok, I might use some older commits. Meanwhile some sniffed B1 data:

18:15:47 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 00AA 0514 0102 55"}}
18:15:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 051E 010102 55"}}
18:15:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0514 0102 55"}}
18:15:48 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0514 0102 55"}}
18:15:50 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 051E 010102 55"}}
18:15:51 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 00AA 051E 0102 55"}}
18:15:54 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0514 0102 55"}}
18:15:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 00B4 0514 0102 55"}}
18:15:59 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 051E 0102 55"}}
18:16:00 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 014A 0528 01 55"}}
18:16:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00B4 0514 0102 55"}}
18:16:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 014A 051E 01 55"}}
18:16:01 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0528 0102 55"}}
18:16:02 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 00AA 0514 10112F 55"}}
18:16:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 00AA 0514 0102 55"}}
18:16:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 00B4 0514 0102 55"}}
18:16:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 051E 0102 55"}}
18:16:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 014A 051E 01 55"}}`

Portisch, 2018-11-19:
I will need the 0xB1 sniffed data with the working version to be able to simulate the signal. The sniffed data with version 0x02 doesn't make sense...

haegar33, 2018-11-20:
Ok some results:
going back to branch "master" with commit "increased RF sync detection" changes nothing. UART still crashes after sending B0 commands. Sniffing a little different:

> 12:20:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 00AA 051E 0102 55"}}
> 12:20:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 00BE 0514 0102 55"}}
> 12:20:20 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 0154 0528 01 55"}}
> 12:20:21 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 015E 0A46 0514 00B4 2896 

> 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:20:21 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0528 0102 55"}}
> 12:20:22 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 0136 053C 01 55"}}
> 12:20:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 0528 0102 55"}}
> 12:20:28 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 0168 0A32 050A 00B4 2896 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:20:30 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 00BE 051E 0102 55"}}
> 12:20:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 014A 051E 01 55"}}
> 12:20:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 015E 0A3C 051E 00B4 28A0 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:20:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 00BE 051E 0102 55"}}
> 12:20:31 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 02 014A 0528 01 55"}}
> 12:20:33 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 015E 00B4 051E 0102 55"}}
> 

flashing back to Version 0 UART is working again. Sniffing with B1:

> 12:40:37 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 014D 0A46 0514 00C1 28B4 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:40:39 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 07 014F 09EC 0152 050F 00C7 0078 21C0 010304030403040403040304030304030404030304040304030304040304030403030403040403040303040304040304030304040304030304030403040304040306 55"}}
> 12:40:41 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 0150 0A50 0511 00C7 28B4 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:40:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 014C 0A50 0513 00C7 28A0 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030204 55"}}
> 12:40:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 07 014A 0A46 0514 00C8 02AD 006E 28AA 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030203030206 55"}}
> 12:40:47 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 06 014A 0A46 0514 00C8 02BC 28AA 010203020302030302030203020203020303020203030203020203030203020302020302030302030202030203030203020203030203020203020302030200000205 55"}}
> 

I hope this helps in this cumbersome search for new protocols. The used switches are the brand "REV Typ 8342LC". Of course I will stay for now with the old version.

Portisch, 2018-11-27:
Should be fixed with [ignore RF pulses < 100µs, increased uart RX buffer](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/10d4905feffa29679e105c0e61c524c95dbc36af)

closed 2018-11-27 by Portisch

haegar33, 2018-11-30:
???

git checkout 10d4905feffa29679e105c0e61c524c95dbc36af
fatal: reference is not a tree: 10d4905feffa29679e105c0e61c524c95dbc36af

Portisch, 2018-11-30:
Working with Webbrowser!?
I am currently working on version 3. This will support more protocols then rc-switch.

Portisch, 2018-12-06:
Please try [beat v3](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)