Portisch/RF-Bridge-EFM8BB1 #56: After update the red led blinkig fast continuously.

spider7611, 2018-11-23:
Updated the bridge EFM8BB1 chip without problem, but after restart the red led blinkig fast and never stopped. I reach the tasmota web page and do a reset but not solved the problem. the red led blinking and never stopped. Seems like always in learning mode. What can I do with this?

Sepp0815, 2018-11-23:
Hey,
I have same effect here. No idea if it's normal or not as this is my first bridge that i use...

Jason2866, 2018-11-25:
My Bridge has the same behaviour and does work well.
It shows that something is received (noise...)
Nothing do you have to fear about...

Sepp0815, 2018-11-25:
@Portisch: Wie Du richtig vermutest, aus dem deutschsprachigen Raum, genauer gesagt aus dem süddeutschen Raum. Allerdings wird hier, wenn man's genau nimmt kein "richtiges" Deutsch gesprochen...;-)

Wollte eigentlich "nur mal schnell" ein paar alte Funksteckdosen, die ich hier noch rumliegen hatte, handyfähig machen und bin, weil das Anlernen der Codes nicht ablief, wie geplant; auf Deine Firmware aufmerksam geworden. Respekt für die Arbeit und das Projekt

Portisch, 2018-11-26:
@Sepp0815 dangsche! Wer schauen das i's wida hin krieag. Schen gruas von Nähe Passau! ;)

Portisch, 2018-11-27:
Should be fixed with [fixed RF sync detection on timing mode sniffing](https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/2a533bf5a664185a24d1db0f8abb39ccda508c9b)

closed 2018-11-27 by Portisch