Portisch/RF-Bridge-EFM8BB1 #57: Delight Kinetic Battery-free Wireless Doorbell

vampywiz17, 2018-11-26:
Hello there,

I flash Tasmota and this latest radio FW. my first plan to use this doorbell:

http://www.globiz.info/index.php?page=product&product_id=55340

But unfortunatelly RF bridge can not see any code. What do you think, it possible to see it is the future? I can not do any info the doorbell (what protocol use it, etc..) i hope somebody able to check it.

Portisch, 2018-11-27:
[0xB1](https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB1)?
Currently I have to stop development of this firmware.
The soic is running out of space.

I think about to split the firmware in "normal" use and "sniffing" use...

vampywiz17, 2018-11-27:
I try this code: `rfraw aab155`

No success.

Portisch, 2018-11-27:
than maybe the sync bucket is < 1300µs!?
You have an Arduino?

vampywiz17, 2018-11-27:
No only a UART converter.

vampywiz17, 2018-12-04:
If the RF bridge not blinking (not see any RF signal) if i push the doorbell button, that mean it no reason to try to flash a newer hex file my RF chip?

So it mean that the doorbell is a different freq?

Or, it only mean that is can not decode it and i have a chance that it will work with newer hex file?

vampywiz17, 2018-12-11:
I try latest firmware (beta 3) but no success. Any possibility that i able to help you to find the problem? unfortunately no any arduino device or similar hardware...

Portisch, 2018-12-11:
> unfortunately no any arduino device or similar hardware...
not without any additional hardware

closed 2018-12-11 by Portisch