Portisch/RF-Bridge-EFM8BB1 #58: Adding protocol for (Mandolyn/Lidl TR-502MSV/RC-402/RC-402DX)

radeknoone, 2018-11-27:
Hi,

I would like to kindly ask you for adding protocol for Mandolyn/Lidl TR-502MSV/RC-402/RC-402DX. I have the TR-502MSV remote which is using it and not able to get it work with Sonoff RF Bridge. I have found the "specification" here: https://github.com/sui77/rc-switch/issues/103

There is written this:
{ 650, {  1, 10 }, {  1,  2 }, {  2,  1 }, true }      // protocol 9 (Mandolyn/Lidl TR-502MSV/RC-402/RC-402DX)

So I hope it helps to add this protocol to this project. In case it helps, I'm gonna buy you a pizza ;).

Thanks!

Radek

radeknoone, 2018-11-27:
This is sniffed data:

22:17:15 CMD: rfraw 177
22:17:15 MQT: stat/sonof_rf_bridge_01/RESULT = {"RfRaw":"ON"}
22:17:15 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AAA055"}}
22:17:28 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 033E 01AE 05BE 1FEE B1AC 012121212121212121210000000000000000000003012121212121212120212101000001000000000103012121212121212120212101200001000000000003012121212121212120212101000001000000000104 55"}}
22:17:32 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 0334 01AE 05C8 1FEE B166 012121212121212121212020200000000000200003012121212121212120212020200000000020202023012121212121212120202020200000000020202023012121212121212120212020200000000020200024 55"}}
22:17:39 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 033E 01AE 05BE 1FEE B1AC 012121212121212121210000000000000000000003012121212121212121212101000001000000000103012121212121212121212101000001000000000003012121212121212120212101000001000000000104 55"}}
22:18:01 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 0334 01AE 05BE 1FE4 B1B6 012121212121212121212020200000000000000023012121212121212120202020200000000020000023012121212121212120202020200000000020200023012121212121212120202120200000000020200024 55"}}
22:18:09 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 0334 01AE 05BE 1FE4 B1AC 012121212121212121212020200000000000000023012121212121212120202020200000000000000023012121212121212120202121200000000000000023012121212121212120212020200000000000000024 55"}}
22:18:32 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 05 0334 01AE 05BE 1FE4 B1B6 012121212121212121212020200000000000020003012121212121212120212020200000000020000023012121212121212120212020200000000000000023012121212121212120212120200000000020000024 55"}}

radeknoone, 2018-11-27:
And litlebit more information from http://www.dbzoo.com/livebox/universalrf?s[]=710#lidl_rf_socket

pulsedef=020320028A05DC028A05DC0514032004B0
burstcount=12
interburstdelay=80000
frames=21
stream=15B15556160

radeknoone, 2018-11-28:
And thats an input from BB Converter:

```
radek@saphir:~$ python BitBucketConverter.py -v -d
Enter B1 line: 21:44:41 MQT: tele/sonof_rf_bridge_01/RESULT = {"RfRaw":{"Data":"AA B1 04 033E 01AE 05BE B198 012121212121212121210000000000000000000003012121212121212121210121000100000000000003012121212121212120212101000101000000000003012121212121212121212121000100000000000004 55"}}
AA B1 04 033E 01AE 05BE B198 012121212121212121210000000000000000000003012121212121212121210121000100000000000003012121212121212120212101000101000000000003012121212121212121212121000100000000000004 55
Repeat: 20
Sync: 40    Data: 12 12 12 12 12 12 12 12 12 10 00 00 00 00 00 00 00 00 00 00 30 12 12 12 12 12 12 12 12 12 10 12 10 00 10 00 00 00 00 00 00 30 12 12 12 12 12 12 12 12 02 12 10 10 00 10 10 00 00 00 00 00 30 12 12 12 12 12 12 12 12 12 12 12 10 00 10 00 00 00 00 00 00 
Sync               0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
Hex: 000000000

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: B198
SYNC_LOW: bucket 0: 033E
BIT_HIGH_TIME: bucket 2: 05BE
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x4d (22%)
BIT_LOW_TIME: bucket 1: 01AE
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x16 (22%)
BIT_COUNT + SYNC_BIT_COUNT: 0x27 (39, SYNC_BIT_COUNT = 0)
Data: 000000000
7FB198033E05BE4d01AE1627000000000
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F B198 033E 05BE 4d 01AE 16 27 000000000' == 0x10

'AA A8 10 7F B198 033E 05BE 4d 01AE 16 27 000000000 55'

```

Portisch, 2018-11-28:
This protocol can't be decoded right now with the rc-switch algorithmen. The sniffed data shows 4 different pulse arrangement for the bits. rc-switch/rf bridge is only supporting 2.
I am currently searching for another "universal" algorithmen to define and decode rf signals.
But this will take a little bit of time because I have to rewrite the whole firmware.
[image]

closed 2018-11-28 by Portisch

radeknoone, 2018-11-28:
Thank you for response and good job!