Portisch/RF-Bridge-EFM8BB1 #59: Need help learning Codes

henfri, 2018-12-04:
Hello,

Thanks for writing the RF-Bridge-Firmware!
I am using Tasmota on the Sonoff RF Bridge and the Firmware "RF-Bridge-EFM8BB1-20180816"
I am entering this on the Commandline of Tasmota:
`21:40:38 CMD: RfRaw AAB155
21:40:38 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
21:40:39 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
21:41:26 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 016E 02C4 3D22 00110011001100110011010100110010101102 55"}}
21:41:28 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02C6 3D22 00110011001100110011010100110010101102 55"}}
21:41:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02C6 3D22 00110011001100110011010100110010101102 55"}}
21:41:44 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0167 02C7 3D22 00110011001100110011010100110010101102 55"}}
21:41:46 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0166 02C8 3D36 00110011001100110011010100110010101102 55"}}
`
So, I run the BitbucketConverter.py:
`>python BitBucketConverter.py -d
Enter B1 line: 21:41:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02C6 3D22 00110011001100110011010100110010101102 55"}}
Filtered 0xB1 data: AAB103016802C63D220011001100110011001101010011001010110255
AAB103016802C63D220011001100110011001101010011001010110255
Second sync bucket is in front
Sync buckets: 20
New data: 20011001100110011001101010011001010110
DrawImage, try to draw a image of: 20011001100110011001101010011001010110
Bitcount: 18
Decoded value: 0x155D1
Resulting 0xB0 data: AA B0 1B 03 14 0168 02C6 3D22 20011001100110011001101010011001010110 55`

And consequently I run:
`22:08:01 CMD: RfRaw AAB01B0314016802C63D222001100110011001100110101001100101011055
22:08:01 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
22:08:01 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
`
But my device is not reacting...

I do realize, that the Codes I get when pressing buttons are not identical. I have tried some, but not all of the strings.

Am I doing everything right?

Regards,
Hendrik

henfri, 2018-12-08:
Hello,

thanks, that works!
So, what did you do different?

My commandline was:

> python BitBucketConverter.py -d
> Enter B1 line: 21:41:30 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0168 02C6 3D22 00110011001100110011010100110010101102 55"}}
> 

Greetings,
Hendrik

degsmfan, 2018-12-09:
There are a few different versions of the script available... I'm guessing that is what the difference is.... Or I used a different B1 reading...

I believe this is the version I used:

``` cat bb.py 
#-------------------------------------------------------------------------------
# Name:        BitBucketConverter.py
# Purpose:     Generate 'B0' message from received 'B1' data.
#
# Author:      gerardovf
#
# Created:     05/09/2018
#-------------------------------------------------------------------------------

import sys

# Output:
# Example: AAB035051412DE05C802D5017223A0012332232323323232322332323232322323323223233223322323323232323223323232232323233455
# 0xAA: sync start
# 0xB0: command
# 0x35: len command (dec 53)
# 0x05: bucket count
# 0x14: repeats
# buckets 0-4
# data
# 0x55: sync end

def main(szInpStr, repVal):
    #print("%s" % szInpStr)
    #print("%d" % repVal)
    listOfElem = szInpStr.split()
    #print(listOfElem)
    iNbrOfBuckets = int(listOfElem[2])
    #print("%d" % iNbrOfBuckets)
    # Start packing
    szOutAux = "AAB0xx"
    szOutAux += listOfElem[2]
    strHex = "%0.2X" % repVal
    #print(strHex)
    szOutAux += strHex
    for i in range(0, iNbrOfBuckets):
        szOutAux += listOfElem[i + 3]
    szOutAux += listOfElem[iNbrOfBuckets + 3]
    szOutAux += listOfElem[iNbrOfBuckets + 4]
    #print(szOutAux)
    iLength = len(szOutAux) / 2
    iLength -= 4
    #print(iLength)
    strHex = "%0.2X" % iLength
    #print(strHex)
    szOutFinal = "'RfRaw "
    szOutFinal += szOutAux.replace('xx', strHex)
    szOutFinal += "'"
    print(szOutFinal)

# In program command line put two values (received Raw between '"' and desired repeats)
# Example: "AA B1 05 12DE 05C8 02D5 0172 23A0 0123322323233232323223323232323223233232232332233223233232323232233232322323232334 55" 20
if __name__ == '__main__':
    szInpStr = sys.argv[1]
    repVal = int(sys.argv[2])
    main(szInpStr, repVal)```

henfri, 2018-12-09:
Thanks. 
do you still have the commandline you used?

Greetings,
Hendrik

degsmfan, 2018-12-09:
```[ ~]$ python -V
Python 2.7.13
[~]$ python bb.py "AA B1 03 0168 02C6 3D22 00110011001100110011010100110010101102 55" 4
'RfRaw AAB01B0304016802C63D220011001100110011001101010011001010110255'```

henfri, 2018-12-09:
Indeed, I get different output of the two converters:
`
AA B0 1B 03 14 0168 02C6 3D22 **2**0011001100110011001101010011001010110  55
AA B0 1B 03 14 0168 02C6 3D22  0011001100110011001101010011001010110**2** 55  
`

The upper one is the output of the one modified by Portisch, the lower one, the one from you.
I called yours with the repeat value of 20 (not 4 like in your example) as the portisch one defaults to 20.

you can see the difference highlighted.

Greetings,
Hendrik

degsmfan, 2018-12-09:
curious, 

though at least you are able to get what you need now I imagine...?

henfri, 2018-12-09:
not sure. testing...

Portisch, 2018-12-09:
My script is moving the sync bucket to the front. Normally this should not be a problem because you repeat it anyway.

henfri, 2018-12-09:
Yes, it happens here:
https://gist.github.com/Portisch/6dd5d6d20465521d76538a2907ba5e55#file-bitbucketconverter-py-L168

henfri, 2018-12-09:
@Portisch: Can you host your bitbucket script properly in the repository, please?
I would like to propose an improvement.

Portisch, 2018-12-09:
Next week - yes. You can also add a comment to the gist.

henfri, 2018-12-09:
Ok.

I can now confirm that moving the sync bucket seems causing my problem:

14:53:58 CMD: RfRaw AAB01B0304016802C63D220011001100110011001101010011001010110255

14:54:25 CMD: RfRaw AAB01B0304016802C63D222001100110011001100110101001100101011055

The first command works, the second not.

Do you have an explaination?

Greetings,
Hendrik

Portisch, 2018-12-09:
Just got the idea why it is not working with the "2" in front. It will invert all buckets. If on the end it is low. If on front it will be high.

Portisch, 2018-12-10:
[image]
Your signal is inverted! This is the reason why the script is moving the sync bucket on the wrong place.
In the picture you can see your "real" protocol. Just use the script and replace the rf data with the data from the B1 command until the script got fixed.
If you tell me a brand and device I can add the protocol decoding.

closed 2018-12-10 by Portisch

henfri, 2018-12-10:
Ok, thanks.

The Remote is this one:
https://manuals.easygates.co.uk/seav-txx-4-manual/

The device is a  extractor hood from Berbel:
https://www.berbel.uk/extractor-hoods/

Greetings,
Hendrik