Portisch/RF-Bridge-EFM8BB1 #60: New protocol? Need help dumping/decoding/resending

degsmfan, 2018-12-07:
Im trying to move away from my homemade Arduino + RF to some commercial hardware and why reinvent the wheel when there are great projects like this and Tasmota. So bought a Sonoff RF Bridge before powering it on, cracked it open and did the hardware mod's to flash the Portisch firmware. Flashed the latest Tasmota out of the `release-6.3` branch yesterday and then uploaded and flashed the latest Portisch firmware which came with the `master` repo: 
https://github.com/arendst/Sonoff-Tasmota/blob/master/tools/fw_efm8bb1/RF-Bridge-EFM8BB1-20181031.hex
(must have made a mixup there somewhere as I see now that that version is now in the `release-6.3` branch.  

So... off I went to see if I could get hold of some data which I could reproduce. 

some of my KaKu wall sockets were detected by the default so that was fine. Though I have another 8 (from Bauhaus) which are super wife friendly having a button on the side of them so you can manually turn them on and off. Rated 3680W and 16A, making me happy :P 

First time round dumping them gave me a double result for each code I had received.
```
15:49:19 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01B8 0438 2D82 01010101010101010101010101010101010101100110010102 55"}}
15:49:21 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 049C 2922 000000000001000100000000010001000000000100010000000001000101000100000000000000010100000000000101000000000100000001010001000000000000010001000002 55"}}
```

Though not being able to resend the data the quest went on... Now it seemed as it was sending just one per code, so I dumped some of those:

```
LR_SHELF ON
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04E2 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00C8 04E2 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04E2 28AA 01020302030203020303020302020302030302030203020302020303020302033030203020302030203020302030203020302020302030302020302030302030204F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04E2 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04EC 28AA 01020302030203020303020302020302030302030201203020203030203020203030203020302030203020302030203020302020302030302020302030302030204F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A82 00BE 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04F6 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0168 0A96 04EC 00C8 28B4 010302030203020302020302030302030202030203020302030302020302030302020302030203020302030203020302030203030203020203030203020203020304 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00BE 04EC 28A0 01020302030203020303020302020302030302030203020302020303020302020303020302030203023020302030203020302020302030302020302030302030204F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A82 00B4 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00BE 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00BE 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
OFF
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00BE 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0168 0A8C 00BE 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00BE 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00D2 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A82 00BE 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A82 00C8 04EC 28B4 01020302030203020303020302020302030302030203020302020303020302020303020302030203020302030203020302030202030203020302030203030201204F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 017C 0A82 00C8 04EC 28A0 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 04EC 00C8 28AA 01030203020302030202030203030203020203020301302030302020302030302020302030203020302030203020302030203030203020302030203020203020304F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0168 0A8C 00B4 04E2 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55"}}
```

What I noticed is that not only being much longer than what I had seen from most of other peoples RfRaw data and using 0,2,3's instead of 0,1's 🤷‍♂️ 

non the less tried to decode:

```
Enter B1 line: tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0172 0A8C 00C8 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55"}}
AA B1 05 0172 0A8C 00C8 04EC 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55
Repeat: 20
Sync: 40    Data: 10 20 30 20 30 20 30 20 30 30 20 30 20 20 30 20 30 30 20 30 20 30 20 30 20 20 30 30 20 30 20 20 30 30 20 30 20 30 20 30 20 30 20 30 20 30 20 30 20 30 20 20 30 20 30 30 20 20 30 20 30 30 20 30 20 
Sync               
Hex: 

The data for command 0xA8 will be:
SYNC_HIGH: bucket 3: 04EC
SYNC_LOW: bucket 0: 0172
BIT_HIGH_TIME: bucket 2: 00C8
BIT_HIGH_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 2: 0x6 (93%)
BIT_LOW_TIME: bucket 1: 0A8C
BIT_LOW_DUTY: (100% / (bucket 2 + bucket 1)) * bucket 1: 0x5d (93%)
BIT_COUNT + SYNC_BIT_COUNT: 0x0 (0, SYNC_BIT_COUNT = 0)
Data: 
7F04EC017200C860A8C5d0
Protocol is 0x7F (unknown), Len xx is counting bytes: '7F 04EC 0172 00C8 6 0A8C 5d 0 ' == 0x0B

'AA A8 0B 7F 04EC 0172 00C8 6 0A8C 5d 0  55'
[... 49 lines cut ...]

Previously on my Arduino I was using a library that supported these new style Kaku switches (where you press the button on the socket it enters learning mode and then you send a code/press button on remote)
https://github.com/1technophile/NewRemoteSwitch

[... cut ...]
home/OpenMQTTGateway/commands/MQTTtoRF2/UNIT_3/PERIOD_256 1
home/OpenMQTTGateway/RF2toMQTT/CODE_3398654/UNIT_3/GROUP_0/PERIOD_256 1
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 02 07B2 0F28 01 55"}}
tele/sonoffbridge/STATE {"Time":"2018-12-07T13:50:04","Uptime":"0T00:10:13","Vcc":3.160,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
```

I hope this didn't become to lengthy, just want to be as complete as possible here.

Portisch, 2018-12-07:
This protocol isn't handled by the firmware < V3. Try this firmware: https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex
And try 0xB1 sniffing again. Bauhaus haben wahrscheinlich das Intertechno V3 Protokoll.

degsmfan, 2018-12-07:
Hey, thanks for your answer, I uploaded the firmware you suggested. Which needed a good pull from the power after it rebooted as no RF was working.. seems to be fine afterwards though.

Here the on followed by the off, what is curious is that as I am writing this post it is still "dumping" data.... Ill add some asterisk's 

It Is possible they use the same protocol, no idea, I know they do use the same as the "new Kaku"...

```
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AAA055"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 00FA 0514 050A 01010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07C6 0154 00F0 050A 28C8 001213121312131213131213111113111313111311131113111113131113111213131113111311131113111311131113121311111312131312121311131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 04F6 00F0 0500 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07C6 0168 00DC 0500 28AA 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131312121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 015E 04EC 00F0 0500 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 015E 00DC 0500 28B4 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131312121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 015E 04EC 00DC 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07B2 015E 00DC 0500 28AA 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131312121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 04F6 00E6 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07BC 015E 00DC 04F6 28B4 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131312121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 0500 00DC 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 02C6 0442 01D6 015E 2D78 01212113212121212121211221212121212121132112211224 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 04F6 01CC 2D8C 00101001101010101010100110101010101010011001100112 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0118 0078 0154 011102 55"}}

tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0172 00C8 04BA 04EC 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07DA 015E 00E6 0500 28BE 001213121312131213131213121213121313121312131213121213131113121213131213111312131113111312131213121312121312131213121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 015E 0500 00E6 050A 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 0168 00DC 0500 28B4 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131213121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 04EC 00F0 04EC 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 0168 00DC 0500 28B4 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131213121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0154 04F6 00E6 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07BC 015E 00DC 0500 28AA 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131213121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 015E 04F6 00E6 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07BC 015E 00DC 0500 28AA 001213121312131213131213121213121313121312131213121213131213121213131213121312131213121312131213121312121312131213121312131312131214 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 015E 04F6 00E6 04F6 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 02C6 0438 01D6 015E 2D82 01212113212121212121211321212121212121132112212124 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 04F6 01CC 015E 2D82 00101001101010101010100210101010101010011001101013 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 04F6 01C2 2D82 00101001101010101010100110101010101010011001101012 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 04EC 01CC 015E 2D8C 00101002101010101010100110101010101010021001101013 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 079E 04CE 0F28 0F3C 0110020000020203 55"}}
*************************
From about here on
*************************
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 07B2 0316 0F1E 0F28 00110203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0082 014A 00D2 0208 0102123F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 006E 01CC 00FA 01B8 012023 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 015E 00C8 015E 011102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 079E 0F32 0F50 00010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0064 07D0 05E6 0F32 01211123 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F1E 0F3C 00010102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07B2 0B40 0154 008C 0F1E 01230004 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0280 07E4 082A 010102 55"}}
[... cut ...] 3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07BC 0F1E 030C 010E 0F0A 00010234 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 06 07A8 0F28 00FA 00BE 05AA 0F14 010002340105 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F28 0F28 010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 08AC 0690 0F28 0F32 01020203 55"}}
```

degsmfan, 2018-12-07:
what maybe worth nothing in the original post each line was a button press, these just dumped was one on and one off press.

degsmfan, 2018-12-07:
FWIW it is still dumping while nothing else is going on, it seems that there is an overflow somewhere...

```
tele/sonoffbridge/STATE {"Time":"2018-12-07T21:35:54","Uptime":"0T00:15:14","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F1E 0E9C 01010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0866 0F28 0F14 001010102F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F14 0F28 000102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 07B2 0F1E 008C 0F1E 01000023 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 021C 1018 0816 05D2 010203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 076C 0F50 0F50 01010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 078A 00E6 02D0 0F3C 000123 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 06 079E 00DC 019A 051E 0F1E 0F1E 01230405 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0078 00E6 04C4 0190 0100203F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 0064 0226 015E 0F1E 00100000200314 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07BC 0F1E 0302 01FE 0F28 01000234 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 06 0410 0776 22D8 008C 583E 0B04 012345 55"}}
tele/sonoffbridge/STATE {"Time":"2018-12-07T21:40:54","Uptime":"0T00:20:14","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 079E 0F3C 047E 01D6 0F46 00010234 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 01F4 0370 0F1E 0F1E 00012003030004 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0064 00B4 015E 011102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 00FA 03CA 0096 0294 012023 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F14 0F14 01010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0082 0118 00BE 0168 0011203F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 07A8 0334 015E 0F1E 000123 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 06 085C 0078 01E0 3CF0 02EE 0F1E 012345 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 079E 0F1E 0370 012C 0F1E 01010234 55"}}
tele/sonoffbridge/STATE {"Time":"2018-12-07T21:45:54","Uptime":"0T00:25:14","Vcc":3.447,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0230 012C 023A 000000012F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0136 080C 0208 006E 049C 00000000000000000000000000000000010000000000000000000000023000000000000000000000000000000000000000000000000004 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 023A 0136 0258 0110112F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0122 0442 1022 00D2 028A 000001234F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F28 0F28 00010102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 0064 0104 04F6 1040 0230 012304 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0794 0F32 0F46 00010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07BC 0F32 0898 00000012 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F28 0F28 010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0078 0122 0208 014A 010213 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 00F0 006E 01FE 000112 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07A8 0B4A 00BE 0320 0F3C 0123000100000104 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 07B2 037A 053C 0F28 00120203 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 006E 00E6 015E 01C2 0011223F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 078A 00BE 0F46 00001002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F14 0E6A 00010102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07BC 0F1E 0F14 000102 55"}}
tele/sonoffbridge/STATE {"Time":"2018-12-07T21:50:54","Uptime":"0T00:30:14","Vcc":3.462,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"PadHap","BSSId":"12:9F:C2:7A:2E:ED","Channel":11,"RSSI":100}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F28 0F1E 010102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 07A8 058C 0122 0F1E 0000012003 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 008C 01CC 0320 000102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0C26 0F1E 0101010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F1E 0F1E 010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 01D6 0384 0154 0DC0 012103 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 04 0244 0820 05BE 104A 010123 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 07B2 058C 0208 0370 0F28 01232204 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07B2 0F1E 0F28 01010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 05 01A4 006E 0280 00A0 01CC 012134 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0082 0104 014A 001102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07A8 0F1E 0F3C 00010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 079E 0F28 0E56 010101002F 55"}}
[... cut ...]
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 07 00BE 0172 0118 0370 021C 0064 01FE 0101210101221111021110110121010101010103011112111014401014201012114444444401135010104410106F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0230 0118 0244 000000012F 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 0258 012C 03D4 00111111111111011011011002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 079E 0F32 0F46 00010002 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 07BC 0F14 0F1E 01000102 55"}}
tele/sonoffbridge/RESULT {"RfRaw":{"Data":"AA B1 03 079E 0F14 0F1E 000102 55"}}
```

Portisch, 2018-12-08:
On the last firmware I decreased the sync pulse detection down to 300ys. Before it was 1300ys. This is may the reason.

degsmfan, 2018-12-09:
Do you have any ideas on what I may need to do to get the raw data in a re-sendable format?

Or is this a it is just not going to work using this stack (at the moment)?

Portisch, 2018-12-10:
```AA B1 05 0172 0A8C 00C8 04E2 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55```
Is converted to:
```AA B0 4E 05 14 0172 0A8C 00C8 04E2 28AA 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030302020302030302030204 55```

```AA B1 05 0172 0A8C 00C8 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55```
Is converted to:
```AA B0 4E 05 14 0172 0A8C 00C8 04EC 28B4 010203020302030203030203020203020303020302030203020203030203020203030203020302030203020302030203020302020302030203020302030302030204 55```

degsmfan, 2018-12-10:
unfortunately neither of these worked...

Portisch, 2018-12-10:
Please try the new firmware from the generic_bucket branch. Do again some 0xB1 sniffing.

degsmfan, 2018-12-10:
uploaded the latest firmware though, no output...

```
16:57:21 CMD: RfRaw 1
16:57:21 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
16:57:27 CMD: rfraw 177
16:57:27 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
```

and that's it :(

Portisch, 2018-12-10:
I don't have any working hardware here. So I can't test it before next week.

degsmfan, 2018-12-10:
I understand, if you need to me to do something let me know. Thanks so far.

degsmfan, 2018-12-10:
well I a refresh and a reboot got it going (sorry)

```
17:08:40 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 058C 0564 808182 55"}}
17:08:40 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0546 0528 818082 55"}}
17:08:40 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 04E2 053C 808182 55"}}
17:08:41 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 019A 04BA 00DC 0514 818283 55"}}
17:08:41 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 0514 0528 818082 55"}}
```

and off
```
17:09:30 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0546 0550 818082 55"}}
17:09:31 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 017C 04E2 00E6 0514 818283 55"}}
17:09:31 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0122 050A 051E 808182 55"}}
```

Portisch, 2018-12-11:
I found a bug in the 0xB1 sniffing. Please try again with this firmware:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

benjaminlundgreen, 2018-12-11:
Hello, I believe i have the same problem as degsmfan. I am using Nexa remote switches which seem to use a similar/same protocol as/to the KaKu switches. If it helps, this is what i get with the latest firmware:

13:12:18 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 012C 0A00 008C 047E 27F6 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:18 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 00D2 04BA 04BA 78182838 55"}}
13:12:18 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 09F6 008C 047E 2800 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:19 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 019A 0442 0082 0424 78182838 55"}}
13:12:19 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0136 0A0A 00AA 0488 27F6 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:19 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0118 00C8 04BA 04BA 78182838 55"}}
13:12:19 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 09F6 008C 047E 2800 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:20 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 014A 00A0 049C 047E 78182838 55"}}
13:12:20 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 012C 0A14 00A0 0488 27F6 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 00D2 04BA 04BA 78182838 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 014A 0A00 0096 047E 2800 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0172 0078 0488 0474 78182838 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 012C 0A0A 00A0 0492 27F6 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0122 00C8 04B0 04BA 78182838 55"}}
13:12:21 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 09F6 008C 047E 27F6 78182838382828383828382838283828283828383828382828383828283838283828382838282838382828382838283838283828283828383828382838283828382800 55"}}

Something must be wrong right? Considering the eights and sevens in the data with only five time buckets.

Portisch, 2018-12-11:
No, this is correct. The first bit of each nibble is the mark if high or low bucket...

Portisch, 2018-12-11:
I pushed a new commit what should fix the bucket marking.
https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/f82219b638ba2b8a34f330d11154472dd9ccd0bf

Portisch, 2018-12-11:
@benjaminlundgreen I just added a new branch what should be able to decode the protocol:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/Kaku/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/5e02efe7bf219ebee2551c2fea60cd59b4f3cbd2

degsmfan, 2018-12-12:
Hey thanks for reopening this, I have been away with work but... with:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

```
18:48:59 CMD: RfRaw 177
18:48:59 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
18:48:59 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00E6 015E 0564 0546 381828 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0104 0546 055A 281808 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 015E 00DC 0528 04EC 381828 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 04F6 04D8 280818 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 014A 00E6 0500 050A 381828 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 0514 0514 280818 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0154 00E6 051E 050A 381828 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 050A 04EC 281808 55"}}
18:49:06 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 014A 00E6 0514 0514 381828 55"}}
18:49:07 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0136 01D6 0442 065E B09282 55"}}
18:49:07 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01B8 044C 0442 281908 55"}}
```

with:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/generic_bucket/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

```
18:53:58 CMD: RfRaw 177
18:53:58 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
18:53:58 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
18:54:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 010E 0578 0550 280818 55"}}
18:54:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 014A 050A 00D2 0500 381828 55"}}
18:54:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 00A0 04F6 0500 381828 55"}}
```

and;
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/Kaku/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex

```
18:58:01 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 01EA 00C8 008C 0154 381829 55"}}
18:58:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0118 0564 0528 281808 55"}}
18:58:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0154 00E6 050A 0500 381828 55"}}
18:58:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 050A 050A 280818 55"}}
18:58:03 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 015E 04D8 00FA 0500 381828 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0154 0294 04E2 289918 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0154 00E6 0500 053C 381828 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 012C 04CE 04EC 280818 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 05 01F4 008C 0140 0500 0514 481A3A 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0136 050A 04F6 280818 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 04 0154 04F6 00A0 04C4 381828 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0140 04F6 0514 280818 55"}}
18:58:04 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AA B1 03 017C 046A 062C A08181 55"}}
```

Also nothing found with 0xA6

```
19:07:51 CMD: RfRaw AAA655
19:07:51 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
19:07:51 MQT: tele/sonoffbridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
```

dkrix, 2019-01-01:
Happy New Year everyone.

I have an smartwares SH5-TSM-A window switch that I bought at a local hagebau hardware store.  It apparently has the same arcswitch/nexa/proove protocol the Kaku switch has. rtl433 from a linux box regcognizes the switch alright.

Using the kaku-Branch RfRaw 177 gives me something like this atleast:
11:16:44 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 05 0140 097E 00C8 04A6 28DC 481828383828283838282838382838283828283838283828382828383828283828382838283828382838283828382838283828382838283838282838283828383828 55"}}

RfRaw 166 does not give me any output with the exception off some AAA055 command confirm. The command does not report a window switch (renkforce CD100S) by Conrad that otherwise gets detected in the regular mode and used the orginal RF bridge protocol, too.

I used the simplicity IDE to compile the branch with no difference to the release HEX fle. I even changed the timing a bit -> no help.

What's interesting is that the Kaku and bucket branch recognize the pause bit first and not the sync, which was different in the master branch.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

Portisch, 2019-02-19:
I updated the Kaku branch with the latest fixes. Maybe it is working now with command 0xA6:
https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/Kaku/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex