Portisch/RF-Bridge-EFM8BB1 #65: Byron BY302 Doorbell

Idcsl, 2018-12-12:
Hi,

Thanks for your efforts in creating this alternative firmware. I have spent much time learning about the RF Bridge and the RF chip. However I have drawn a blank when it came to trying to decode the codes for this unit and I would therefore appreciate very much help in determining whether it is actually feasible to decode the codes or if there is something unusual with this protocol.

Here is a snapshot of codes:

14:58:35 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03F2 0BB8 010010110011001100101100101011010101010102 55"}}
14:58:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BCC 010010110011001100101100101011010101010102 55"}}
14:58:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03DE 0BAE 010010110011001100101100101011010101010102 55"}}
14:58:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 0208 03DE 0BB8 010010110011001100101100101011010101010102 55"}}
14:58:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03F2 0BC2 010010110011001100101100101011010101010102 55"}}
14:58:36 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01FE 03DE 0BA4 010010110011001100101100101011010101010102 55"}}
14:58:37 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BCC 010010110011001100101100101011010101010102 55"}}
14:58:43 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01AE 0424 0BE0 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01FE 03DE 0BCC 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BC2 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 024E 03DE 0BC2 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BC2 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BAE 010010110011001100101100101011010101010102 55"}}
14:58:44 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01EA 03E8 0BAE 010010110011001100101000101011010101010102 55"}}
14:58:45 MQT: tele/SonoffRFBridge/RESULT = {"RfRaw":{"Data":"AA B1 03 01F4 03E8 0BC2 010010110011001100101100101011010101010102 55"}}

Regards
Ian

degsmfan, 2018-12-12:
what have you tried to decode?

```
$ python bb.py "AA B1 03 01EA 03E8 0BAE 010010110011001100101100101011010101010102 55" 4
'RfRaw AAB01D030401EA03E80BAE01001011001100110010110010101101010101010255'
```

Idcsl, 2018-12-13:
I think I have been confused with a varying number of versions of the bit bucket converter that I have tried. However, I am able to apparently decode the codes shown now as I see the resulting RfRaw data as you have shown but sending this data via command 176 does not activate the doorbell. Trying various codes as I had shown has the same result.

I am actually only interested in receiving the codes as would be using this via MQTT in Home Assistant if I can get it working.

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch