Portisch/RF-Bridge-EFM8BB1 #66: after flashing with latest (20181127) file in rf bridge, it looks like rf part is dead

woudenbe, 2018-12-12:
th bridge seems to be working but rf part is dead. red led is not even blinking.

pls help

woudenbe, 2018-12-12:
gave it other try...
22:55:00 UPL: File RF-Bridge-EFM8BB1-20181127.hex ...
22:55:05 HTP: Upload done
22:55:05 Upload: Failed to write to RF chip

Portisch, 2019-01-07:
The EFM8B1 died on my side 2 times. Now I am waiting for RF Bridge nr. 3. I used the https://github.com/conorpp/efm8-arduino-programmer without problems for a long time. After I switched to the Tasmota methode I killed two EFM8. So be carefully when flashing the IC.
The EFM8 will be dead if the flashing stops and the red LED will lighten and the beeper make a constant sound... The problem is that the flash can't be erased anymore.

closed 2019-01-07 by Portisch

jluzonpr, 2019-02-28:
@Portisch Does failed to write to RF Chip pretty much mean chip died?

Portisch, 2019-02-28:
No idea! I got a lot of write faild by the Tasmota flash update. After a while the flash could not be erased anymore. I using now an extra Arduino to flash the EFM8B1 and got no problems so far...

jluzonpr, 2019-02-28:
I actually got to write it with some help from Folks at the Tasmota Discord, so definitely does not automatically mean dead chip. :D

vmazmaz, 2019-12-09:
I'm facing the same 'Failed to write RF chip' error, would you mind sharing how you manage to get it to write ?

jluzonpr, 2019-12-12:
I can't remember what it was exactly, but I remember having an issue of 2 of the pins I soldered being shorted somehow and cleaning up the pins' solder helped fix SOMETHING.

pabloeorsi, 2020-05-30:
> The EFM8B1 died on my side 2 times. Now I am waiting for RF Bridge nr. 3. I used the https://github.com/conorpp/efm8-arduino-programmer without problems for a long time. After I switched to the [...]

Hi! I have the same problem and I can`t solve it! Is it possible? Please help me, because I don't purchase other :-(

Pardon for my English.
Thanks!!!