Portisch/RF-Bridge-EFM8BB1 #69: KERUI F56 Kinetic Doorbell

vampywiz17, 2018-12-17:
Hello there,

I bought a new Kinetic doorbell (KERUI F56 ), but it not see it the RF Bridge. Now, i able to take some picture inside, i hope it help...

https://i.imgur.com/cfHs56t.jpg

The chip text is: cmt2150A

Portisch, 2019-01-07:
I think you have to meassure with any analyse hardware to find the used protocol of this transmitter.
As example: if the data get only repeated 2 times like my weather station it can't be sniffed by 0xB1!

closed 2019-01-07 by Portisch

vampywiz17, 2019-01-09:
@Portisch 

i bought a Arduino (Wemos® Mega +WiFi R3 Module ATmega2560+ESP8266 ) and RF receiver[ this type](https://www.aliexpress.com/item/QIACHIP-433mhz-Mini-Low-Power-RF-Relay-Receiver-Transmitter-Module-Remote-Control-Switch-For-Arduino-Uno/32831237349.html?spm=2114.13010708.0.0.3a024c4drdOmyN) It help to find out the protocol?

I install my arduino to [RFlink](http://www.rflink.nl/blog2/)) but it not see this (like Sonoff RF)

Any way to analyse it?

Portisch, 2019-01-09:
Maybe not 433MHz?

vampywiz17, 2019-01-09:
It say that is 433 mhz. [Link](https://www.aliexpress.com/item/KERUI-WaterProof-Touch-Doorbell-Button-Wireless-32-Songs-Optional-Door-Bell-Kits-EU-US-UK-plug/32818601228.html?spm=2114.13010708.0.0.3f974c4dVr6Zr9)

and the chip that use, is 433 mhz.  (cmt2150A)

Portisch, 2019-01-09:
Part Number Frequency Package MOQ
CMT2150A-ESR Random SOP14 2,500 pcs
CMT2150A-ESR3 315.00 MHz SOP14 2,500 pcs
CMT2150A-ESR4 433.92 MHz SOP14 2,500 pcs

vampywiz17, 2019-01-09:
Well, i will buy a 315 mhz receiver and we will see it :)

Thanks!

vampywiz17, 2019-02-01:
@Portisch I got the 315 mhz receiver but no any signal... I think it is 433mhz (it say all possible order page), now that you continue the development, that not possible that somehow find out, why not see any kinetic devices? 

I think the Kinetic devices use a different sending method, that the "regular" devices can't recognise it...