Portisch/RF-Bridge-EFM8BB1 #71: B0 codes not activating device

DarkStorm99, 2018-12-31:
This is a bit of a steep learning curve for me, but I'll try and explain this as best I can. I have a couple of ceiling fans that I was hoping to use the bridge with. I have flashed the custom rf firmware to the bridge. Before flashing the custom firmware, nothing would show in the tasmota console when snooping in on transmissions. 

On rf firmware Keil 8051 v9.53, I have to press the buttons on the remote quite a few times for anything to show in the console after typing rfraw 177. The console picks up a fraction of these only:

```
06:36:13 CMD: rfraw 177
06:36:13 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
06:36:13 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
06:36:17 MQT: Attempting connection...
06:36:18 DNS: Query done. MQTT services found 0
06:36:29 MQT: Attempting connection...
06:36:30 DNS: Query done. MQTT services found 0
06:36:38 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 019A 00E6 0302 22E2 01200000000000000120000003 55"}}
06:36:38 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 01A4 00E6 02EE 22EC 01200000000000000120000003 55"}}
06:36:41 MQT: Attempting connection...
06:36:42 DNS: Query done. MQTT services found 0
```

Probably 20 or so keypresses to get that result. Running those through the bitbucketconverter gave me the B0 codes, which I tried to insert as follows:

```
06:46:51 CMD: rfraw AA B0 17 04 04 019A 00E6 0302 22E2 012000000000000001200000 03 55
06:46:51 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
06:46:51 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
06:48:01 CMD: rfraw AA B0 17 04 04 01A4 00E6 02EE 22EC 012000000000000001200000 03 55
06:48:01 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
06:48:01 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
```
Neither had a result

I then tried the copy that came with the Tasmota data in the Tools folder. 
RF-Bridge-EFM8BB1-20181127.hex had the same results as above, needing to spam keys just to pick up anything, but nothing gets changed when sending the B0 code
RF-Bridge-EFM8BB1-20181102.hex kept the red rf led flashing sporadically without any input at all and didnt pick up any B1 codes
RF-Bridge-EFM8BB1-20181023.hex seemed the most promising, as every keypress shows in the console, but sometimes one keypress resulted in many lines. This example was a different button on the remote. 

```
06:05:59 CMD: rfraw 177
06:05:59 RSL: stat/sonoff/RESULT = {"RfRaw":"ON"}
06:06:00 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AAA055"}}
06:06:04 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 028F 0165 22CE 10101100101010101100101012 55"}}
06:06:04 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 027D 0176 0104 22E2 10101200101010101200101013 55"}}
06:06:05 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 028C 0169 0104 22E2 10101200101010101100101013 55"}}
06:06:05 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0293 0159 22F6 10101100101010101100101012 55"}}
06:06:07 MQT: Attempting connection...
06:06:08 DNS: Query done. MQTT services found 0
06:06:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0290 015F 22C4 10101100101010101100101012 55"}}
06:06:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0281 0177 00FA 22EC 10101200101010101200101013 55"}}
06:06:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0292 0159 22E2 10101100101010101100101012 55"}}
06:06:08 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0281 0172 00FA 22E2 10101200101010101200101013 55"}}
06:06:10 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0292 015F 22EC 10101100101010101100101012 55"}}
06:06:10 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 027D 0178 0104 22E2 10101200101010101200101013 55"}}
06:06:10 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0293 0159 22EC 10101100101010101100101012 55"}}
06:06:12 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0290 015C 22F6 10101100101010101100101012 55"}}
06:06:12 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0281 0177 00FA 22EC 10101200101010101200101013 55"}}
06:06:12 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0293 0159 22F6 10101100101010101100101012 55"}}
06:06:14 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 028F 015F 22C4 10101100101010101100101012 55"}}
06:06:14 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 027E 017A 00FA 22E2 10101200101010101200101013 55"}}
06:06:14 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0298 0156 22EC 10101100101010101100101012 55"}}
06:06:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 0294 0167 00FA 22BA 10101200101010101100101013 55"}}
06:06:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 028D 0162 22F6 10101100101010101100101012 55"}}
06:06:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 04 027E 0176 00FA 22E2 10101200101010101200101013 55"}}
06:06:17 RSL: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0293 0159 22EC 10101100101010101100101012 55"}}
```

Unfortunately converting each of these into B0 codes, either one by one and trying it, or by trying 
bitbucketconverter.py -f v1firmware.txt -e 192.168.1.129 , all again showed the same result of not having any effect. 

I'm out of ideas at this point. My gut feeling is the B0 codes aren't correct somehow, and I've tried different firmware and button combinations for about 6 hours now without any progress.

sarreck, 2019-01-07:
I am seeing the same issue with a LUCCI Slim Line Fan Remote Control. This appears to be a Rhine Electronic Co remote control model uc7216t.

It looks like RF control of this fan series has been implemented in the RFXCom device with the quirk that only the fan receiver works. That is the remote control RF commands are not recognised by the RFXCom device, but the RFXCom device can send commands to the directly to the fan receiver. 

This appears to reflect the behaviour we are seeing. 

There is a little more information here:
https://www.domoticz.com/forum/viewtopic.php?t=15499
https://github.com/openhab/openhab1-addons/issues/5155

So far I haven't been able to find how the RFXCom system talks to the fan.

Final-Hawk, 2021-02-24:
> I am seeing the same issue with a LUCCI Slim Line Fan Remote Control. This appears to be a Rhine Electronic Co remote control model uc7216t.

Have you found a solution yet? Having the same issue