Portisch/RF-Bridge-EFM8BB1 #73: Grundig doorbell

niekvlessert, 2019-01-05:
Hello,

I'm pretty new to this whole RF world, but I got advice to add a ticket here for my issue. I have the Sonoff Bridge 2.0 and I've installed Tasmota and the Portisch RF firmware. Thanks for all the effort!

I have a Grundig doorbell and I want to add a rule to Tasmota so stuff happens when the doorbell gets pressed. The data shows up when pressing the doorbell using rfraw 177 and rfraw 166, but not using rfraw 0 or rfraw 1. However I can't create a rule based on rfraw data in Tasmota, only on rfreceived. So there's no way to do this afaik.

Some data that comes in on rfraw 177 is:

23:53:03 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 042E 0186 16E4 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16B2 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16B2 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0168 16A8 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16A8 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16B2 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
23:53:04 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
23:53:05 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16B2 010101001011010010110100101101010102 55"}}
23:53:05 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 169E 010101001011010010110100101101010102 55"}}

and

00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 042E 0186 16DA 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16B2 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
00:09:42 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
00:09:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}
00:09:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0172 16A8 010101001011010010110100101101010102 55"}}
00:09:43 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0168 16A8 010101001011010010110100101101010102 55"}}

I've been told this might mean an unsupported protocol?

Portisch, 2019-01-30:
Try sniffing again with the latest [master](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF-Bridge-EFM8BB1.hex)!

I close the issue because of a buggy EFM8 firmware - please do a 0xB1 sniffing again and I will reopen the issue.

closed 2019-01-30 by Portisch

niekvlessert, 2019-11-08:
Hello, I picked this up again and I installed the latest firmware, RF-Bridge-EFM8BB1-20191006.hex.

I now get this when i push the doorbell:

23:09:23 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 042E 0190 1702 281818180909181809091818090918181818 55"}}
23:09:24 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 044C 0172 16B2 281818180909181809091818090918181818 55"}}
23:09:25 MQT: tele/sonoff/RESULT = {"RfRaw":{"Data":"AA B1 03 0442 0168 16BC 281818180909181809091818090918181818 55"}}

By using the B1 to B0 convertor I got the data to send and it works, my bell is ringing, pretty awesome:

Rfraw "AA B0 1A 03 08 042E 0190 1702 281818180909181809091818090918181818 55"
Rfraw "AA B0 1A 03 08 044C 0172 16B2 281818180909181809091818090918181818 55"
Rfraw "AA B0 1A 03 08 0442 0168 16BC 281818180909181809091818090918181818 55"

However it would be even better if I can make the rf bridge act on the incoming signals, so I can make an event happen when the doorbell gets pressed.

I think the doorbell has to be added to the code, or is there already some way to make the bridge act on those signals? I guess not since it requires rfraw 177 to get the data...